pymodbus-dev/pymodbus #2266: Dealing with a device response that has a wrong length field

michael-sayapin, 2024-07-26:
### Versions

- Python: 3.11.9
- OS: Linux xxx 5.15.139 #0 SMP Sat Dec 9 12:39:53 2023 aarch64 GNU/Linux
- Pymodbus: 3.6.9
- Modbus Hardware (if used): TCP

### Pymodbus Specific

- Server: 3rd party
- Client: tcp sync/async

### Description

I'm connecting to commercial Midea HVAC gateway device (very popular in China). Certain versions of this device have a bug, response length field is not correct. In the example below, it promises 6 bytes but sends 5 bytes. It should be 5 bytes indeed. I'm trying to find a way to deal with this, since these gateways are installed all over in places like universities, hospitals, and airports, where updating/fixing them will take years, if ever. 

What would be the best way to tackle this? I'm thinking to do something similar to uModbus, where they just infer `expected_response_size` from the request ADU. (However, uModbus is broken is many other ways...) Would this belong to the framer?.. 🤔

### Code and Logs

```python
import pymodbus
from pymodbus.client import ModbusTcpClient
print(pymodbus.__version__)
pymodbus.pymodbus_apply_logging_config("DEBUG")
client = ModbusTcpClient('192.168.1.200', 502, timeout=5)
client.connect()
response = client.read_input_registers(1923, count=1, slave=1)
print(response)
client.close()
```

```
>>> client.connect()
2024-07-26 03:05:52,548 DEBUG logging:102 Connection to Modbus server established. Socket ('192.168.1.50', 52076)
>>> response = client.read_input_registers(1923, count=1, slave=1)
2024-07-26 03:05:52,552 DEBUG logging:102 Current transaction state - IDLE
2024-07-26 03:05:52,554 DEBUG logging:102 Running transaction 1
2024-07-26 03:05:52,555 DEBUG logging:102 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x7 0x83 0x0 0x1
2024-07-26 03:05:52,556 DEBUG logging:102 New Transaction state "SENDING"
2024-07-26 03:05:52,557 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-07-26 03:05:57,565 DEBUG logging:102 Incomplete message received, Expected 12 bytes Received 11 bytes !!!!
2024-07-26 03:05:57,566 DEBUG logging:102 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-07-26 03:05:57,568 DEBUG logging:102 RECV: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x2 0x0 0x1a
2024-07-26 03:05:57,571 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x2 0x0 0x1a
2024-07-26 03:05:57,572 DEBUG logging:102 Frame check failed, ignoring!!
2024-07-26 03:05:57,574 DEBUG logging:102 Getting transaction 1
2024-07-26 03:05:57,577 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
>>> print(response)
Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
```

janiversen, 2024-07-26:
This is not something we support, mainly because building a logic to determine what is correct is complicated.

Framer or transport sounds like places to patch, but I do not see this as a product feature.

janiversen, 2024-07-26:
Normally you would not patch transport, as this is too low level for most situations.

You can wait, use dev or a release, it all depends on your needs and not something I can help you with. We will keep touching the framers for a foreseeable future, since we need to integrate the new framer structure.

janiversen, 2024-07-26:
Patching framer is actually very simple, we really have no idea how to make it simpler for general use, but as with anything in life, you need to know what you are doing.

janiversen, 2024-07-30:
Just one more comment, we actually provide hooks in transport, allowing you to intercept incoming messages and outgoing messages...this is a standard functionality, so thinking about it you do not even need to patch anything.

Closing as solved.

closed 2024-07-30 by janiversen