pymodbus-dev/pymodbus #2273: Issue reading several consecutive register with serial pymodbus simulator

Tamanar, 2024-07-31:
### Versions

- Python: Python 3.11.9
- OS: W10
- Pymodbus: 3.6.9
- Modbus Hardware (if used): Modbus simulator  

### Pymodbus Specific

- Server: rtu - sync
- Client: rtu sync

### Description

I'm using pymodbus with the serial pymodbus simulator. I can successfully read a single registers even if it's made of several register, but when trying to read a several consecutive register, I encounter an error. 
Here's a snippet of the json I use for the modbus simulator:
```json
"string": [
		    {"addr": [30000, 30014], "value": "model_name12345"},
		    {"addr": [30015, 30024], "value": "sn01234567"},...
]
```

- Reading 13 registers works fine:
  `response = self.serial.read_holding_registers(30000, 13)`
  This successfully retrieves the model_name12345.

- However, when trying to read consecutive registers to get all the data at once:
  `response = self.serial.read_holding_registers(30000, 23)`
  I get the following error:
```shell
DEBUG:pymodbus.logging:Current transaction state - IDLE
2024-07-31 09:28:53,752 DEBUG logging:103 Running transaction 1
DEBUG:pymodbus.logging:Running transaction 1
2024-07-31 09:28:53,753 DEBUG logging:103 SEND: 0x0 0x3 0x75 0x30 0x0 0x17 0x1e 0x16
DEBUG:pymodbus.logging:SEND: 0x0 0x3 0x75 0x30 0x0 0x17 0x1e 0x16
2024-07-31 09:28:53,754 DEBUG logging:103 Resetting frame - Current Frame in buffer -
DEBUG:pymodbus.logging:Resetting frame - Current Frame in buffer -
2024-07-31 09:28:53,755 DEBUG logging:103 New Transaction state "SENDING"
DEBUG:pymodbus.logging:New Transaction state "SENDING"
2024-07-31 09:28:53,756 DEBUG logging:103 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-07-31 09:28:54,760 DEBUG logging:103 Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received))
DEBUG:pymodbus.logging:Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)) 
2024-07-31 09:28:54,761 DEBUG logging:103 Processing:
DEBUG:pymodbus.logging:Processing:
2024-07-31 09:28:54,762 DEBUG logging:103 Getting transaction 0
DEBUG:pymodbus.logging:Getting transaction 0
2024-07-31 09:28:54,763 DEBUG logging:103 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
ERROR:MarsSoilingSensor:Error reading from sensor: Modbus Error: Modbus error reading serial number: Modbus Error: [Input/Output] Modbus Error: 
[Invalid Message] No response received, expected at least 4 bytes (0 received)
INFO:root:Error: Modbus Error: Modbus error reading serial number: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
```
I need to get several register at once because I use a package ([huawei-solar](https://github.com/wlcrs/huawei-solar-lib)) that use this.
### Code and Logs

Serial modbus simulator :

```python
import asyncio
import logging
import sys
from pymodbus.server import ModbusSimulatorServer, get_simulator_commandline
import pymodbus
pymodbus.pymodbus_apply_logging_config("DEBUG")
logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger('main')
async def run_simulator(device_to_simulate):
	"""Run server."""
	_logger.info("### start server simulator")
	cmdline = [
		"--json_file", "simulator_config.json",
		"--modbus_server", "server_serial",
		"--modbus_device", device_to_simulate,
		"--log", "debug"
	]
	cmd_args = get_simulator_commandline(cmdline=cmdline)
	task = ModbusSimulatorServer(**cmd_args)
	await task.run_forever()

	_logger.info("### simulator running, press Ctrl+C to stop")
	try:
		while True:
			await asyncio.sleep(1)
	except KeyboardInterrupt:
		pass

	_logger.info("### shutdown server")
	await task.stop()
	_logger.info("### Thanks for now.")
if __name__ == "__main__":
	if len(sys.argv) != 2:
		print("Usage: python .\serial_modbus_simulator.py <device_to_simulate>")
		sys.exit(1)
	
	device_to_simulate = sys.argv[1]
	asyncio.run(run_simulator(device_to_simulate), debug=True)
```
Snippet of one of my sensor
```python
# coding=utf-8
import logging
from typing import Tuple
import struct

from pymodbus.client import ModbusSerialClient
from pymodbus.exceptions import ModbusException
from pymodbus.exceptions import ModbusIOException
import pymodbus
pymodbus.pymodbus_apply_logging_config("DEBUG")
# Attempt to import config
try:
    # When run as part of the package
    from . import config as c
except ImportError:
    # When run as a standalone script
    import config as c
class MarsSoilingSensor:
	"""
	Class that represents a Mars Soiling sensor attached via Modbus RTU
	"""
	def __init__(self, tty: str):
		self.logger = logging.getLogger(self.__class__.__name__)
		self.tty = tty
		self.serial = ModbusSerialClient(
			port=self.tty,
			method=c.MODE,
			baudrate=c.BAUD_RATE,
			stopbits=c.STOP_BITS,
			bytesize=c.BYTE_SIZE,
			timeout=c.SERIAL_TIMEOUT,
			parity=c.PARITY
		)
		self._transmission_loss = 0.0
		self._soiling_ratio = 0.0
		self._part_number = ""
		self._serial_number = ""
		self._software_version = 0
		self._calibration_date = (0, 0, 0, 0, 0, 0)
		self._uptime = 0
		self._status_code = 0

	def __enter__(self):
		self.serial.connect()
		return self

	def __exit__(self, exc_type, exc_val, exc_tb):
		self.serial.close()

	def read_all_values(self):
		"""Get all values from Modbus registers"""
			
		try:
			# So here if I do (30000, 13) works but (30000, 23 ) no
			response = self.serial.read_holding_registers(30000, 23)
			if response.isError():
				raise ModbusException(f"Modbus error reading serial number: {response}")
			self._serial_number = self._decode_string(response.registers)
			
		except ModbusException as e:
			self.logger.error(f"Error reading from sensor: {e}")
			raise

if __name__ == "__main__":
	logging.basicConfig(level=logging.INFO)
	with MarsSoilingSensor("COM6") as sensor:
		try:
			values = sensor.read_all_values()
			logging.info("Sensor values:")
			for key, value in values.items():
				logging.info(f"{key}: {value}")
		except Exception as e:
			logging.info(f"Error: {e}") 
```

janiversen, 2024-07-31:
Seems like a bug, I actually thought we had tests that covered it, but let me try it.

janiversen, 2024-08-01:
Before testing I noted that your addressing is incorrect! each address contains 2 bytes (address == register == 16bit).

but still the second read holding should work, starting to test.

Tamanar, 2024-08-01:
Ah ok, I thought that was one byte by address, it explains why when I print the result I get results like this 'model_name12345            '

janiversen, 2024-08-01:
Well it is a matter of reading the documentation! but byte address would not be logical since you cannot address a byte.

Tamanar, 2024-08-01:
Yes, I corrected the addressing but the error came not from here.

janiversen, 2024-08-01:
Just made an extra test case (based on the standard json):
```
    async def test_simulator_server_string(self, simulator_server, use_port):
        """Test simulator server end to end."""
        client = AsyncModbusTcpClient(NULLMODEM_HOST, port=use_port)
        assert await client.connect()
        result = await client.read_holding_registers(43, 2)
        assert result.registers[0] == int.from_bytes(bytes("St", "utf-8"), "big")
        assert result.registers[1] == int.from_bytes(bytes("r ", "utf-8"), "big")
        result = await client.read_holding_registers(43, 6)
        assert result.registers[0] == int.from_bytes(bytes("St", "utf-8"), "big")
        assert result.registers[1] == int.from_bytes(bytes("r ", "utf-8"), "big")
        assert result.registers[2] == int.from_bytes(bytes("St", "utf-8"), "big")
        assert result.registers[3] == int.from_bytes(bytes("rx", "utf-8"), "big")
        assert result.registers[4] == int.from_bytes(bytes("yz", "utf-8"), "big")
        assert result.registers[5] == int.from_bytes(bytes("12", "utf-8"), "big")
        result = await client.read_holding_registers(21, 23)
        assert len(result.registers) == 23
        client.close()
```

As you can see it is not a problem reading across 2 string definitions, neither is it a problem to read 23 registers.

I suspect that you have defined the 'hr' parameter too small.

Tamanar, 2024-08-01:
Here's the setup part :
```json
"setup": {
		    "co size": 0,
		    "di size": 0,
		    "hr size": 65536,
		    "ir size": 65536,
		    "shared blocks": true,
		    "type exception": true,
```
It's big enough no?

janiversen, 2024-08-01:
that might actually be too big, but is seems big enough.

But as you can see from the test case, it seems to be a problem in your configuration.

janiversen, 2024-08-01:
One way to isolate the problem is to test with growing count until it breaks, and then look if there are something special with that address.

Tamanar, 2024-08-01:
It break when I move from 
```python
response = self.serial.read_holding_registers(30000, 15)
```
to

```python
response = self.serial.read_holding_registers(30000, 16)
```
With this configuration :
```json
{
    "server_list": {
        "server_serial": {
            "comm": "serial",
            "port": "COM5",
            "stopbits": 1,
            "bytesize": 8,
            "parity": "N",
            "baudrate": 9600,
            "timeout": 1,
            "framer": "rtu",
            "identity": {
                "VendorName": "example",
                "ProductCode": "example",
                "VendorUrl": "https://example.com",
                "ProductName": "exampleSensor",
                "ModelName": "example",
                "MajorMinorRevision": "1.0"
            }
        }
    },
"sun2000_3ktl_device": {
		"setup": {
		    "co size": 0,
		    "di size": 0,
		    "hr size": 65535,
		    "ir size": 65535,
		    "shared blocks": true,
		    "type exception": true,
		    "defaults": {
		        "value": {
		            "bits": 0,
		            "uint16": 0,
		            "uint32": 0,
		            "float32": 0.0,
		            "string": ""
		        },
		        "action": {
		            "bits": null,
		            "uint16": null,
		            "uint32": null,
		            "float32": null,
		            "string": null
		        }
		    }
		},
		"invalid": [],
		"write": [],
		"bits": [],
		"uint16": [
		    {"addr": [47000, 47000], "value": 0},
		    {"addr": [37000, 37000], "value": 2},
		    {"addr": [43006, 43006], "value": 5},
		    {"addr": [47089, 47089], "value": 6}
		],
		"uint32": [],
		"float32": [],
		"string": [
		    {"addr": [30000, 30014], "value": "model_name12345678910111213141"},
		    {"addr": [30015, 30024], "value": "sn012345678910111"},
		    {"addr": [30025, 30034], "value": "PN01234567"},
		    {"addr": [30035, 30049], "value": "FW_version"},
		    {"addr": [30050, 30064], "value": "SW_version"}
		],
		"repeat": []
	}
    }
}
```
(I haven't corrected all the addressing problems for the string yet)

janiversen, 2024-08-01:
seems there are a problem, between the 2 strings, why not define them as one.

janiversen, 2024-08-01:
you have type_exception true so of course you cannot combine 2 strings !!

It really helps reading the documentation.

PR #2277 (2024-08-01): Proof for issue 2273.

fixes #2273
closed 2024-08-01 by janiversen