pymodbus-dev/pymodbus #2283: ModbusTcpClient constructor BC break between 3.6.9 and 3.7.0

rtek, 2024-08-06:
The ModbusTcpClient had a signature change between 3.6 and 3.7 for the port parameter.

see
https://github.com/pymodbus-dev/pymodbus/blob/v3.6.9/pymodbus/client/tcp.py#L61
```
 def __init__(
        self,
        host: str,
        port: int = 502,
        framer: Framer = Framer.SOCKET,
        source_address: tuple[str, int] | None = None,
        **kwargs: Any,
    ) -> None:
```
vs
https://github.com/pymodbus-dev/pymodbus/blob/v3.7.0/pymodbus/client/tcp.py#L61
```
 def __init__(  # pylint: disable=too-many-arguments
        self,
        host: str,
        framer: FramerType = FramerType.SOCKET,
        port: int = 502,
        name: str = "comm",
        source_address: tuple[str, int] | None = None,
        reconnect_delay: float = 0.1,
        reconnect_delay_max: float = 300,
        timeout: float = 3,
        retries: int = 3,
        on_connect_callback: Callable[[bool], None] | None = None,
    ) -> None:
```

This results in the following error if using positional arguments for port.
```
Traceback (most recent call last):
  File "main.py", line 13, in <module>
    client = ModbusTcpClient('localhost', 5502)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "../pymodbus/client/tcp.py", line 158, in __init__
    super().__init__(framer, retries)
  File "../pymodbus/client/base.py", line 190, in __init__
    self.framer: ModbusFramer = FRAMER_NAME_TO_CLASS.get(
                                ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'str' object is not callable
```

Reverting to 3.6.9 or using kwargs resolves the issue.

janiversen, 2024-08-07:
Yes we even documented it, see API_changes.rst

This is not a bug.

closed 2024-08-07 by janiversen

rtek, 2024-08-07:
Hello - which item referenced this? 

https://github.com/pymodbus-dev/pymodbus/blob/dev/API_changes.rst#api-changes-370

Thats the first place and I couldnt find it.

janiversen, 2024-08-07:
Ahh it seems API changes was cut short sorry about that. I will update it.

janiversen, 2024-08-07:
Not my day, all the api changes are correctly listed, the individual commits are listed in changeling, you are looking for 

* Remove kwargs. (#2236, #2237)

janiversen, 2024-08-07:
We strongly advise against using positional parameters to set non-positional parameters!

The port= have not changed signature, you are just not using it as the signature defines it.

lafrech, 2024-08-26:
Got caught by this as well. This could be considered a breaking change needing a major release but I won't complain. It only took me a few minutes to find this issue and fix my code.

> We strongly advise against using positional parameters to set non-positional parameters!

If I may suggest, those kwargs could be made keyword-only in the next pymodbus major release (v4).

```py
 def __init__(  # pylint: disable=too-many-arguments
        self,
        host: str,
        *,
        framer: FramerType = FramerType.SOCKET,
        port: int = 502,
        name: str = "comm",
        source_address: tuple[str, int] | None = None,
        reconnect_delay: float = 0.1,
        reconnect_delay_max: float = 300,
        timeout: float = 3,
        retries: int = 3,
        on_connect_callback: Callable[[bool], None] | None = None,
    ) -> None:
```

Up to you where to put the limit. Can host be positional?

At this point, you may even consider doing this right away since code using positional parameters is broken already.

janiversen, 2024-08-26:
First of all, there are no change of signature, port is an optional parameter NOT a positional parameter! 

Secondly we have a definition that API changes are only allowed in x.y.0 releases, NOT in x.y.z releases, so going to 3.7.0 means API changes, which are documented in API_changes.rst.

janiversen, 2024-08-26:
the positional parameters are NOT broken, again the problem is that you do NOT follow the signature in 3.7 nor in earlier versions.

The signature says "port = 502" meaning it is an optional parameter without a fixed position.

We do not use kmarts ! that was removed in 3.7 especially to make the signatures clearly readable.

lafrech, 2024-08-26:
> we have a definition that API changes are only allowed in x.y.0 releases

I assumed that would be only in x.0.0, because this is typical semver. But each project is free to decide otherwise. My fault for not reading the docs.

Anyway, I mainly stopped here for the friendly kw-only suggestion, not to rant about the change.

> port is an optional parameter NOT a positional parameter!

Then making it [keyword-only](https://docs.python.org/3/glossary.html#term-parameter) as I suggest above is the best way to express and enforce it IMHO.

> The signature says "port = 502" meaning it is an optional parameter without a fixed position.

Well, I don't think that is the admitted convention. By default in Python, keyword parameters can be supplied positionally. Hence the [kw-only PEP](https://peps.python.org/pep-3102/).

> We do not use kwargs ! that was removed in 3.7 especially to make the signatures clearly readable.

I'm not asking for 

```py
 def __init__(self, host: str, **kwargs):
    ...
```

What I'm talking about is this:

```py
 def __init__(  # pylint: disable=too-many-arguments
        self,
        host: str,
        *,  # <-- args after the star can only be supplied by name, otherwise an exception is raised
        framer: FramerType = FramerType.SOCKET,
        port: int = 502,
        name: str = "comm",
        source_address: tuple[str, int] | None = None,
        reconnect_delay: float = 0.1,
        reconnect_delay_max: float = 300,
        timeout: float = 3,
        retries: int = 3,
        on_connect_callback: Callable[[bool], None] | None = None,
    ) -> None:
```

Doing this prevents user from supplying args positionally, therefore from being caught by a signature change in the future.

Anyway, just wanted to make my earlier message clearer. I don't mean to argue about this. Code fixed on my side and everything works a treat. Thanks for pymodbus.

janiversen, 2024-08-26:
Feature added to the 3.8 list, just have to check if that works across all supported python versions.

thanks for the pointer.

lafrech, 2024-08-26:
No pb.

This was introduced much earlier than 3.9. I can't remember but from a quick search it may well be Python 3.0.

ekscmorente, 2024-11-11:
> Got caught by this as well. This could be considered a breaking change needing a major release but I won't complain. It only took me a few minutes to find this issue and fix my code.

Same here. I totally believe this should have been considered as a breaking change (and therefore a new major version).

+1 here to include keyword-only arguments as explicit keyword arguments using the star definition, otherwise position of the keyword arguments should be considered👍🏻

janiversen, 2024-11-19:
Or read the README, where the release cycle is explained!

3.6 to 3.7 have API changes (as documented in the readme), all change are documented in API_changes.rst.

3.7.x do not have API changes.

so it is just a question of reading the documentation.

janiversen, 2024-11-19:
If you want a warning, then feel free to submit a pull request, this is open source where you are expected to help.

rtek, 2025-01-27:
This happened again in 3.7.4 when the entire ModbusResponse class was removed.

To anyone who finds this - lock your versions explicitly since the version numbers are not semantic.

janiversen, 2025-01-30:
If you bothered read the README.rst you would understand the version scheme used in this project.

between 3.6.x and 3.7.x your app need to be slightly adapted (no major changes). All e.g 3.6.x version are without any API changes.

rtek, 2025-01-30:
Yup just warning others. 

Good luck.

janiversen, 2025-01-30:
If you say so, but please check the inheritance tree. V3.7.4 is plugin compatible with v3.7.0-3, of course unless you refer to internal parts of pymodbus.

Anyhow if you do not like the library do not use it, alternatively particate and submit pull requests and e,g, help by commenting.

Solely being negative

rtek, 2025-01-30:
Here are the signatures between the two versions.

3.7.0
```
class ModbusTcpClient(ModbusBaseSyncClient):
class ModbusBaseSyncClient(ModbusClientMixin[ModbusResponse]):
def read_holding_registers(self, address: int, count: int = 1, slave: int = 1) -> T:
```

3.7.4
```
class ModbusTcpClient(ModbusBaseSyncClient):
class ModbusBaseClient(ModbusClientMixin[Awaitable[ModbusPDU]]):
class ModbusClientMixin(Generic[T]):
def read_holding_registers(self, address: int, count: int = 1, slave: int = 1, no_response_expected: bool = False) -> T
```

My requirements.txt contained the following, based on your notes about patch version changes from the summer.

`pymodbus >=3.7, <4.0`

Heres my sample code that broke when someone else ran it against 3.7.4.

```
from pymodbus.client import ModbusTcpClient
from pymodbus.pdu import ModbusResponse

def decodeResponse(response: ModbusResponse)
   something(response.registers)

client = ModbusTcpClient(....)
client.connect()
response =client.read_holding_registers(...)

decoded= decodeResponse(response)
```

My feedback is:

If you want flexibility in your API to use duck typing, don't have a return type. We have no problem duck typing when consuming API - we just guard around thing we're looking for - `response.registers` in this case. We used the return type as the guard so when it was removed it broke the code.

janiversen, 2025-01-30:
As I wrote earlier, do not just complain ! if you the library to be better participate. People just complaining tend to be ignored.