pymodbus-dev/pymodbus #2293: Module 'serial' has no attribute 'serial_for_url' when using ModbusSerialClient

TowardtheStars, 2024-08-15:
### Versions

- Python: 3.9.6
- OS: Windows 10
- Pymodbus: 3.7.0
- Modbus Hardware (if used): A custom hardware, but that should not be the problem.

### Pymodbus Specific

- Client: Serial, sync

### Description
Prompt gives `Module 'serial' has no attribute 'serial_for_url'` when using `ModbusSerialClient::connect`

### Code and Logs

```python
>>> import pymodbus
>>>
>>> pymodbus.pymodbus_apply_logging_config("DEBUG")
>>> from pymodbus.client import ModbusTcpClient, ModbusSerialClient
>>> client = ModbusSerialClient('COM6', baudrate=9600)
>>> client.connect()
2024-08-15 11:07:41,014 ERROR logging:115 module 'serial' has no attribute 'serial_for_url'
False
>>>
```

TowardtheStars, 2024-08-15:
It seems like I accidentally installed package `serial`, and the interpreter uses `serial` package instead of `pyserial` package when importing `serial` module. 

To fix this, simply uninstall the `serial` package and reinstall `pyserial`.

Hopefully this can be helpful to someone encounters the same problem.

closed 2024-08-15 by TowardtheStars

janiversen, 2024-08-15:
Actually if it is called "read the documentation" if you installed pymodbus as suggested you would not have had this problem.