pymodbus-dev/pymodbus #2304: ImportError: cannot import name 'ModbusTcpClient' from 'pymodbus.client'

jsanchez73, 2024-09-02:
Hello,

I have installed pymodbus on a Windows 11 Pro machine and it works fine, but when installing it and running it on a raspberry pi I get the following error when I try to import ModbusTcpClient:

ImportError: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/home/pi/.local/lib/python3.7/site-packages/pymodbus/client/__init__.py)

I am using a Raspberry Pi 3 Model B with Raspbian GNU/Linux 10 (buster) and using a Waveshare 8-ch Ethernet Relay Module (Modbus POE ETH Relay). I have installed pymodbus with pip install and have noticed that the __init__.py file in the client directory of the installation is empty. As shown above, I am running my script on python3.7.
Thanks in advance for any help to solve this issue. Best Regards,

Jose

janiversen, 2024-09-02:
First of all, we no longer support python 3.7, however I suspect it still works, at least the error have nothing to do with python version.

You do not write which version of pymodbus you are using, but I guess your problem is that the documentation you read do not correspond to the version you have installed.

We have had "ModbusTcpClient" in pymodbus.client.__init__.py for a very long time...however raspbian is known to be years behind with versions.

Please check the version of pymodbus, current version is 3.7.2.

janiversen, 2024-09-02:
you can fix the version by doing "pip install pymodbus==3.7.2"

jsanchez73, 2024-09-02:
Thanks a lot Jan!!! The version installed by default is 2.5.3 . I´ll try to
force to 3.7.2 and let you know

jsanchez73, 2024-09-02:
I tried to force 3.7.2 installation but go the following error message:

 Could not find a version that satisfies the requirement pymodbus==3.7.2 (from versions: 0.9.0, 1.0.0, 1.1.0, 1.2.0, 1.3.0rc1, 1.3.0rc2, 1.3.1, 1.3.2, 1.4.0rc3, 1.4.0, 1.5.0, 1.5.1, 1.5.2, 2.0.0, 2.0.1, 2.1.0, 2.2.0rc1, 2.2.0rc2, 2.2.0rc3, 2.2.0rc4, 2.2.0, 2.3.0rc1, 2.3.0, 2.4.0rc1, 2.4.0, 2.5.0rc1, 2.5.0rc2, 2.5.0, 2.5.1rc1, 2.5.1, 2.5.2, 2.5.3rc1, 2.5.3)
No matching distribution found for pymodbus==3.7.2

janiversen, 2024-09-02:
That is because raspian have not yet upgraded pymodbus to 3.7.2 (Version 2.5.3 died about 3 years ago).

You need to tell pip to look in the official pypi distribuition.

Closing as this is not a pymodbus problem.

closed 2024-09-02 by janiversen