pymodbus-dev/pymodbus #2306: Losing packets in modbus rtu

Gradhoc-aeimil, 2024-09-05:
### Versions

- Python: 3.11.2
- OS:Raspbian
- Pymodbus: 3.7.2

### Pymodbus Specific

- Server: rtu

### Description

Hi, I', trying to set a server modbus rtu, i can set a conection but when a try to connect a modbus client to it, it lost most of the packages. We are using modbus converters of automatic flow control. The exact same server on windows it works. I post the code and the logs below.

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
![imagen (3)](https://github.com/user-attachments/assets/7362b68c-bdac-4a99-9397-007680f5747e)

pymodbus.pymodbus_apply_logging_config("DEBUG")
![imagen (3)](https://github.com/user-attachments/assets/29db87b2-187a-477f-a2fd-2aecd6595420)

janiversen, 2024-09-05:
You have posted a picture not the code nor a debug log. please post the code directly as well as a debug log.

You write the problem is with rtu, but you also talk about an i0 address which is tcp/ip, so excuse me that I am confused.

Gradhoc-aeimil, 2024-09-05:
Sorry, i ddint express myself right. I'm trying to raise a modbus rtu server on a raspberry, and I can do it. But, when I try to access to that server through Modscan on Windows, I lost most of the packages. I'm using automatic flow control converters. The strange thing is that, if I tried to connect to that server through a client on linux I can do it perfectly. So my question is, is ther any problem accessing one server setted up on Linux through a software of Windows.

The code is next one:
[image]

janiversen, 2024-09-05:
I cannot talk about how modscan works, did you try with the pymodbus client?

You need to add a debug log as explained when you opened the issue, if you believe the pymodbus server have a problem.

In general a lot of things can be wrong, the usb you are using, the rs485 cabling (e.g. not terminated correctly), but it's all problems outside pymodbus.

Since the server works with a linux client, it does not seem to be a pymodbus problem.

janiversen, 2024-09-05:
Pymodbus server do not care about which OS the client uses!

Closing issue as it does not seem to be a pymodbus problem.

closed 2024-09-05 by janiversen

gradhocJcalvo, 2024-09-06:
I'm having same issue with his code. If you use a pymodbus client on the same OS the system is stable, but when you go with your pymodbus server on Windows and your pymodbus client on Linux (or reverse) it lost a lot of packages. I'm using automatic flow direction [converter](https://www.amazon.es/DIGITUS-adaptador-Conversor-Chipset-FT232RL/dp/B007VZY4CW/ref=asc_df_B007VZY4CW/?tag=googshopes-21&linkCode=df0&hvadid=699690028215&hvpos=&hvnetw=g&hvrand=6719471590987304720&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=20272&hvtargid=pla-563872658051&psc=1&mcid=c85908776ab73a5aad88fa6222df8bad&gad_source=1).
The same client with the same code on the same OS as server works fine. I already try gateway rtu to tcp converter example with the same result. I really can not face what is going on, it look like a configuration though OS, but don't know what can be.
Maybe do you have some idea what is going on?

Thank you very much.

janiversen, 2024-09-06:
I have no idea if pymodbus server on windows have a problem, we have about 50 test cases of the server on windows, which run without problems. pymodbus client on windows have been reported to work.

But please do not just claim you loose a lot of packages, add a debug as pr issue template (see above), so we can see what happens.

Preferable with a pymodbus client, so we have 2 complete logs.

I cannot help further without debug logs.