pymodbus-dev/pymodbus #2308: Type Hint for write_registers value parameter wrong?

rauberdaniel, 2024-09-07:
`BinaryPayloadBuilder.build(…)` returns `list[bytes]`
```python
def build(self) -> list[bytes]:
```
but `ModbusTcpClient.write_registers(…)` expects `list[int]` as values argument
```python
def write_registers(self, address: int, values: list[int], slave: int = 1, skip_encode: bool = False) -> T:
``` 

According to my tests, write_registers however actually expects `list[bytes]` so I think this type hint there is wrong? Can anybody confirm or deny that?

janiversen, 2024-09-07:
I will take a look.

janiversen, 2024-09-07:
You are right it should be list[bytes], while write_register expects an int.

You are welcome to make a pull request if you want, otherwise I will correct it when I get around to it,
.

PR #2309 (2024-09-07): Fix type hints for write_register and write_registers
Fixes for type hints on `write_register` and `write_registers` which actually expect `bytes` / `list[bytes]` instead of `int` / `list[int]`

Closes #2308
closed 2024-09-07 by janiversen