pymodbus-dev/pymodbus #2320: ModbusTcpClient doesn't do reconnects?

camtarn, 2024-09-14:
### Versions

- Python: N/A
- OS: N/A
- Pymodbus: 3.7.2
- Modbus Hardware (if used): N/A

### Pymodbus Specific

- Client: tcp - sync

### Description

On:
https://github.com/pymodbus-dev/pymodbus/blob/dcbcfd7f16fb3be82525bccf9699855a06f900c5/pymodbus/client/tcp.py#L129C1-L129C74

there is the comment:
`    Remark: There are no automatic reconnect as with AsyncModbusTcpClient`

Why is this?

This also suggests that this part of the docstring for ModbusTcpClient might be incorrect, if there is no automatic reconnection:

```    .. tip::
        **reconnect_delay** doubles automatically with each unsuccessful connect, from
        **reconnect_delay** to **reconnect_delay_max**.
        Set `reconnect_delay=0` to avoid automatic reconnection.
```

Not a problem if it's not supported, I was just surprised to find this.

janiversen, 2024-09-14:
the async client do automatic reconnect, wh6 do you think there are no automatic reconnect?

If you think there is a bug, then please add a debug log so we can see what happens.

janiversen, 2024-09-14:
Of course it seems to be a wrong documentation, 8n that case a pull request is welcome.

camtarn, 2024-09-14:
This is specifically for the sync client.

Happy to make a pull request if you confirm the sync client also does reconnects.

janiversen, 2024-09-14:
The sync client do not do automatic reconnect, as far as I remember.

janiversen, 2024-09-14:
sync uses a different approach, it connects if a request is sent and it's not connected.

camtarn, 2024-09-14:
Ah, yes, I think I see - https://github.com/pymodbus-dev/pymodbus/blob/aa7e556cb53c4270c4e0b74b771ba430332c62e8/pymodbus/client/base.py#L235 causes the client to try connecting, which returns immediately if it's already connected, or kicks off a new connection.

I'll see if there's a better way to word the documentation.

camtarn, 2024-09-14:
I've found a different way to phrase the docs.

However, it seems like it would make sense to remove the reconnect_delay/reconnect_delay_max params from the sync client, since it doesn't use them. I'm guessing that's an API change? For the moment I can flag them as 'not used in the sync client'.

PR #2321 (2024-09-24): Improve docs around sync clients and reconnection
Sync clients don't do reconnection in the same way as async clients, and should have a little more explanation as to what they do.

fixes #2320

Verified that docs successfully build and look right.
closed 2024-09-24 by janiversen