pymodbus-dev/pymodbus #2368: struct.error: required argument is not an integer

MAKOMO, 2024-10-10:
### Versions

- Python: 3.12
- OS: macOS 15
- Pymodbus: 3.7.3
- Modbus Hardware (if used):

### Description

The type of the Client methods `write_register` (and `write_registers`) was changed in `client/mixin.py`. Now the value to be sent should be of type `bytes` instead of `int` as in the previous releases.

Converting the int `value` to bytes using `value.to_bytes(2, 'big')` makes mypy happy, but results in

```
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pymodbus/client/base.py", line 104, in async_execute
    packet = self.ctx.framer.buildPacket(request)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pymodbus/framer/base.py", line 85, in buildPacket
    data = message.function_code.to_bytes(1,'big') + message.encode()
                                                     ^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/pymodbus/pdu/register_write_message.py", line 42, in encode
    packet += struct.pack(">H", self.value)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: required argument is not an integer
```

Sending the int value as before works, but makes mypy sad.

What to do?

janiversen, 2024-10-10:
You should be able to send bytes or int, seems I did a bad review on #2309.

Fix will be available in a moment.

closed 2024-10-10 by janiversen

MAKOMO, 2024-10-10:
Thanks for your quick reaction. Your fix looks good to me.

MAKOMO, 2024-10-24:
Thanks for providing v3.7.4 of your great lib!

However, I am not completely happy with the new type annotation of the values in mixin:write_registers which is now `list[bytes | int]`.

As [`list` is invariant](https://github.com/python/mypy/issues/3351) it is now not type correct to provide this function with a homogeneous list of either bytes or int.

For example on a `test.py` with

```
from typing import Sequence

def f(l:list[int | bytes]) -> None:
    return
def g(l:list[int] | list[bytes]) -> None:
    return
def h(l:Sequence[int | bytes]) -> None:
    return

l:list[int] = [1,2,3]

f(l) # type error
g(l) # type correct
h(l) # type correct
```

mypy reports

```
# mypy test.py
test.py:13: error: Argument 1 to "f" has incompatible type "list[int]"; expected "list[int | bytes]"  [arg-type]
test.py:13: note: "List" is invariant -- see https://mypy.readthedocs.io/en/stable/common_issues.html#variance
test.py:13: note: Consider using "Sequence" instead, which is covariant
Found 1 error in 1 file (checked 1 source file)
```

The simplest solution to this would be to restrict the use to homogenous lists of either type using `list[int]|list[bytes]` or using `Sequence[int | bytes]` as suggested by mypy.

janiversen, 2024-10-24:
I can live with both solutions, at the end of the day, the call accept a list of int or bytes. It also accepts a list of mixed bytes/int but that is more an accident.

Looking forward to see a PR.

PR #2369 (2024-10-10): write_register/s accept bytes or int

fixes #2368