pymodbus-dev/pymodbus #2371: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response

joseph112, 2024-10-11:
The instruction I sent was that RTU should respond with 8 bits, but the program calculates it as 7 bits.
The correct input and response is as follows
send: 01 03 FA 0C 00 01 74 D1
rec: 01 03 00 02 23 2C FC E7
Here's the translation:

I modified `self.base_adu_size` to be `address(2)`, making `self.base_adu_size = 4`. The response length indeed became 8 bits. However, `self.transactions` remains empty or throws the same error. What should I do?

joseph112, 2024-10-11:
My test code is as follows
```
from pymodbus.client import ModbusSerialClient

client = ModbusSerialClient(port="COM1", baudrate=38400)
if client.connect():
    print("The connection was successful.")
else:
    print("Failed to connect")
result = client.read_holding_registers(address=64012, count=1, slave=1)
if result.isError:
    print(result)
else:
    print(result.registers)
client.close()
```

janiversen, 2024-10-11:
You are reading a holding register, which is a register not bits. A register is defined by the standard to be 2bytes.

for bits use a bit function like e.g. read_coils, that will deliver the bits requested rounded up to bytes as pr standard.

janiversen, 2024-10-11:
And just for information:
```
send: 01 03 FA 0C 00 01 74 D1
rec: 01 03 00 02 23 2C FC E7
```

The send asks for 1 registers (00 01)
and your device responds with length 2 bytes (00 02) == 1 register so you get 1 register back with the value (23 2c)

that is all correct, but has nothing to do with bits.

joseph112, 2024-10-12:
Regarding the instruction length, it includes both the high and low bits separately. Another issue is why self.transactions is empty, leading to "No Response received from the remote slave / Unable to decode response".

janiversen, 2024-10-12:
I have no idea what you talk about! your modifications are wrong, mainly because because you use a call that always will return a multiple of 16 bits.

Stay assured the function call you use, work as it should do...and as you saw in my analysis you get the correct answer corresponding to the request you make.

janiversen, 2024-10-12:
If you doubt that read_holding_register should always return a multiple of 16bits == number of register x 2, then please read the modbus standard.

As I said earlier if you want bits the use a bit call....this is also described in the standard.

joseph112, 2024-10-13:
I'm sorry I didn't get it clear. I mean why using the pymodbus library it only returns an error, whereas I use other tools it is correct.
[image]

janiversen, 2024-10-13:
What I tried to say is that pymodbus works without your modifications it will return the same as the other tool.....which also returns 16bit.

Don't blame pymodbus blame your wrong changes.

joseph112, 2024-10-13:
I did not modify it, using it directly will return an error. It is correct with other tools. Other tools are shown in the screenshot above.

janiversen, 2024-10-13:
hmm your text says "I modified self.base_adu_size to be address(2), making self.base_adu_size = 4. The response length indeed became 8 bits. However, self.transactions remains empty or throws the same error. What should I do?"

You also wrote "The instruction I sent was that RTU should respond with 8 bits" but read_holding_register respond with a multiple of 16 bits.

janiversen, 2024-10-13:
If you want help, you need to add a debug log, with a not modified pymodbus !! so we can see what your problem might be !

With the debug log we can see exactly what is being sent and what is being received.

joseph112, 2024-10-14:
Sorry, I tried modifying the base_adu_size, but it always returned the same error, so I reset it.

janiversen, 2024-10-14:
I think you are not talking about BITS but BYTES (a BYTE is 8 BITS), that at least explains a lot.

Yes pymodbus do not receive the last BYTE, probably because your usb does not deliver it im time.

Which version of pymodbus are you using ? this should be solved in v3.7.3

janiversen, 2024-10-14:
When pymodbus does not receive a full frame, it will try to read again….that does not happen in your example, so I think you are using an old version.

janiversen, 2024-10-14:
Try to set timeout=5 that will make the receiver wait for 5 seconds before retrying.

joseph112, 2024-10-14:
I apologize, my description contained an error; indeed, I was referring to bytes. The version of pymodbus is: 3.7.3

joseph112, 2024-10-14:
This is my test code, the error still exists.
`from pymodbus.client import ModbusSerialClient

client = ModbusSerialClient(port="COM1", baudrate=38400, framer=FramerType.RTU, timeout=5)
if client.connect():
    print("The connection was successful.")
else:
    print("Failed to connect")
result = client.read_holding_registers(address=64012, count=1, slave=1)
if result.isError:
    print(result)
else:
    print(result.registers)
client.close()`

janiversen, 2024-10-14:
please add retries=10, I just now saw that you are using the sync. version, which have a different implementation (old, and more limited than the async version).

AND please add the debug log, so I can see the effect.

joseph112, 2024-10-14:
I tried retries=10 but it still doesn't work. I'm not quite sure what you mean, are you asking me to try the old version or the asynchronous version?

janiversen, 2024-10-14:
The sync version is in maintenance mode, pending a major rewrite, so a number of the features is not available.

the async version uses both retries= and timeout=

I am not sure what you mean by "old version", I did not talk about that.

I am preparing a bug fix to the sync version, that implements timeout= for receive.

joseph112, 2024-10-14:
Thank you very much, I understand.

janiversen, 2024-10-14:
Now I did a bit more analyzing and testing....you claim the correct response is:
```
01 03 00 02 23 2C FC E7
```
First byte is slave_id and the 2 last bytes are CRC, so the PDU is:
```
03 00 02 23 2C
```
The modbus standard defined function code 0x03 as:
[image]

So decoding the PDU:
03 = Function code
00 = 2 x N, where N is number of registers
02 23 2c = Data which is should be a multiple of 2 (since a register is 2 bytes).

As you can see your device responds with a wrong byte count, and that is the real problem, why you do not receive anything.

janiversen, 2024-10-14:
There are however a bug in the sync code, which I still have to fix.

joseph112, 2024-10-14:
I used this code to test the asynchronous client, but the error still persists.
https://github.com/pymodbus-dev/pymodbus/blob/dev/examples/simple_async_client.py

janiversen, 2024-10-14:
Yes because the message your device sends (count = 0) is not correct.

joseph112, 2024-10-14:
02 is the data number, 23 2C is the returned data.

joseph112, 2024-10-14:
My device's error frame returns 01 03 80 01 00 04 3C 09

joseph112, 2024-10-14:
I feel that the data frame returned by my device, the data length should occupy two bytes.

janiversen, 2024-10-14:
Look at the standard, I sent you !!! data length is 1 byte NOT 2.

closed 2024-10-14 by janiversen

joseph112, 2024-10-15:
I consulted with the device manufacturer, and they informed me that Modbus is not a standard; data length spans 2 bytes. Therefore, the data length for 00 02 bits.

janiversen, 2024-10-15:
Modbus is a standard, a very official standard, but it seems your device is not following the modbus standard.

Pymodbus support the modbus standard, and not vendor variations, so please use another library.

PR #2377 (2024-10-14): Improve retries for sync client.

fixes #2371