pymodbus-dev/pymodbus #2373: Modbus TCP crashed reading a PAC3220 after several hours

JuanPaEsAr, 2024-10-11:
### Versions

- Python: 3.9.2
- OS: Linux debian 11
- Pymodbus: 3.6.9
- Modbus Hardware (if used): TCP

### Pymodbus Specific

- Server: tcp - sync/async
- Client: tcp - sync/async

### Description

Hi guys, in this moment, I am reading a tcp device, specically, PAC 32220 of brand SIEMENS, I am reading this device very quickly, with a delta time equal to 0.2 seconds such as shown in the code below, the problem is after the six hours or more the code crash and shown the result below:

### Code and Logs

```python
2024-10-10 14:57:19,382 WARNING logging:109 Failed to connect
2024-10-10 14:57:19,382 WARNING logging:109 Failed to connect
2024-10-10 14:57:19,382 WARNING logging:109 Failed to connect
2024-10-10 14:57:19,382 WARNING logging:109 Failed to connect
2024-10-10 14:57:19,382 WARNING logging:109 Failed to connect
2024-10-10 14:57:19,382 WARNING logging:109 Failed to connect
Executing <Task finished name='Task-406397' coro=<run_async_simple_client() done, defined at /home/erco_to/FW2/modbus_tcp.py:6> exception=AssertionError() created at /usr/lib/python3.9/asyncio/base_events.py:621> took 6.756 seconds
Traceback (most recent call last):
  File "/home/erco_to/FW2/modbus_tcp.py", line 71, in <module>
    registers_read = asyncio.run(
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/erco_to/FW2/modbus_tcp.py", line 32, in run_async_simple_client
    assert client.connected
AssertionError

import asyncio
import pymodbus.client, pymodbus.framer, pymodbus.exceptions, pymodbus.pdu
import time
from pymodbus import Framer

async def run_async_simple_client(
    host,
    port,
    function_code,
    address_register,
    amount_registers,
    slave,
    framer=pymodbus.framer.ModbusSocketFramer,
):
    """Run async client."""
    # activate debugging
    pymodbus.pymodbus_apply_logging_config("INFO")

    print("Obteniendo cliente")

    client = pymodbus.client.AsyncModbusTcpClient(
        host=host,
        port=port,
        framer=framer,
        timeout=10,
    )

    print("Conectando al servidor")
    await client.connect()
    # test client is connected
    assert client.connected

    print("Conseguiendo datos y verificando")
    try:
        if function_code == 3:
            rr = await client.read_holding_registers(
                address=address_register, count=amount_registers, slave=slave
            )
        else:
            rr = await client.read_input_registers(
                address=address_register, count=amount_registers, slave=slave
            )
    except pymodbus.exceptions.ModbusException as exc:
        print(f"Recibiendo una excepción Modbus desde la libreria({exc})")
        client.close()
        return
    if rr.isError():
        print(f"Recibiendo un error Modbus de la libreria({rr})")
        client.close()
        return
    if isinstance(rr, pymodbus.pdu.ExceptionResponse):
        print(f"Recibiendo una excepción de la libreria Modbus({rr})")
        # THIS IS NOT A PYTHON EXCEPTION, but a valid modbus message
        client.close()

    print("Cerrando conexion")
    client.close()
    return rr.registers

if __name__ == "__main__":
    host = "10.6.20.77"
    port = 502
    function_code = 3
    address_regsiter = 1
    amount_registers = 70
    slave = 1

    while True:
        registers_read = asyncio.run(
            run_async_simple_client(
                host=host,
                port=port,
                function_code=function_code,
                address_register=address_regsiter,
                amount_registers=amount_registers,
                slave=slave,
            ),
            debug=True,
        )
        print(registers_read)

        time.sleep(0.2)

```

janiversen, 2024-10-11:
Seems your connection broke down, but I cannot tell you why. The async client automatically tries to reconnect, as you can see from the failed connect, but it seems your device rejects (or something in between) the new connection request. 

I have a solar inverter that I read every 100ms, and that works for at least weeks, so in general it should be working.

Please change INFO in the debug log call to DEBUG, that gives a lot more information, and post the log here.

A little tip on performance: Put the whole while loop in a async function and replace time.sleep with asyncio.sleep, that will work a lot faster, because you do not initialize asyncio every 0.2 seconds.

Thinking about it, that could be your problem, that the frequent call to asyncio.run, accumulates open sockets, so you run out of resources.

JuanPaEsAr, 2024-10-11:
I followed your suggestion, and I enable the library debug, and I was looking the logs and I am worried about the log that say: "Very short frame (NO MBAP):  wait for more data", Is this log normal or is there any issue into the communication?

````python 
Obteniendo cliente
Conectando al servidor
2024-10-11 10:46:10,291 DEBUG logging:103 Connecting to 10.6.20.77:502.
2024-10-11 10:46:10,292 DEBUG logging:103 Connecting comm
2024-10-11 10:46:10,294 DEBUG logging:103 Connected to comm
2024-10-11 10:46:10,295 DEBUG logging:103 Resetting frame - Current Frame in buffer -
Conseguiendo datos y verificando
2024-10-11 10:46:10,296 DEBUG logging:103 Adding transaction 1
2024-10-11 10:46:10,296 DEBUG logging:103 Resetting frame - Current Frame in buffer -
2024-10-11 10:46:10,297 DEBUG logging:103 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x46
2024-10-11 10:46:10,298 DEBUG logging:103 recv: 0x0 0x1 0x0 0x0 0x0 0x8f 0x1 0x3 0x8c 0x42 0xf6 0x5e 0xe4 0x42 0xf6 0x4e 0xa5 0x42 0xf6 0x2f 0x28 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3e 0x1c 0x61 0xc3 0x3e 0x1c 0x67 0x28 0x3e 0x1c 0x57 0x4f 0x41 0x96 0x7f 0xe9 0x41 0x96 0x7b 0x2d 0x41 0x96 0x58 0xb2 0x40 0xdf 0xa5 0xb6 0x40 0xdf 0x5a 0x4c 0x40 0xdf 0x1a 0xec 0xbf 0xf7 0x9 0x7 0xbf 0xf9 0x1a 0xce 0xbf 0xf9 0x31 0x94 0x3e 0xbe 0x36 0x4d 0x3e 0xbd 0xfc 0x23 0x3e 0xbd 0xf1 0xc1 0x40 0x75 0xf9 0x95 0x40 0x75 0xe7 0x27 0x40 0x76 0x4 0x16 0x43 0x71 0x2e 0x64 0x43 0x71 0x2e 0xf7 0x43 0x71 0x2f 0xa 0x42 0x6f 0xe5 0xc2 0x42 0xf6 0x49 0x90 0x0 0x0 0x0 0x0 0x3e 0x1c 0x60 0x14 0x42 0x61 0xa9 0xe4 0x41 0xa7 0x86 0xbc 0xc0 0xba 0x55 0x5a 0x3e 0xbe 0xc 0x13 old_data:  addr=None
2024-10-11 10:46:10,299 DEBUG logging:103 Processing: 0x0 0x1 0x0 0x0 0x0 0x8f 0x1 0x3 0x8c 0x42 0xf6 0x5e 0xe4 0x42 0xf6 0x4e 0xa5 0x42 0xf6 0x2f 0x28 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3e 0x1c 0x61 0xc3 0x3e 0x1c 0x67 0x28 0x3e 0x1c 0x57 0x4f 0x41 0x96 0x7f 0xe9 0x41 0x96 0x7b 0x2d 0x41 0x96 0x58 0xb2 0x40 0xdf 0xa5 0xb6 0x40 0xdf 0x5a 0x4c 0x40 0xdf 0x1a 0xec 0xbf 0xf7 0x9 0x7 0xbf 0xf9 0x1a 0xce 0xbf 0xf9 0x31 0x94 0x3e 0xbe 0x36 0x4d 0x3e 0xbd 0xfc 0x23 0x3e 0xbd 0xf1 0xc1 0x40 0x75 0xf9 0x95 0x40 0x75 0xe7 0x27 0x40 0x76 0x4 0x16 0x43 0x71 0x2e 0x64 0x43 0x71 0x2e 0xf7 0x43 0x71 0x2f 0xa 0x42 0x6f 0xe5 0xc2 0x42 0xf6 0x49 0x90 0x0 0x0 0x0 0x0 0x3e 0x1c 0x60 0x14 0x42 0x61 0xa9 0xe4 0x41 0xa7 0x86 0xbc 0xc0 0xba 0x55 0x5a 0x3e 0xbe 0xc 0x13
2024-10-11 10:46:10,300 DEBUG logging:103 Factory Response[ReadHoldingRegistersResponse': 3]
2024-10-11 10:46:10,300 DEBUG logging:103 Getting transaction 1
2024-10-11 10:46:10,301 DEBUG logging:103 Very short frame (NO MBAP):  wait for more data
Cerrando conexion
[17142, 24292, 17142, 20133, 17142, 12072, 0, 0, 0, 0, 0, 0, 15900, 25027, 15900, 26408, 15900, 22351, 16790, 32745, 16790, 31533, 16790, 22706, 16607, 42422, 16607, 23116, 16607, 6892, 49143, 2311, 49145, 6862, 49145, 12692, 16062, 13901, 16061, 64547, 16061, 61889, 16501, 63893, 16501, 59175, 16502, 1046, 17265, 11876, 17265, 12023, 17265, 12042, 17007, 58818, 17142, 18832, 0, 0, 15900, 24596, 16993, 43492, 16807, 34492, 49338, 21850, 16062, 3091]
````

janiversen, 2024-10-12:
you cut the log just as it gets interesting, I need to see more of the log.

janiversen, 2024-10-12:
A little analyzing:

```
send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x46
```
You request 0x46 registers == 70 from address 1, which is all correct. So we expect 140 bytes of data.

```
recv: 0x0 0x1 0x0 0x0 0x0 0x8f 0x1. <== MBAP with payload length 0x8F == 143
0x3 0x8c <== Read_holding response, data length 0x8c == 140
0x42 0xf6 0x5e 0xe4 0x42 0xf6 0x4e 0xa5 0x42 0xf6
0x2f 0x28 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
0x0 0x0 0x0 0x0 0x3e 0x1c 0x61 0xc3 0x3e 0x1c
0x67 0x28 0x3e 0x1c 0x57 0x4f 0x41 0x96 0x7f 0xe9
0x41 0x96 0x7b 0x2d 0x41 0x96 0x58 0xb2 0x40 0xdf
0xa5 0xb6 0x40 0xdf 0x5a 0x4c 0x40 0xdf 0x1a 0xec
0xbf 0xf7 0x9 0x7 0xbf 0xf9 0x1a 0xce 0xbf 0xf9
0x31 0x94 0x3e 0xbe 0x36 0x4d 0x3e 0xbd 0xfc 0x23
0x3e 0xbd 0xf1 0xc1 0x40 0x75 0xf9 0x95 0x40 0x75
0xe7 0x27 0x40 0x76 0x4 0x16 0x43 0x71 0x2e 0x64
0x43 0x71 0x2e 0xf7 0x43 0x71 0x2f 0xa 0x42 0x6f
0xe5 0xc2 0x42 0xf6 0x49 0x90 0x0 0x0 0x0 0x0
0x3e 0x1c 0x60 0x14 0x42 0x61 0xa9 0xe4 0x41 0xa7
0x86 0xbc 0xc0 0xba 0x55 0x5a 0x3e 0xbe 0xc 0x13
```
So the response is OK.

```
2024-10-11 10:46:10,301 DEBUG logging:103 Very short frame (NO MBAP):  wait for more data
```
This comes because pymodbus have received less than 7 bytes, but that is not something that the app is notified about.
I am puzzled about:
```
Cerrando conexion
```
That surely is not from pymodbus, but from the app.

With the log you supply I see no problems in the communication nor in pymodbus...BUT as I said I need to see the continuation of the debug log, to see what happens next
- do the device disconnect
- do pymodbus correctly try to reconnect etc.

JuanPaEsAr, 2024-10-15:
I did the modifications in the code above that you suggest me:
```python
import asyncio
import pymodbus.client, pymodbus.framer, pymodbus.exceptions, pymodbus.pdu

async def run_async_simple_client(
    host,
    port,
    function_code,
    address_register,
    amount_registers,
    slave,
    framer=pymodbus.framer.ModbusSocketFramer,
):
    """Run async client."""
    
    pymodbus.pymodbus_apply_logging_config("DEBUG")

    

    client = pymodbus.client.AsyncModbusTcpClient(
        host=host,
        port=port,
        framer=framer,
        timeout=1,
    )

    await client.connect()
    # test client is connected
    assert client.connected

    try:
        if function_code == 3:
            rr = await client.read_holding_registers(
                address=address_register, count=amount_registers, slave=slave
            )
        else:
            rr = await client.read_input_registers(
                address=address_register, count=amount_registers, slave=slave
            )
    except pymodbus.exceptions.ModbusException as exc:
        print(f"Modbus library exception({exc})")
        client.close() 
        return
    if rr.isError():
        print(f"Modbus error({rr})")
        client.close() 
        return
    if isinstance(rr, pymodbus.pdu.ExceptionResponse):
        print(f"Modbus instance exception({rr})")
        client.close() 

    client.close() 
    return rr.registers

async def read_data():
    host = "10.6.20.77"
    port = 502
    function_code = 3
    address_register = 1
    amount_registers = 70
    slave = 1

    while True:
        registers_read = await run_async_simple_client(
            host=host,
            port=port,
            function_code=function_code,
            address_register=address_register,
            amount_registers=amount_registers,
            slave=slave,
        )
        print(registers_read)

        await asyncio.sleep(10) 

if __name__ == "__main__":
    asyncio.run(read_data())

```
I run this code only one time and the complete output, that is to say, all logs in DEBUG are:
```python
2024-10-15 11:18:47,075 DEBUG logging:103 Connecting to 10.6.20.77:502.
2024-10-15 11:18:47,076 DEBUG logging:103 Connecting comm
2024-10-15 11:18:47,089 DEBUG logging:103 Connected to comm
2024-10-15 11:18:47,089 DEBUG logging:103 Resetting frame - Current Frame in buffer -
2024-10-15 11:18:47,090 DEBUG logging:103 Adding transaction 1
2024-10-15 11:18:47,091 DEBUG logging:103 Resetting frame - Current Frame in buffer -
2024-10-15 11:18:47,091 DEBUG logging:103 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x1 0x0 0x46
2024-10-15 11:18:47,093 DEBUG logging:103 recv: 0x0 0x1 0x0 0x0 0x0 0x8f 0x1 0x3 0x8c 0x42 0xfb 0xf2 0xb3 0x42 0xfb 0xe2 0xa9 0x42 0xfb 0xc2 0x3d 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3f 0x80 0x0 0x0 0x3f 0x80 0x0 0x0 0x3f 0x80 0x0 0x0 0x40 0x88 0xdf 0x10 0x40 0x88 0xca 0x68 0x40 0x88 0xb7 0x23 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x42 0x70 0x4 0x27 0x42 0xfb 0xdd 0x33 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3f 0x80 0x0 0x0 old_data:  addr=None
2024-10-15 11:18:47,094 DEBUG logging:103 Processing: 0x0 0x1 0x0 0x0 0x0 0x8f 0x1 0x3 0x8c 0x42 0xfb 0xf2 0xb3 0x42 0xfb 0xe2 0xa9 0x42 0xfb 0xc2 0x3d 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3f 0x80 0x0 0x0 0x3f 0x80 0x0 0x0 0x3f 0x80 0x0 0x0 0x40 0x88 0xdf 0x10 0x40 0x88 0xca 0x68 0x40 0x88 0xb7 0x23 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x42 0x70 0x4 0x27 0x42 0xfb 0xdd 0x33 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3f 0x80 0x0 0x0
2024-10-15 11:18:47,094 DEBUG logging:103 Factory Response[ReadHoldingRegistersResponse': 3]
2024-10-15 11:18:47,095 DEBUG logging:103 Getting transaction 1
2024-10-15 11:18:47,095 DEBUG logging:103 Very short frame (NO MBAP):  wait for more data
[17147, 62131, 17147, 58025, 17147, 49725, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 16256, 0, 16256, 0, 16256, 0, 16520, 57104, 16520, 51816, 16520, 46883, 65535, 65535, 65535, 65535, 65535, 65535, 17008, 1063, 17147, 56627, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 16256, 0]
```
Is some wrong or all is alright?

janiversen, 2024-10-15:
if you get your data, like it seems everything is ok.

closed 2024-10-16 by janiversen