pymodbus-dev/pymodbus #2398: 3.7.3 has a delay in the serial communication

SaraEersel, 2024-10-21:
### Versions

- Python: 3.11.8
- OS:Pop!_OS 22.04 LTS
- Pymodbus: 3.7.3
- Modbus Hardware (if used): custom PCB

### Pymodbus Specific
- Client: rtu - sync

### Description
We use ModbusSerialClient(port=port, framer=FramerType.RTU, baudrate=115200, timeout=1)

When upgrading the package to the newest release 3.7.3, we noticed an rather large delay between messages. These delays were about 2 seconds between sending a message and then receiving a message. We then downgraded the pymodbus back to version 3.7.2, and the communication delay's were not there anymore.

janiversen, 2024-10-21:
You do not tell if you use sync or async version, but I assume sync.

Please add a debug log as pr issue template, so we can see where the time is spent.

For performance you should use the async client, but the sync should not have a delay of 2 seconds.

janiversen, 2024-10-21:
Just ran a short test on my macmini, it made approx. 10 calls pr second.

janiversen, 2024-10-21:
The typical problem for a long delay is if the usb driver do not deliver the whole frame and pymodbus have to make several recv calls.

janiversen, 2024-10-21:
Just tested on my Mac mini, with the standard examples:

```
% time ./client_calls.py --comm serial  --port /dev/ttys010
./client_calls.py --comm serial --port /dev/ttys010  0,10s user 0,03s system 30% cpu 0,407 total
```

0,4 second for 38 requests...so it seems the delay is due to how you use pymodbus (or how the usb driver delivers data).

Once you post a debug log, it is easy to see why you have delay.

buganini, 2024-10-21:
My test code
```python
import pymodbus.client as ModbusClient
import time

client = ModbusClient.ModbusSerialClient(
    "/dev/cu.usbserial-110",
    timeout=0.1,
    retries=3,
    baudrate=19200,
    bytesize=8,
    parity="N",
    stopbits=1,
)

t0 = time.time()
client.connect()
while True:
    r = client.read_holding_registers(0x4, 1, slave=0x1)
    print(r.registers)
    print(time.time() - t0)
```

With 3.7.2
```
[65534]
0.5415279865264893
[65534]
0.5757172107696533
[65534]
0.6073360443115234
[65534]
0.6386821269989014
[65534]
0.6724522113800049
[65534]
0.7062540054321289
[65534]
0.7371621131896973
[65534]
0.7709403038024902
[65534]
0.8046951293945312
[65534]
0.8361661434173584
[65534]
0.8725309371948242
...
```

With 3.7.3:
```
[65534]
2.0475847721099854
[65534]
4.094855785369873
[65534]
6.142652750015259
[65534]
8.194524765014648
[65534]
10.24574899673462
[65534]
12.294071674346924
```

janiversen, 2024-10-21:
As I wrote, that does not help anything....please run with debug log enabled, as the issue template showed how to do.

Without a debug log, we cannot help you!

buganini, 2024-10-21:
Ah, sorry, I missed the chance to read the template..

Here is the log, `<PAUSE HERE>` is where the output paused 

```python
import pymodbus
import pymodbus.client as ModbusClient
import time

pymodbus.pymodbus_apply_logging_config("DEBUG")

client = ModbusClient.ModbusSerialClient(
    "/dev/cu.usbserial-110",
    timeout=0.1,
    retries=3,
    baudrate=19200,
    bytesize=8,
    parity="N",
    stopbits=1,
)

t0 = time.time()
client.connect()
while True:
    r = client.read_holding_registers(0x4, 1, slave=0x1)
    print(r.registers)
    print(time.time() - t0)
```

```
2024-10-21 23:07:12,125 DEBUG transaction:182 Current transaction state - IDLE
2024-10-21 23:07:12,125 DEBUG transaction:188 Running transaction 1
2024-10-21 23:07:12,125 DEBUG transaction:296 SEND: 0x1 0x3 0x0 0x4 0x0 0x1 0xc5 0xcb
2024-10-21 23:07:12,125 DEBUG base:247 New Transaction state "SENDING"
2024-10-21 23:07:12,126 DEBUG transaction:320 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-10-21 23:07:12,155 DEBUG transaction:412 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-10-21 23:07:12,155 DEBUG transaction:327 RECV: 0x1 0x3 0x2 0xff 0xfe 0x78 0x34
2024-10-21 23:07:12,155 DEBUG base:101 Processing: 0x1 0x3 0x2 0xff 0xfe 0x78 0x34
2024-10-21 23:07:12,155 DEBUG factory:259 Factory Response[ReadHoldingRegistersResponse': 3]
2024-10-21 23:07:12,155 DEBUG base:119 Frame advanced, resetting header!!
2024-10-21 23:07:12,155 DEBUG transaction:66 Adding transaction 0
2024-10-21 23:07:12,155 DEBUG transaction:77 Getting transaction 1
2024-10-21 23:07:12,155 DEBUG transaction:77 Getting transaction 0

<PAUSE HERE>

2024-10-21 23:07:13,661 DEBUG transaction:253 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
[65534]
2.0459089279174805
```

janiversen, 2024-10-21:
Seems you found something, which seems to be a bug, I will try to simulate it.

If possible please do the same test with dev, that would help quite a lot to know if we already solved it (by accident).

janiversen, 2024-10-21:
the pull request will be part of v3.7.4 that are due in 1-2 weeks.

closed 2024-10-21 by janiversen

buganini, 2024-10-21:
I can confirm it's fix with [be46b60](https://github.com/pymodbus-dev/pymodbus/commit/be46b605b0a358c620f523d298d5c329f1bd4645)

janiversen, 2024-10-21:
Thanks for the confirmation....have a nice day/night in your lovely country.

SaraEersel, 2024-10-22:
@buganini Thank you for posting the debug log. I wasnt able to do it due to not having the hardware there yesterday.
@janiversen Thank you for the information. en your help.

PR #2401 (2024-10-21): Wrong close, when transaction do not match.

fixes #2398 

a client.close() was not indented correctly (should have been in an else: and not general).

While at it, optimized the transaction code a little bit, by always retrieving the newest transaction.