pymodbus-dev/pymodbus #2409: struct.error: required argument is not an integer when sending list of bytes to write_registers in v3.7.4

willborghei, 2024-10-23:
### Versions

- Python: 3.11
- OS: Linux
- Pymodbus: 3.7.4
- Modbus Hardware (if used):

### Description

This relates to #2368  and #2309 
I am sending a list of bytes to `write_registers` as `values` and still getting the `struct.error: required argument is not an integer` error. I may be missing something here, but I thought with 3.7.4 and #2368 we should be able to send a list of bytes?
I also tried setting the `skip_encode` argument to `True` so that it would skip the `message.encode()` section of the error in `framer/base.py`, but that had the same exact traceback.
Is this an issue on my end?

### Code and Logs

```python
~/modbus_stack.py:80: in write_der_on
    response = self.client.write_registers(address=enable_reg["address"],values=values,slave=int(unit_id))
.venv/lib/python3.11/site-packages/pymodbus/client/mixin.py:358: in write_registers
    return self.execute(no_response_expected, pdu_req_write.WriteMultipleRegistersRequest(address, values,slave=slave,skip_encode=skip_encode))
.venv/lib/python3.11/site-packages/pymodbus/client/base.py:239: in execute
    return self.transaction.execute(no_response_expected, request)
.venv/lib/python3.11/site-packages/pymodbus/transaction.py:218: in execute
    response, last_exception = self._transact(
.venv/lib/python3.11/site-packages/pymodbus/transaction.py:292: in _transact
    packet = self.client.framer.buildFrame(request)
.venv/lib/python3.11/site-packages/pymodbus/framer/base.py:63: in buildFrame
    data = message.function_code.to_bytes(1,'big') + message.encode()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <pymodbus.pdu.register_write_message.WriteMultipleRegistersRequest object at 0x7b62548aac10>

    def encode(self):
        """Encode a write single register packet packet request.
    
        :returns: The encoded packet
        """
>       packet = struct.pack(">HHB", self.address, self.count, self.byte_count)
E       struct.error: required argument is not an integer

.venv/lib/python3.11/site-packages/pymodbus/pdu/register_write_message.py:179: error
------------------------------------------------------------- Captured stderr setup -------------------------------------------------------------
2024-10-24 09:40:21,285 DEBUG tcp:169 Connection to Modbus server established. Socket ('127.0.0.1', 54254)
2024-10-24 09:40:21.285 | INFO     | modbus_stack:connect_to_device:36 - Connected to device at localhost
-------------------------------------------------------------- Captured log setup ---------------------------------------------------------------
DEBUG    pymodbus.logging:tcp.py:169 Connection to Modbus server established. Socket ('127.0.0.1', 54254)
------------------------------------------------------------- Captured stderr call --------------------------------------------------------------
2024-10-24 09:40:21,285 DEBUG transaction:184 Current transaction state - IDLE
2024-10-24 09:40:21,285 DEBUG transaction:193 Running transaction 1
--------------------------------------------------------------- Captured log call ---------------------------------------------------------------
DEBUG    pymodbus.logging:transaction.py:184 Current transaction state - IDLE
DEBUG    pymodbus.logging:transaction.py:193 Running transaction 1
============================================================ short test summary info ============================================================
FAILED tests/test_modbus_stack.py::TestModbusStack::test_write_holding_registers - struct.error: required argument is not an integer
=============================================================== 1 failed in 0.30s ===============================================================
```

willborghei, 2024-10-24:
Found the issue, that error is also raised if the address field is not int and I was not sending int.

closed 2024-10-24 by willborghei

janiversen, 2024-10-24:
Yes, the address needs to be an int.