pymodbus-dev/pymodbus #2423: Modbus TLS ADU Packets faulty recv with 8 Bytes packets? Why not read the complete frame?

hf81-ble, 2024-10-28:
### Versions

- Python: 3.11
- OS: Windows
- Pymodbus: 
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

TLS Client

### Description

The TLS Transactions not not valid at the moment. 

TLS connection with FrameType.Socket will generate correct Frames. My Server will process them an writes the response. BUT the response is not correct received. 

READing Holding Register => 3 Registers (6Bytes)

The Response has 15 Bytes. 7 Bytes the MBAP and 8 Bytes with the payload. 

The transaction part will first look for recv(8) and then try to read 7 Bytes.

BUT in The SSL Socket, it seams that the buffer is cleared! 

Solution;

PyMod 3.7.4

transaction.py Line 343:

If i set the min_size for for the FramerSocket from 8 to my payload size. The message is correctly received. 

There must be a issue in reading from the TLS socket, because if you setting the correct ADU Size, the message processing is working. 

### Code and Logs

```python
# code and logs here.
2024-10-28 17:43:33,133 DEBUG transaction:184 Current transaction state - IDLE
2024-10-28 17:43:33,133 DEBUG transaction:193 Running transaction 1
2024-10-28 17:43:33,133 DEBUG transaction:293 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0xa 0x0 0x3
2024-10-28 17:43:33,133 DEBUG base:250 New Transaction state "SENDING"
2024-10-28 17:43:33,133 DEBUG transaction:317 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-10-28 17:43:33,133 DEBUG transaction:338 transaction None , full False
2024-10-28 17:43:33,133 DEBUG tcp:222 TCP readsize is 8
2024-10-28 17:43:35,200 DEBUG tcp:222 TCP readsize is 7
2024-10-28 17:43:40,313 DEBUG tcp:222 TCP readsize is 7
2024-10-28 17:43:45,319 DEBUG transaction:409 Incomplete message received, Expected 15 bytes Received 8 bytes !!!!
2024-10-28 17:43:45,319 DEBUG transaction:420 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-10-28 17:43:45,319 DEBUG transaction:324 RECV: 0x0 0x1 0x0 0x0 0x0 0x9 0x1 0x3
2024-10-28 17:43:45,319 DEBUG base:92 Processing: 0x0 0x1 0x0 0x0 0x0 0x9 0x1 0x3
2024-10-28 17:43:45,319 DEBUG socket:31 Short frame: 0x0 0x1 0x0 0x0 0x0 0x9 0x1 0x3 wait for more data
2024-10-28 17:43:45,319 DEBUG transaction:78 Getting transaction 1
2024-10-28 17:43:45,319 DEBUG transaction:258 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2024-10-28 17:43:45,319 DEBUG payload:294 None

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config("DEBUG")
```

janiversen, 2024-10-28:
A tls connection is only valid with FramerType.TLS, as pr documentation.

Framer TLS secures a whole package is read in one recv().

janiversen, 2024-10-28:
You do not show any example code, so I cannot see if you use AsyncModbusTlsClient or ModbusTlsClient, which is needed.

Btw framertype.Socket do NOT produce correct frames for TLS, but only for TCP/UDP.

hf81-ble, 2024-10-29:
But as i read the TCP security specs (https://modbus.org/docs/MB-TCP-Security-v21_2018-07-24.pdf) and read the issues (#2203 #2014 ) the modbus TLS implementation is wrong in pymodbus! 

The Current FramerType.TLS only sends the PDU in the ADU Message. The whole MBAP header is missing. So the socket  implementation which sends also the correct MBAP Frames is still valid for TLS ADU packets, But the whole TLS transaction but is lagging. Sometimes sockets are read segmented (min_size=8), but then on TCP/IP the expected response size ist used. 

I made a fix, by considering also the expected response size for the TLS implementation, and it works like charm

fix 
transaction.py 
add line 218
`
##Fix TLS and Socket missing bytes
                elif self.client.comm_params.comm_type == CommType.TLS:
                    Log.debug("TLS and Socket frame: - ")
                    full = True
                    response_pdu_size = request.get_response_pdu_size()
                    expected_response_length = self._calculate_response_length(response_pdu_size)
`

janiversen, 2024-10-29:
The tls frame do not have the mdap header, the socket frame do.

janiversen, 2024-10-29:
Please remember this is open source "Take care about this." is very provoking, if you want a change feel free to submit a pull request.

janiversen, 2024-10-29:
And just the record your line is wrong, or does it also support the ascii and rtu framers ?

it also only covers the sync client.

closed 2024-10-29 by janiversen

hf81-ble, 2024-10-29:
Thanks for your feedback and the fast work. 

Maybe we could use a different FramerType for the legacy implementation and the TLS12?

janiversen, 2024-10-30:
Pull requests are welcome.

PR #2426 (2024-10-29): Check client and frametype.

fixes #2423
(together with #2424)