pymodbus-dev/pymodbus #2432: asyncio lock in client misbehaving

peufeu2, 2024-10-30:
### Versions

- Python: 3.11
- OS: Linux 5.15.80-sunxi
- Pymodbus: 3.7.4
- Modbus Hardware (if used): Waveshare USB-RS485 interface

### Pymodbus Specific

- Client: rtu - async

### Description

In client/base.py:

```
        while count <= self.retries:
            async with self._lock:
                ...do the request...

```

I discovered asyncio.Lock doesn't behave the way I thought: if other tasks want to use this modbus client, the above code will not let them, because the "async with" releases the lock, loops back to the "while", and immediately re-acquires it without yielding. The result is, if a slave is not responding, pymodbus will do all the retries without letting other tasks use the client instance to talk to other slaves on the same bus.

The following code demonstrates this:

```
import asyncio, time

class Test:

    def start( self ):
        self.lock = asyncio.Lock()

        with asyncio.Runner() as runner:
            runner.run(self.astart())

    async def astart( self ):
        async with asyncio.TaskGroup() as tg:
            tg.create_task( self.task1() )
            tg.create_task( self.task2() )

    async def task1( self ):
        while True:
            async with self.lock:
                print("Task 1")
                time.sleep(0.1)
            await asyncio.sleep(0)

    async def task2( self ):
        while True:
            async with self.lock:
                print("Task 2")
                time.sleep(0.1)
            await asyncio.sleep(0)

if __name__ == '__main__':
    Test( ).start()
```

If both lines "await asyncio.sleep(0)" are commented out, only one of the tasks will run. It releases and re-acquires the lock without yielding to the other waiting tasks.

The fix is to yield by adding "await asyncio.sleep(0)" in client/base.py right after line 117. Then other tasks can use the same client between retries to an unresponsive slave.

janiversen, 2024-10-30:
???? pymodbus (and modbus) only allows one request at a time, the lock ensures that !

If you define multiple clients, each client have its own lock, and thus you can make requests on each client in parallel.

The while loop, is only active if the request is not responded to (it is the retry mechanism) and therefore the request is still open.

closed 2024-10-30 by janiversen

janiversen, 2024-10-30:
yes one unresponsive slave delays all others, that is how a bus works !

Normally if a device is unresponsive, the app do not send requests to it, and thus only have a delay when the failure occurs.

Your desired behavior would mean that there are 2 requests open one in each task, and that is not permitted, if you want your desired behavior you need to set retries to zero (as you do) and handle it in the app.

The cause for the problem however is a design error, you try to use the same singleton in 2 tasks.

If you want the code to work differently then feel free to submit a pull request, but be aware, the only one API call open at any time is fundamental.

peufeu2, 2024-10-30:
> yes one unresponsive slave delays all others, that is how a bus works !

Indeed

> Normally if a device is unresponsive, the app do not send requests to it, and thus only have a delay when the failure occurs.

Then how do you know when the device comes back online?

Problem with these "a bit flaky" devices is they often stop responding for at most 4 seconds, then resume as if nothing had happened. Especially the Chinese inverters. It's not serial bit errors, because it fails all the retries for a few seconds, then works again. If I stopped sending requests when that occurs, then the whole system would stop working.

> Your desired behavior would mean that there are 2 requests open one in each task, and that is not permitted

I guess the root cause of our disagreement is the what an "open request" means.

What I believe is your interpretation: an "open request" remains open until it either succeeds or all the retries are done, so it locks the bus exclusively during all that time.

My interpretation: each attempt (retry) opens a new request, then either succeeds or times out, then the request is closed. So it is possible to release the bus between retries.

In fact I believe this is what you intended, otherwise in client/base.py you would have put the "[with lock](https://github.com/pymodbus-dev/pymodbus/blob/5addc5100b02350bad97a4efada05233d1cb7323/pymodbus/client/base.py#L105)" outside of the while so the lock was held during the first try and all the retries. But it is inside the while, which I guess means you intended to release the lock between each retry.

> if you want your desired behavior you need to set retries to zero (as you do) and handle it in the app.

Yes, that's what I did. I set retries to zero, and put another Lock around it so only one task can hit the client at the same time. Requests do interleave properly, everything works fine, but it feels like duplicating code.

> The cause for the problem however is a design error, you try to use the same singleton in 2 tasks.

I don't think it would make any sense to have 2 Client instances on the same serial port, how would they coordinate? The Client instance represents the bus as seen from the master side, not the devices on the bus, so it makes sense that the Client is a singleton. I mean a singleton per bus, since I have 8 serial ports, I have 6 clients and 2 servers. Each Client is a shared resource accessed by several tasks, protected with a lock to interleave requests without overwriting each other. So I disagree on the design error ;)

> One solution which would be acceptable is to have a lock pr slave

That would not prevent two requests being sent to the same bus by two different tasks at the same time, so they would overwrite each other on the bus. Per-slave lock would be useful if I had to perform atomic (read-modify-write) operations on the slaves, but I don't.

peufeu2, 2024-10-30:
Ok, I looked into it some more and I'm officially an idiot 🤣
The example python code in the first post was wrong. "async with lock" does release and let waiters take the lock when it exits, instead of holding onto it as I said, which means pymodbus already behaves like I wanted! So... no pull request.

janiversen, 2024-10-30:
Don't be hard on yourself, you pointed out a bug, thanks for that.

peufeu2, 2024-10-30:
Thanks!

About the lock in Client... it protects against concurrent requests, but if two tasks call connect() at the same time, it will crash. A fix would be something like this in connect():

```
with self._lock:
    if self.connected:
        return
   ...previous connect code

```

janiversen, 2024-10-31:
Pull requests are welcome, but be aware that the library is not build for multithreading, there are a number of calls that probably will crash due to the use of global object in the not refactored code.