pymodbus-dev/pymodbus #2449: instantiation fails with custom port

tpd-opitz, 2024-11-10:
### Versions

- Python: 3.10.12
- OS: Ubuntu 22.04.5 LTS
- Pymodbus: pymodbus-3.7.4
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: not applicapable
- Client: tcp- sync

### Description

initialization of client fails when using custom port, works when no port given

### Code and Logs

#### code
```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

from pymodbus.client import ModbusTcpClient

pymodbus.pymodbus_apply_logging_config("DEBUG")

def no_port(ip_address):
    client = ModbusTcpClient(ip_address)

def custom_port(ip_address, port=1502):
    client = ModbusTcpClient(ip_address, port)
        

# Usage
ip_address = "0.0.0.0"  # any
print('call no_port')
no_port(ip_address)
print('successs', end= '\n\n\n')
print('call custom_port')
custom_port(ip_address)

```
#### log
```text
$ python3 ./kostalWkModbus.py 
call no_port
successs

call custom_port
Traceback (most recent call last):
  File "/home/thomas/Dokumente/Haus/E-Auto/OWB/erweiterung/python/./kostalWkModbus.py", line 20, in <module>
    custom_port(ip_address)
  File "/home/thomas/Dokumente/Haus/E-Auto/OWB/erweiterung/python/./kostalWkModbus.py", line 11, in custom_port
    client = ModbusTcpClient(ip_address, port)
  File "/home/thomas/.local/lib/python3.10/site-packages/pymodbus/client/tcp.py", line 151, in __init__
    super().__init__(framer, retries)
  File "/home/thomas/.local/lib/python3.10/site-packages/pymodbus/client/base.py", line 191, in __init__
    self.framer: FramerBase = (FRAMER_NAME_TO_CLASS[framer])(DecodePDU(False))
KeyError: 1502

```

janiversen, 2024-11-10:
Please look at the documentation, your call does not conform with the API.

janiversen, 2024-11-10:
At the very least look at the call signature, then you will see your parameters are incorrect.

closed 2024-11-10 by janiversen

tpd-opitz, 2024-11-10:
@janiversen could you please be more specific? The [Docu](https://pymodbus.readthedocs.io/en/latest/source/client.html#client-device-addressing) states that only the host is mandatory an the and the parameter name for the port is `port`...