pymodbus-dev/pymodbus #2459: ModbusClientMixin.write_registers typing could be loosened

vfazio, 2024-11-19:
Note: all code is untested except by unit tests.

### Pymodbus Specific

- Client

### Description

The typed interface for `ModbusClientMixin.write_registers` has flip-flopped a few times in recent releases, but it's still slightly wrong IMO.

The current typing, as of the time of writing, is:

```python
    def write_registers(
        self,
        address: int,
        values: list[bytes | int],
        *,
        slave: int = 1,
        no_response_expected: bool = False
    ) -> T:
```

The problem with this typing is that any code calling into this function that is explicitly typed as `values: Sequence[int]` or `values: list[int]` will fail type checks.

It seems the motivation to migrate this from `list[int]` to `list[int] | list[bytes]` to `list[int|bytes]` is to denote that the input can be a mixture of ints or bytes and that the interface will handle conversions as necessary.

Unfortunately, because `list` is mutable, it is invariant. Type checkers, like mypy, assume because it's typed as `list[int|byte]` that the function could _potentially_ mutate this list to contain a type that a caller, who has typed their interfaces to be `list[int]` to be unaware of and would be unable to handle.

These interfaces _should not_ mutate the value, so can and probably should loosen their types as much as possible... with a few tweak, it could be as generic as:

```python
values: Iterable[int|bytes]
```

The underlying code really only needs it to be iterable (via `enumerate`). The only exception is:

```python
self.count: int = count or len(registers)
```

Which would mean it needs to be `Container` or `Sequence` however I think this could be avoided by simply doing:

```python
self.count: int = count or len(self.registers)
```

### Code and Logs

```diff
diff --git a/examples/client_async_calls.py b/examples/client_async_calls.py
index d2c118e5..2c119e8d 100755
--- a/examples/client_async_calls.py
+++ b/examples/client_async_calls.py
@@ -163,13 +163,15 @@ async def async_write_registers_mypy(client):
     await client.write_registers(1, regs1, slave=SLAVE)
     rr = await client.read_holding_registers(1, count=len(regs1), slave=SLAVE)
     assert not rr.isError()  # test that call was OK
+    assert rr.registers is not regs1
     assert rr.registers == regs1
 
     regs2: list[bytes] = [b'\x01\x02', b'\x03\x04']
     await client.write_registers(1, regs2, slave=SLAVE)
     rr = await client.read_holding_registers(1, count=len(regs2), slave=SLAVE)
     assert not rr.isError()  # test that call was OK
-    assert rr.registers == regs2
+    assert rr.registers is not regs2
+    assert rr.registers == [int.from_bytes(i, byteorder="big") for i in regs2]
 
 
 async def async_handle_input_registers(client):
diff --git a/pymodbus/client/mixin.py b/pymodbus/client/mixin.py
index 23eebc89..215e7353 100644
--- a/pymodbus/client/mixin.py
+++ b/pymodbus/client/mixin.py
@@ -2,6 +2,7 @@
 from __future__ import annotations
 
 import struct
+from collections.abc import Iterable
 from enum import Enum
 from typing import Generic, TypeVar
 
@@ -501,7 +502,7 @@ class ModbusClientMixin(Generic[T]):  # pylint: disable=too-many-public-methods
     def write_registers(
         self,
         address: int,
-        values: list[bytes | int],
+        values: Iterable[bytes | int],
         *,
         slave: int = 1,
         no_response_expected: bool = False
(venv) vfazio@vfazio4 /mnt/development/pymodbus $ PAGER= git diff
diff --git a/examples/client_async_calls.py b/examples/client_async_calls.py
index d2c118e5..2c119e8d 100755
--- a/examples/client_async_calls.py
+++ b/examples/client_async_calls.py
@@ -163,13 +163,15 @@ async def async_write_registers_mypy(client):
     await client.write_registers(1, regs1, slave=SLAVE)
     rr = await client.read_holding_registers(1, count=len(regs1), slave=SLAVE)
     assert not rr.isError()  # test that call was OK
+    assert rr.registers is not regs1
     assert rr.registers == regs1
 
     regs2: list[bytes] = [b'\x01\x02', b'\x03\x04']
     await client.write_registers(1, regs2, slave=SLAVE)
     rr = await client.read_holding_registers(1, count=len(regs2), slave=SLAVE)
     assert not rr.isError()  # test that call was OK
-    assert rr.registers == regs2
+    assert rr.registers is not regs2
+    assert rr.registers == [int.from_bytes(i, byteorder="big") for i in regs2]
 
 
 async def async_handle_input_registers(client):
diff --git a/pymodbus/client/mixin.py b/pymodbus/client/mixin.py
index 23eebc89..215e7353 100644
--- a/pymodbus/client/mixin.py
+++ b/pymodbus/client/mixin.py
@@ -2,6 +2,7 @@
 from __future__ import annotations
 
 import struct
+from collections.abc import Iterable
 from enum import Enum
 from typing import Generic, TypeVar
 
@@ -501,7 +502,7 @@ class ModbusClientMixin(Generic[T]):  # pylint: disable=too-many-public-methods
     def write_registers(
         self,
         address: int,
-        values: list[bytes | int],
+        values: Iterable[bytes | int],
         *,
         slave: int = 1,
         no_response_expected: bool = False
diff --git a/pymodbus/pdu/pdu.py b/pymodbus/pdu/pdu.py
index b65426c6..67a9fdd6 100644
--- a/pymodbus/pdu/pdu.py
+++ b/pymodbus/pdu/pdu.py
@@ -4,7 +4,7 @@ from __future__ import annotations
 import asyncio
 import struct
 from abc import abstractmethod
-from collections.abc import Sequence
+from collections.abc import Iterable
 from enum import Enum
 from typing import cast
 
@@ -26,7 +26,7 @@ class ModbusPDU:
             address: int = 0,
             count: int = 0,
             bits: list[bool] | None = None,
-            registers: Sequence[int | bytes] | None = None,
+            registers: Iterable[int | bytes] | None = None,
             status: int = 1,
         ) -> None:
         """Initialize the base data for a modbus request."""
@@ -36,11 +36,16 @@ class ModbusPDU:
         self.bits: list[bool] = bits or []
         if not registers:
             registers = []
-        self.registers: list[int] = cast(list[int], registers)
-        for i, value in enumerate(registers):
+        self.registers: list[int] = []
+        # copy values to not mutate the original sequence of values
+        for value in registers:
             if isinstance(value, bytes):
-                self.registers[i] = int.from_bytes(value, byteorder="big")
-        self.count: int = count or len(registers)
+                self.registers.append(int.from_bytes(value, byteorder="big"))
+            elif isinstance(value, int):
+                self.registers.append(value)
+            else:
+                raise ValueError("Invalid value in iterable")
+        self.count: int = count or len(self.registers)
         self.status: int = status
         self.fut: asyncio.Future
```

```shell
vfazio@vfazio4 /mnt/development/pymodbus $ cat test.py 
from pymodbus.client import ModbusTcpClient

mbc = ModbusTcpClient("localhost")

z: list[int] = [1, 2]

mbc.write_registers(10, z)

vfazio@vfazio4 /mnt/development/pymodbus $ mypy test.py 
test.py:7: error: Argument 2 to "write_registers" of "ModbusClientMixin" has incompatible type "list[int]"; expected "list[bytes | int]"  [arg-type]
test.py:7: note: "List" is invariant -- see https://mypy.readthedocs.io/en/stable/common_issues.html#variance
test.py:7: note: Consider using "Sequence" instead, which is covariant
Found 1 error in 1 file (checked 1 source file)
vfazio@vfazio4 /mnt/development/pymodbus $ git stash pop
On branch dev
Your branch is up to date with 'origin/dev'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   pymodbus/client/mixin.py
        modified:   pymodbus/pdu/pdu.py

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        build/lib/
        test.py

no changes added to commit (use "git add" and/or "git commit -a")
Dropped refs/stash@{0} (75b0cf37e3435354a7f5c6911224ff32831bfbca)
vfazio@vfazio4 /mnt/development/pymodbus $ mypy test.py 
Success: no issues found in 1 source file

vfazio@vfazio4 /mnt/development/pymodbus $ mypy pymodbus
Success: no issues found in 47 source files
```

vfazio, 2024-11-19:
If this seems reasonable, i can submit a PR

janiversen, 2024-11-19:
We have added several tests, among other one that ensures you can call with list[int], so I am not sure what the problem is.

'''
async def async_write_registers_mypy(client):
    """Read/write holding registers."""
    regs1: list[int] = [10] * 8
    await client.write_registers(1, regs1, slave=SLAVE)
    rr = await client.read_holding_registers(1, count=len(regs1), slave=SLAVE)
    assert not rr.isError()  # test that call was OK
    assert rr.registers == regs1

    regs2: list[bytes] = [b'\x01\x02', b'\x03\x04']
    await client.write_registers(1, regs2, slave=SLAVE)
    rr = await client.read_holding_registers(1, count=len(regs2), slave=SLAVE)
    assert not rr.isError()  # test that call was OK
    assert rr.registers == regs2
'''

The reason it has changed during the last releases was to make it mypy complaint.

The current signature indicates that the list can be a mixture of int and bytes which is what is allowed.

The idea of mypy is not to relax the typing as much as possible, but to be specific.....my personal preference would be to remove typing, but for some reason a couple of users have had strong opinions on how the typing should be.

I cannot judge your code changes, without a pr, but it looks like a lot of changes just to change typing.

janiversen, 2024-11-20:
Having modbusPdu validate data is a nice idea, but that is very extensive and needs a lot of testing, so that is yet another PR.

vfazio, 2024-11-21:
When i get time, maybe tomorrow or next week I will make 3 MRs:

1. Make a copy of the sequence to be owned by `ModbusPDU` (internal change)
2. Validate data type in sequence and raise a TypeError/ValueError on invalid value (internal change, depends on 1)
3. Update the types used in the APIs (external change, depends on 1, not on 2). I will potentially update the tests to use typing for these interfaces so that they are caught via mypy should there be a regression/breakage.

janiversen, 2024-11-24:
Closing this as there is a test securing that list[int] and list[bytes] are all allowed:
```
async def async_write_registers_mypy(client):
    """Read/write holding registers."""
    regs1: list[int] = [10] * 8
    await client.write_registers(1, regs1, slave=SLAVE)
    rr = await client.read_holding_registers(1, count=len(regs1), slave=SLAVE)
    assert not rr.isError()  # test that call was OK
    assert rr.registers == regs1

    regs2: list[bytes] = [b'\x01\x02', b'\x03\x04']
    await client.write_registers(1, regs2, slave=SLAVE)
    rr = await client.read_holding_registers(1, count=len(regs2), slave=SLAVE)
    assert not rr.isError()  # test that call was OK
    assert rr.registers == regs2
```
We do not want to allow random typings.

janiversen, 2024-11-24:
I am taking quite a lot of time to discuss the issues, which you can see if you look at other issues.

The mypy test was added due to an outside contributor, who had a problem with exact the same call as you ! He confirmed that the typing worked for him.

Your PR is not bad, but very inefficient which I why I requested changes.

I have taken quite a time to understand your problem....the call is typed "values: list[int | bytes]", we test that "values: list[int]" and "values: list[bytes]" passed mypy (otherwise the other contributor would have been very unhappy).

You refer partly to an internal typing in pdu.py, which should not be a concern for users of the API. The Sequence in pdu.py have (in response to your comments) been changed and are no longer !

Please feel free to state (again) what is your problem with the API typing... (We intentionally do not want to have a very open typing), but do not tell me, that I do not spend time listening and discussing !!

janiversen, 2024-11-24:
ps if we wanted a loosened API, we would not have typed it all !!!

vfazio, 2024-11-24:
The mypy checks are not _complete_. I mentioned this in my previous comment

The test functions are not typed properly, so they are defaulting to the `Any` type which allows for any typing to pass.

When the proper typing is applied, they fail:

```diff
(venv) vfazio@vfazio4 /mnt/development/pymodbus $ git diff
diff --git a/examples/client_async_calls.py b/examples/client_async_calls.py
index d2c118e5..6dad4a95 100755
--- a/examples/client_async_calls.py
+++ b/examples/client_async_calls.py
@@ -34,6 +34,7 @@ import logging
 import sys
 
 from pymodbus import ModbusException
+from pymodbus.client import ModbusBaseClient
 from pymodbus.pdu import FileRecord
 
 
@@ -157,7 +158,7 @@ async def async_handle_holding_registers(client):
     assert not rr.isError()  # test that call was OK
     assert rr.registers == arguments["values"]
 
-async def async_write_registers_mypy(client):
+async def async_write_registers_mypy(client: ModbusBaseClient) -> None:
     """Read/write holding registers."""
     regs1: list[int] = [10] * 8
     await client.write_registers(1, regs1, slave=SLAVE)
```

```shell
(venv) vfazio@vfazio4 /mnt/development/pymodbus $ mypy examples/client_async_calls.py 
examples/client_async_calls.py:164: error: Argument 2 to "write_registers" of "ModbusClientMixin" has incompatible type "list[int]"; expected "list[bytes | int]"  [arg-type]
examples/client_async_calls.py:164: note: "List" is invariant -- see https://mypy.readthedocs.io/en/stable/common_issues.html#variance
examples/client_async_calls.py:164: note: Consider using "Sequence" instead, which is covariant
examples/client_async_calls.py:170: error: Argument 2 to "write_registers" of "ModbusClientMixin" has incompatible type "list[bytes]"; expected "list[bytes | int]"  [arg-type]
examples/client_async_calls.py:170: note: "List" is invariant -- see https://mypy.readthedocs.io/en/stable/common_issues.html#variance
examples/client_async_calls.py:170: note: Consider using "Sequence" instead, which is covariant
examples/client_async_calls.py:173: error: Non-overlapping equality check (left operand type: "list[int]", right operand type: "list[bytes]")  [comparison-overlap]
Found 3 errors in 1 file (checked 1 source file)
```

janiversen, 2024-11-24:
hmmm looks like you have a point, do not know why the other contributor did not catch that.

That will be solved in a couple of hours.

vfazio, 2024-11-24:
I'm sorry for my attitude earlier. it was inappropriate. I've spent a lot of time working on typing not only on my project but other projects as well. I have actually just helped rework much of https://github.com/brgl/libgpiod/commits/master/bindings/python so i do have some experience when it comes to properly typing interfaces.

janiversen, 2024-11-24:
I have never doubted your experience (just a little bit your implementation :-) ). Apology accepted, we all have bad days....and frankly I did not understand the mypy problem...seems there is a difference between these 2 calls:
```
mypy examples (this works)
mypy client_async_calls.py (this one complains, including files that are not used)
```

Anyhow the solution is easy, and I will have a patch in a moment.

janiversen, 2024-11-24:
mypy is just about the worst type checker I have ever experienced (coming from C++ / C# and Swift)....if I define a type as
```
   values: list[int] | list[bytes]
```
Normal developers read this as we accept list[int] OR list[bytes] but not mypy it demands the parameter to be "list[int] | list[bytes]".

Well I circumvented it.

closed 2024-11-24 by janiversen

janiversen, 2024-11-24:
Solution added to dev, only list[int] is allowed....the request (PR) to allow list[bytes] came from a contributor, but have proven far to difficult to manage in practice.

PR #2468 (2024-11-24): Write_registers/pdu typing again.

fixes #2459