pymodbus-dev/pymodbus #2482: Modbus connect successful but fail for read_input_registers and read_holding_registers

mintheinwin, 2024-11-29:
I am using the Calypso wind direction sensor from the product link provided below:  
[https://calypsoinstruments.com/shop/product/ultra-low-power-ultrasonic-wind-meter-ulp-standard-6#attr=76,71,65](url).

The device connected successfully, but reading the registers failed. The device contains 14 registers. What parameters should be passed for read_input_registers  and read_holding_registers?

#my full code.
import asyncio
from pymodbus.client.asynchronous.serial import AsyncModbusSerialClient
from pymodbus.framer.rtu_framer import ModbusRtuFramer

async def modbus_client_test():
    # Initialize Modbus client
    modbus_client = AsyncModbusSerialClient(
        port='/dev/ttyS0',
        framer=ModbusRtuFramer,
        baudrate=38400,
        stopbits=1,
        bytesize=8,
        parity='N',
        timeout=1
    )

    # Use the client
    async with modbus_client as client:
        if client.connected:
            print("Connection successful!")

            result = await client.read_holding_registers(address=1, count=1, slave=2)

            if not result.isError():
                print("Registers read successfully:", result.registers)
            else:
                print("Error reading registers:", result)
        else:
            print("Failed to connect!")

if _name_ == "_main_":
    asyncio.run(modbus_client_test())

Error reading registers: Exception Response (131,3,None)
[image]

janiversen, 2024-11-29:
As you can see in the log, your device does not respond.

That can be due to quite a number of parameters being wrong.

That connect() succeeds only means pymodbus can open the port and set the communication parameters.

janiversen, 2024-11-29:
Potentially it can also be your rs485 converter or the wiring.

mintheinwin, 2024-11-29:
How can I verify the device's communication parameters or ensure the slave configuration is correct?
Which third-party tools are easy to use for Modbus?
Calypso wind direction sensor Register number starting from 200.

janiversen, 2024-11-29:
Read the manual, apart from that try and error.

What do you mean by third party tools ? search in google.

janiversen, 2024-11-29:
Not sure what you need, pymodbus offer REPL, please see the documentation.

This is the pymodbus project we generally do not speak about other projects or software.

janiversen, 2024-11-29:
There are no indication this is a pymodbus problem, so closing.

closed 2024-11-29 by janiversen