pymodbus-dev/pymodbus #2496: Modbus client continuously reading data even if it is not being written by the server at that frequency

GargiGit24, 2024-12-11:
### Versions

- Python: 3.9
- OS:
- Pymodbus: 3.8.0
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async : using client object to write data in registers  
- Client: tcp/rtu/ascii - sync/async : tcp async

### Description

In my server, for a particular register. I am writing data in a one-minute frequency.
I expect that in my client the data should come as it is written in the server (one value per minute).
But, my client is reading data continuously from the server. Most of the values are repeated as the server writes one value per minute.
I want the client to only bring the value when server has written it and avoid repeated calls.
The client code should be generic and handle data of varying frequencies.
### Code and Logs

```python
# code and logs here.

import time
import pandas as pd
import asyncio
from pymodbus.client import AsyncModbusTcpClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian
from pymodbus.exceptions import ModbusException

def decode_float64(registers):
    decoder = BinaryPayloadDecoder.fromRegisters(registers, byteorder=Endian.BIG, wordorder=Endian.LITTLE)
    return decoder.decode_64bit_float()
async def read_modbus_data(client: AsyncModbusTcpClient):
    start_address = 1801
    num_of_reg = 4
    datatype = "64_bit_float"

    try:
        response = await client.read_holding_registers(address=start_address, count=num_of_reg)
        if response is None:
            return None
        if response.isError():
            print(f"Error reading registers for : {response}")
            return None

        if datatype == '64_bit_float':
            if len(response.registers) < 4:
                print("Not enough registers to decode 64-bit float. Will need atleast 4 registers.")
            read_value = decode_float64(response.registers[:4])
            return read_value
    except ModbusException as exc:
        print(f"ModbusException: {exc}")
        return None
    

async def main():
    async with AsyncModbusTcpClient(host=host, port=port, timeout=timeout) as client:
        while True:
            result = await read_modbus_data(client)
            print(result)
            
# Run the main function
asyncio.run(main())

janiversen, 2024-12-11:
Its strange ! you state you are using v3.8.0, but the read_holding_registers does not correspond to the call signature of v3.8.0 ?? your call should be rejected by python.

you use BinaryPayloadDecoder.fromRegisters, did you note the warning it issues, that it is deprecated and will be removed soon ?

Apart from that I have a client that read in 10ms cycles, so pymodbus have no problem doing that. However your code loops as fast as possible that does not sound like the best idea.

You can add a debug log (using the pymodbus call to do so), and then you get detailed what pymodbus sends and receives, that will help to confirm that your server sends the same data multiple times.

janiversen, 2024-12-11:
Can it be you have misunderstood how modbus works......the client does not know when data in the server is updated, whenever it calls it get the current value.

janiversen, 2024-12-12:
Closing as it seems the client works correctly, but it is a user misunderstanding as to how the protocol works.

closed 2024-12-12 by janiversen

GargiGit24, 2024-12-17:
Hello,
I use BinaryPayloadDecoder.fromRegisters because when I try to use convert_to_registers I cannot give the datatype properly. In my case I want int, float, string and bool datatypes with all bit combinations in int and float.

Also, I see that it is a warning, as it will be removed in the version 3.9.0, I am currently using version 3.8.0
 Could you please give me a short example of the usage?

However, to better explain my use case. I am building a streaming plugin which will stream data from the registers continuously as they send it. Hence for continuous writing I had written the while True loop.
If I remove the loop the data is written only once. Could you please explain your approach and provide a short snippet for clarity?

janiversen, 2024-12-17:
convert_to_registers takes a parameter DATATYPE, just see the signature.

int can be anything between INT16, INT32 and INT64 (same for float), you know which type to use, since it is directly related to the number of registers you read.

I have no idea what "bit combinations in int and float" means, that does not make sense to me. You have the raw registers and can do anything.

Your loop contains a READ no WRITE, so data is never written, but only read ! 

Just see the examples already at hand, a client typically have a read loop, but with a sleep, not to overload your server.

janiversen, 2024-12-17:
ps. the current version of convert_to_registers use word_order == "big", because that is what most devices do.

If you want word_order == "little" there are 2 options in order to use the upcoming v3.9:
- Do the conversion in your app, using the standard python functions.
- Make a pull request, adding a parameter word_order to the convert methods
  (see #2506, which is an attempt to do that, but with a major misunderstanding)

GargiGit24, 2025-01-09:
Hi, 

- The answer to "Your loop contains a READ no WRITE, so data is never written, but only read !" is that I write in a local dataset. I am using dataiku for that, my modbus client runs in dataiku as a continuous code and writes the output to a dataiku dataset which is nothing but a CSV file.

- By bit combinations in int and float I mean the following datatypes:
16_bit_int, 32_bit_int, 64_bit_int, 32_bit_float, 64_bit_float, string, boolean

- For float32 and float64 decoding, I need word_order == "little", I will try doing the conversion in my app first.
- I will need to keep the loop as per my understanding as I expect my client to read the registers whenever they give data.
- As for now I assume, that you are suggesting using convert_to_registers, will solve this problem.
- I will reach out if it does not.

Thanks for your help, please let me know your thoughts on it

janiversen, 2025-01-09:
Be aware a modbus server responds imidiatly to a request, delivering the content of its registers....totally independent from when these registers change.

In other words modbus is a polling protocol not a push protocol.