pymodbus-dev/pymodbus #2505: Cannot get coils to read correctly

carthoris, 2024-12-14:
We are using out-of-the-box modbus simulator. We successfully were able to read registers but are having a hard time reading coils.
We initialized all the coils with a 0x1 and only get 1 coil back with a 1.

When running and stopping in debug we get 

[True, False, False, False, False, False, False, False]

instead of what we expect to be all True.

We have been playing with the client.read_coils method trying different addresses and counts to no success.

We are using a modified sample client to try to get the coils to read as follows:

#!/usr/bin/env python3
"""Pymodbus synchronous client example.

An example of a single threaded synchronous client.

usage: simple_sync_client.py

All options must be adapted in the code
The corresponding server must be started before e.g. as:
    python3 server_sync.py
"""

# --------------------------------------------------------------------------- #
# import the various client implementations
# --------------------------------------------------------------------------- #
import pymodbus.client as ModbusClient
from pymodbus import (
    FramerType,
    ModbusException,
    pymodbus_apply_logging_config,
)

def run_sync_simple_client(comm, host, port, framer=FramerType.SOCKET):
    """Run sync client."""
    # activate debugging
    pymodbus_apply_logging_config("DEBUG")

    print("get client")
    client: ModbusClient.ModbusBaseSyncClient
    if comm == "tcp":
        client = ModbusClient.ModbusTcpClient(
            host,
            port=port,
            framer=framer,
            # timeout=10,
            # retries=3,
            # source_address=("localhost", 0),
        )
    elif comm == "udp":
        client = ModbusClient.ModbusUdpClient(
            host,
            port=port,
            framer=framer,
            # timeout=10,
            # retries=3,
            # source_address=None,
        )
    elif comm == "serial":
        client = ModbusClient.ModbusSerialClient(
            port,
            framer=framer,
            # timeout=10,
            # retries=3,
            baudrate=9600,
            bytesize=8,
            parity="N",
            stopbits=1,
[... 297 lines cut ...]
                }
            },
            "invalid": [
            ],
            "write": [
			    [0, 7],
                [21, 25]
            ],
            "bits": [
			  {"addr": [0, 15], "value": 1}
            ],
            "uint16": [
                [16, 25]
            ],
            "uint32": [
            ],
            "float32": [
            ],
            "string": [
            ],
            "repeat": [
            ]
        }
    }
}

janiversen, 2024-12-15:
You have not defined 8 bit as true, but only 1 bit. I'm the bit section address is (if I remember correctly) for a single bit.

janiversen, 2024-12-15:
You client call looks ok, please add a debug log (as pr issue template) so we can see what the server actually sends.

janiversen, 2024-12-15:
The problem might be your addressing, coils are addressed as bits not registers, so the first 16bit are stored in register 1 etc.

janiversen, 2024-12-15:
A good idea is to reduce the json to only contain the server type and device you are actually using, otherwise it's quite pure guesswork to help.

janiversen, 2024-12-18:
Closing as this seems to be a configuration issue, and not a bug.

Feel free to correct if I am wrong.

closed 2024-12-18 by janiversen

carthoris, 2024-12-24:
most likely it is configuration but not sure what.

janiversen, 2024-12-24:
Looks like a adressing as I wrote, but you have multiple devices definedd do not tell what you use, so it's hard to help, please follow the guidelines I gave you earlier.

janiversen, 2024-12-24:
And just for information I do not think you initialized coils to be true, the default is still set to false.

carthoris, 2024-12-29:
we are using devicev

"devicev": {
"setup": {
"co size": 8,
"di size": 8,
"ir size": 5,
"hr size": 5,
"shared blocks": false,
"type exception": true,
"defaults": {
"value": {
"bits": 0,
"uint16": 0,
"uint32": 0,
"float32": 0.0,
"string": " "
},
"action": {
"bits": null,
"uint16": null,
"uint32": null,
"float32": null,
"string": null
}
}
},
"invalid": [
],
"write": [
[0, 7],
[21, 25]
],
"bits": [
{"addr": [0, 15], "value": 1}
],
"uint16": [
[16, 25]
],
"uint32": [
],
"float32": [
],
"string": [
],
"repeat": [
]
}

janiversen, 2024-12-29:
But value 1 only sets 1 bit and leaves the rest 0.