pymodbus-dev/pymodbus #2509: Lacking examples and documentation on deprecation of BinaryPayloadDecoder etc

MichielfromNL, 2024-12-17:
I am trying to migrate from the BinaryPayloadDecoder approach to client.convert_from_registers() and client.convert_to_registers() as suggested by the deprecation warning in 3.8.0

However, there is no documentation, and there are no examples yet for these procedures. Which even confuses Copilot in Github, it has no clue in how to use them to convert "new style" .
It would be nice to have the old approach around for 1 or two releases, until doc & examples are updated, to allow for a smoother code upgrade process

Tx !

janiversen, 2024-12-17:
Copilot in vscode can use the signatures of the methods which we provide, it cannot use arbitrary examples.

Binarypayload have been around for many years, and it was first announced as deprecated (not in the code) with 3.1, so we do not want to postpone it more....users are welcome to download the binarypayload from an old release, the only real difference is that we do not maintain it.

janiversen, 2024-12-17:
And just to correct you, there are 3 examples using convert_from.

janiversen, 2024-12-17:
But as always pull requests are welcome if you wan5 to add more examples.

janiversen, 2024-12-18:
Closing as this seems to be a non-issue.

closed 2024-12-18 by janiversen

MichielfromNL, 2024-12-19:
My bad, apologies, looked in the wrong place for the examples.