pymodbus-dev/pymodbus #2513: no current event loop when running ModbusSerialClient in separate thread

flavius-t, 2024-12-19:
### Versions

- Python: 3.9.2
- OS: Debian GNU/Linux 11
- Pymodbus: 3.8.1
- Pyserial: 3.5
- Modbus Hardware (if used):
    - A raspberry pi-based device with the modbus ID, baudrate, and readable address indicated in the code demo below, reading over RS485-to-USB cable.

### Pymodbus Specific

- Client: rtu - sync

### Description

Hi, I've been facing an issue since upgrading to pymodbus `3.8.1`. Using `ModbusSerialClient` in a separate thread from the main thread raises a runtime error related to asyncio usage. This issue was not present in `3.7.4`, but is also present in `3.8.0`.

### Code and Logs

```python
import traceback
from concurrent.futures import ThreadPoolExecutor

from pymodbus.client import ModbusSerialClient

def read_modbus():
    try:
        print("Reading modbus registers")
        with ModbusSerialClient(port='/dev/ttyUSB0', baudrate=115200) as client:
            out = client.read_holding_registers(address=4144, count=1, slave=241)
            print(f"Reading holding registers: slave_id={241}, address={4144}, num_regs={1}: response={out}")
    except Exception:
        print(traceback.format_exc())

if __name__ == "__main__":
    print("running from main thread")
    read_modbus()

    print("running from thread pool")
    executor = ThreadPoolExecutor()
    executor.submit(read_modbus)
```

### Logs running from 3.8.1
```
running from main thread
Reading modbus registers
2024-12-19 23:22:25,722 DEBUG base:91 Processing: 0xf1 0x3 0x2 0xb 0xe8 0xff 0x2f
2024-12-19 23:22:25,723 DEBUG decoders:113 decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[3048], status=1) 
2024-12-19 23:22:25,723 DEBUG base:101 Frame advanced, resetting header!!
Reading holding registers: slave_id=241, address=4144, num_regs=1: response=ReadHoldingRegistersResponse(dev_id=241, transaction_id=0, address=0, count=0, bits=[], registers=[3048], status=1)
running from thread pool
Reading modbus registers
Traceback (most recent call last):
  File "/<path>/modbus_script.py", line 12, in read_modbus
    with ModbusSerialClient(port='/dev/ttyUSB0', baudrate=115200) as client:
  File "/<path>/venv_demo/lib/python3.9/site-packages/pymodbus/client/serial.py", line 200, in __init__
    super().__init__(
  File "/<path>/venv_demo/lib/python3.9/site-packages/pymodbus/client/base.py", line 135, in __init__
    self.transaction = TransactionManager(
  File "/<path>/venv_demo/lib/python3.9/site-packages/pymodbus/transaction/transaction.py", line 65, in __init__
    self.response_future: asyncio.Future = asyncio.Future()
  File "/usr/lib/python3.9/asyncio/events.py", line 642, in get_event_loop
    raise RuntimeError('There is no current event loop in thread %r.'
RuntimeError: There is no current event loop in thread 'ThreadPoolExecutor-0_0'.
```

### Logs running from 3.7.4
```
running from main thread
Reading modbus registers
2024-12-19 23:38:15,176 DEBUG transaction:184 Current transaction state - IDLE
2024-12-19 23:38:15,176 DEBUG transaction:193 Running transaction 0
2024-12-19 23:38:15,176 DEBUG transaction:293 SEND: 0xf1 0x3 0x10 0x30 0x0 0x1 0x94 0x35
2024-12-19 23:38:15,177 DEBUG base:250 New Transaction state "SENDING"
2024-12-19 23:38:15,177 DEBUG transaction:317 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-12-19 23:38:15,192 DEBUG transaction:419 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-12-19 23:38:15,192 DEBUG transaction:324 RECV: 0xf1 0x3 0x2 0xb 0xe8 0xff 0x2f
2024-12-19 23:38:15,192 DEBUG base:92 Processing: 0xf1 0x3 0x2 0xb 0xe8 0xff 0x2f
2024-12-19 23:38:15,193 DEBUG decoders:103 decode PDU for 3
2024-12-19 23:38:15,193 DEBUG base:102 Frame advanced, resetting header!!
2024-12-19 23:38:15,193 DEBUG transaction:67 Adding transaction 0
2024-12-19 23:38:15,193 DEBUG transaction:78 Getting transaction 0
2024-12-19 23:38:15,193 DEBUG transaction:258 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Reading holding registers: slave_id=241, address=4144, num_regs=1: response=ReadHoldingRegistersResponse (1)
running from thread pool
Reading modbus registers
2024-12-19 23:38:15,197 DEBUG transaction:184 Current transaction state - IDLE
2024-12-19 23:38:15,198 DEBUG transaction:193 Running transaction 0
2024-12-19 23:38:15,199 DEBUG transaction:293 SEND: 0xf1 0x3 0x10 0x30 0x0 0x1 0x94 0x35
2024-12-19 23:38:15,199 DEBUG base:250 New Transaction state "SENDING"
2024-12-19 23:38:15,200 DEBUG transaction:317 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-12-19 23:38:15,214 DEBUG transaction:419 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-12-19 23:38:15,215 DEBUG transaction:324 RECV: 0xf1 0x3 0x2 0xb 0xe8 0xff 0x2f
2024-12-19 23:38:15,215 DEBUG base:92 Processing: 0xf1 0x3 0x2 0xb 0xe8 0xff 0x2f
2024-12-19 23:38:15,216 DEBUG decoders:103 decode PDU for 3
2024-12-19 23:38:15,216 DEBUG base:102 Frame advanced, resetting header!!
2024-12-19 23:38:15,217 DEBUG transaction:67 Adding transaction 0
2024-12-19 23:38:15,217 DEBUG transaction:78 Getting transaction 0
2024-12-19 23:38:15,217 DEBUG transaction:258 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Reading holding registers: slave_id=241, address=4144, num_regs=1: response=ReadHoldingRegistersResponse (1)
```

Thank you!

FranzEricSchneider, 2024-12-20:
Sounds like it might be related, before I was able to connect to the modbus client in an ipython window, but today when I try I get the same runtime error. I'm not exactly sure what version I was running, but I confirmed now that 3.8.0 has the same behavior and 3.7.4 works.

Code:
```
In [1]: from pymodbus.client import ModbusTcpClient
In [2]: client = ModbusTcpClient("127.0.0.1", port=502)
```

Error:
```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In[2], line 1
----> 1 client = ModbusTcpClient("127.0.0.1", port=502)

File /usr/local/lib/python3.10/dist-packages/pymodbus/client/tcp.py:170, in ModbusTcpClient.__init__(self, host, framer, port, name, source_address, reconnect_delay, reconnect_delay_max, timeout, retries, trace_packet, trace_pdu, trace_connect)
    159         raise TypeError("Only FramerType SOCKET/RTU/ASCII allowed.")
    160     self.comm_params = CommParams(
    161         comm_type=CommType.TCP,
    162         host=host,
   (...)
    168         timeout_connect=timeout,
    169     )
--> 170 super().__init__(
    171     framer,
    172     retries,
    173     self.comm_params,
    174     trace_packet,
    175     trace_pdu,
    176     trace_connect,
    177 )
    178 self.socket: socket.socket | None = None

File /usr/local/lib/python3.10/dist-packages/pymodbus/client/base.py:135, in ModbusBaseSyncClient.__init__(self, framer, retries, comm_params, trace_packet, trace_pdu, trace_connect)
    133 # Common variables.
    134 self.framer: FramerBase = (FRAMER_NAME_TO_CLASS[framer])(DecodePDU(False))
--> 135 self.transaction = TransactionManager(
    136     self.comm_params,
    137     self.framer,
    138     retries,
    139     False,
    140     trace_packet,
    141     trace_pdu,
    142     trace_connect,
    143     sync_client=self,
    144 )
    145 self.reconnect_delay_current = self.comm_params.reconnect_delay or 0
    146 self.use_udp = False

File /usr/local/lib/python3.10/dist-packages/pymodbus/transaction/transaction.py:65, in TransactionManager.__init__(self, params, framer, retries, is_server, trace_packet, trace_pdu, trace_connect, sync_client)
     63     self._lock = asyncio.Lock()
     64     self.low_level_send = self.send
---> 65 self.response_future: asyncio.Future = asyncio.Future()

File /usr/lib/python3.10/asyncio/events.py:656, in BaseDefaultEventLoopPolicy.get_event_loop(self)
    653     self.set_event_loop(self.new_event_loop())
    655 if self._local._loop is None:
--> 656     raise RuntimeError('There is no current event loop in thread %r.'
    657                        % threading.current_thread().name)
    659 return self._local._loop

RuntimeError: There is no current event loop in thread 'MainThread'.

```

janiversen, 2024-12-20:
Hmmm sounds like a bug, it's strange though that 3.8.0 have the same problem, I will investigate.

closed 2024-12-20 by janiversen

farzadpanahi, 2024-12-20:
@janiversen tnx for the quick fix. do you think you will have a new release soon to include the fix?

farzadpanahi, 2024-12-20:
sorry, I think my browser is playing a trick on me:
[image]
I didn't see the new release. tnx

janiversen, 2024-12-20:
?? that is not the current release.

farzadpanahi, 2024-12-20:
I know, it was my browser showing the old releases data on the repo main page. looks like some caching issue either on my browser or gh! when I go to the releases page I can see the new release.

farzadpanahi, 2024-12-21:
@janiversen oh I see where the problem is. you forgot to move the `latest` tag : D

janiversen, 2024-12-21:
Which tag ? v3.8.2 exist as tag.

farzadpanahi, 2024-12-21:
[image]
It is not really a tag. Just a checkbox in release page.

janiversen, 2024-12-21:
It is a proper tag, check in git

[image]

Just not marked as latest release, which is something different.