pymodbus-dev/pymodbus #2519: No mixed type client.convert_from_registers() examples for migrating from BinaryPayloadDecoder.

jameshilliard, 2024-12-21:
I'm not seeing examples demonstrating how to convert registers with mixed types using `client.convert_from_registers()` while trying to migrate from `BinaryPayloadDecoder` to `client.convert_from_registers`, for example I have a lot of code decoding register groups with mixed type decoding using `BinaryPayloadDecoder` along these lines:
```python
decoder = BinaryPayloadDecoder.fromRegisters(
    data.registers,
    byteorder=Endian.BIG,
    wordorder=Endian.BIG,
)
manufacturer = decoder.decode_string(8).rstrip(b"\x00").decode("utf-8")
model = decoder.decode_string(32).rstrip(b"\x00").decode("utf-8")
hardware_version = decoder.decode_string(2).lstrip(b"\x00").decode("utf-8")
software_version = decoder.decode_16bit_uint()
protocol_version = (
    f"{decoder.decode_16bit_uint()}.{decoder.decode_16bit_uint()}"
)
compilation_time = decoder.decode_string(8)
rated_voltage = decoder.decode_16bit_uint()
rated_current = decoder.decode_16bit_uint()
rated_frequency = decoder.decode_16bit_uint()
outputs = decoder.decode_16bit_uint()
output_switch_function = bool(decoder.decode_16bit_uint())
output_current_function = bool(decoder.decode_16bit_uint())
output_degree_function = bool(decoder.decode_16bit_uint())
decoder.decode_16bit_uint()  # voltage
decoder.decode_16bit_uint()  # amps, not accurate, just p1
decoder.decode_16bit_uint()  # active power
decoder.decode_16bit_uint()  # reactive power
power_factor = decoder.decode_16bit_uint()
active_electrical_degree = decoder.decode_32bit_uint()
```

janiversen, 2024-12-21:
It's simple you cannot!

Mixed types are very advanced usage, and need to be solved at app level.

Please remember pymodbus implement the modbus protocol, the payload decoder have nothing to do with the protocol.

It surely helps when using mixed typed, but it is very old code that is quite hard to maintain, it needs a rewrite so we cannot continue maintaining it.

janiversen, 2024-12-21:
ps nobody blocks you from copying the code into your app and continue using it, you just do not get free support.

jameshilliard, 2024-12-21:
> It surely helps when using mixed typed, but it is very old code that is quite hard to maintain, it needs a rewrite so we cannot continue maintaining it.

Yeah, it's def a weird style, I'll maybe pull it into our codebase and see if I can come up with something a bit nicer for mixed type decoding and then see if that makes sense to upstream here.

janiversen, 2024-12-21:
ok.

closed 2024-12-21 by jameshilliard