pymodbus-dev/pymodbus #2530: Repl is broken

jondo2010, 2025-01-03:
### Versions

- Python: 3.12
- OS: Ubuntu 24.04
- Pymodbus: 3.8.2

### Description

Trying to use the repl.

### Code and Logs

```
$ pymodbus.console
Traceback (most recent call last):
  File "/home/jhughes/tmp/bin/pymodbus.console", line 5, in <module>
    from pymodbus_repl.client.main import main
  File "/home/jhughes/tmp/lib/python3.12/site-packages/pymodbus_repl/client/main.py", line 15, in <module>
    from pymodbus.transaction import (
ImportError: cannot import name 'ModbusAsciiFramer' from 'pymodbus.transaction' (/home/jhughes/tmp/lib/python3.12/site-packages/pymodbus/transaction/__init__.py)
```

janiversen, 2025-01-03:
Please file this issue in the repl repo https://github.com/pymodbus-dev/repl

As documented repl is not maintained as part of pymodbus.

closed 2025-01-03 by janiversen

dhoomakethu, 2025-01-04:
REPL requires pymodbus <= 3.7.

dhoomakethu, 2025-01-05:
This is being tracked here https://github.com/pymodbus-dev/repl/pull/26