pymodbus-dev/pymodbus #2539: ModbusSerialServer time delayed crash

flavius-t, 2025-01-10:
### Versions

- Python: 3.9.2
- OS: Debian GNU/Linux 11
- Pymodbus: 3.8.2
- Modbus Hardware (if used):
   - modbus master device reading from serial base with clients connected, via rs458 to usb cable (connected to modbus master's usb port).
   - modbus master is also running a modbus rtu server (described below), modbus master's rs485 port is wired back to the serial base, allowing the modbus rtu server to also be read as a client over usb (by the master itself).

### Pymodbus Specific

- Server: rtu- async
- Client: rtu - sync

### Description

Hi,

I've implemented an async modbus rtu server using `ModbusSerialServer`. This is being run inside a thread. After a period of time that varies (so far between 1-6 hours), clients can no longer read from the server. In the server thread, an exception related to `SerialTransport.intern_read_ready()` is repeatedly thrown. I included below the logs for the `SerialTransport.intern_read_ready()` error, and a log from one of the clients' failed reads.

I'm working on putting together some scripts for a simple demo to replicate the error, which I will include here. But for now I was wondering if there are any suggestions or ideas about where this error may be coming from?

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config("DEBUG")
```

```
Jan 09 23:46:03 osensa-hmi python3[65543]: 2025-01-09 23:46:03 +0000 : thread_id 548078088640 :    ERROR: base_events : Exception in callback SerialTransport.intern_read_ready()
Jan 09 23:46:03 osensa-hmi python3[65543]: handle: <Handle SerialTransport.intern_read_ready()>
Jan 09 23:46:03 osensa-hmi python3[65543]: Traceback (most recent call last):
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/usr/lib/python3.9/asyncio/events.py", line 80, in _run
Jan 09 23:46:03 osensa-hmi python3[65543]:     self._context.run(self._callback, *self._args)
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/transport/serialtransport.py", line 134, in intern_read_ready
Jan 09 23:46:03 osensa-hmi python3[65543]:     self.intern_protocol.data_received(data)  # type: ignore[attr-defined]
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/transport/transport.py", line 304, in data_received
Jan 09 23:46:03 osensa-hmi python3[65543]:     self.datagram_received(data, None)
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
Jan 09 23:46:03 osensa-hmi python3[65543]:     cut = self.callback_data(self.recv_buffer, addr=addr)
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/transaction/transaction.py", line 189, in callback_data
Jan 09 23:46:03 osensa-hmi python3[65543]:     used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/framer/base.py", line 76, in processIncomingFrame
Jan 09 23:46:03 osensa-hmi python3[65543]:     data_len, pdu = self._processIncomingFrame(data[used_len:])
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/framer/base.py", line 97, in _processIncomingFrame
Jan 09 23:46:03 osensa-hmi python3[65543]:     if (result := self.decoder.decode(frame_data)) is None:
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/pdu/decoders.py", line 112, in decode
Jan 09 23:46:03 osensa-hmi python3[65543]:     pdu.decode(frame[1:])
Jan 09 23:46:03 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/pdu/bit_message.py", line 26, in decode
Jan 09 23:46:03 osensa-hmi python3[65543]:     self.address, self.count = struct.unpack(">HH", data)
Jan 09 23:46:03 osensa-hmi python3[65543]: struct.error: unpack requires a buffer of 4 bytes
...
Jan 09 23:46:00 osensa-hmi python3[65543]: Traceback (most recent call last):
Jan 09 23:46:00 osensa-hmi python3[65543]:   File "/<path>/biz_logic/modbus/modbus_reads.py", line 227, in read_modbus_rtu
Jan 09 23:46:00 osensa-hmi python3[65543]:     regs, error = read_multiple_regs(
Jan 09 23:46:00 osensa-hmi python3[65543]:   File "/<path>biz_logic/modbus/modbus_reads.py", line 71, in read_multiple_regs
Jan 09 23:46:00 osensa-hmi python3[65543]:     response = get_response(
Jan 09 23:46:00 osensa-hmi python3[65543]:   File "/<path>/biz_logic/modbus/modbus_reads.py", line 48, in get_response
Jan 09 23:46:00 osensa-hmi python3[65543]:     return client.read_holding_registers(address=address, count=num_regs, slave=slave_id)
Jan 09 23:46:00 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
Jan 09 23:46:00 osensa-hmi python3[65543]:     return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
Jan 09 23:46:00 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/client/base.py", line 186, in execute
Jan 09 23:46:00 osensa-hmi python3[65543]:     return self.transaction.sync_execute(no_response_expected, request)
Jan 09 23:46:00 osensa-hmi python3[65543]:   File "/<path>/venv/lib/python3.9/site-packages/pymodbus/transaction/transaction.py", line 120, in sync_execute
Jan 09 23:46:00 osensa-hmi python3[65543]:     raise ModbusIOException(txt)
Jan 09 23:46:00 osensa-hmi python3[65543]: pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

janiversen, 2025-01-10:
The error in the server is because it receives a short packet (less than 4 bytes), which is not a legal modbus packet.

janiversen, 2025-01-10:
Your problem is most likely that you use the server on a multipoint line, something that is stated in the documentation to be unsupported!

The reason that we do not support it, is that pymodbus cannot reliably detect the 3.5char pause between frames because it does not have hardware access.

janiversen, 2025-01-10:
Pull requests are more than welcome if you think you can solve the detection problem.

farzadpanahi, 2025-01-10:
thanks for looking into this.

> Your problem is most likely that you use the server on a multipoint line,

you mean having specifically pymodbus server (slave) on multipoint modbus bus is not supported? 

because pymodbus client (master) on multipoint modbus bus works fine.

> something that is stated in the documentation to be unsupported!

is [this](https://pymodbus.readthedocs.io/en/latest/source/library/framer.html#module-pymodbus.framer.FramerRTU) the section you are referring to?

> The reason that we do not support it, is that pymodbus cannot reliably detect the 3.5char pause between frames because it does not have hardware access.

a while back I was looking at the frame delay issue in a different C lib. please take a look at these links:
https://github.com/debevv/nanoMODBUS/issues/29

https://github.com/stephane/libmodbus/commit/a11805cbfda23da36aed84bca3c5edc1413d664c#diff-6d1f2c1d14cf673b3d523523e9b2042ed1774fbcdd202b9de3d1a76050e5aa78R66
"The libmodbus implementation of RTU isn't time based as stated in original
Modbus specification, instead all bytes are sent as fast as possible and a
response or an indication is considered complete when all expected characters
have been received."

if I remember correctly in their implementation they ignore frame delay and try to look at all the incoming bytes and try to determine the frame based on which byte range makes sense. 

is this approach tested in pymodbus?

> Due to that the server only looks for requests and gets confused when it gets a response (sent from another device) which leads to your breakdown.

is [this](https://github.com/pymodbus-dev/pymodbus/pull/1435) the PR that tries to address this issue?

farzadpanahi, 2025-01-10:
for the record [here](https://pymodbus.readthedocs.io/en/latest/source/server.html#pymodbus.server.ModbusSerialServer:~:text=Warning%20The%20current%20framer%20implementation%20does%20not%20support%20running%20the%20server%20on%20a%20shared%20rs485%20line%20(multipoint).) is the relevant documentation warning: 
> Warning The current framer implementation does not support running the server on a shared rs485 line (multipoint).

flavius-t, 2025-01-10:
I will start a long term test of the modbus rtu server without the multpoint setup, to make sure it cannot be reproduced. I will report back.

janiversen, 2025-01-10:
Be my guest....I have a server running, that is now entering its second month, without restart.

Closing issue, as it is not a pymodbus problem.

closed 2025-01-10 by janiversen