pymodbus-dev/pymodbus #2542: DataStore offset

filippobrogi, 2025-01-21:
**Versions**

- python 3.10
- OS: ubuntu 22 x86_64
- pymodbus: v3.8.3

**Pymodbus speficic**
Using the sync modbus server  StartTcpServer

**Issue**
The client sends 2 requests to read data from starting address 30005 and 30055. These addresses are increased by one in pymodbus, I guess because of the zero norm, however I need to add an offset value to read the expected values on the client side.

`        self.__byte_order = Endian.BIG
        offset = 2
        self.registers[RegisterID.VOLTAGE_L1N] = Float32Register(self, 30005 + offset, Scale.X1)
        self.registers[RegisterID.VOLTAGE_L2N] = Float32Register(self, 30007 + offset, Scale.X1)
        self.registers[RegisterID.VOLTAGE_L3N] = Float32Register(self, 30009 + offset, Scale.X1)
        self.registers[RegisterID.VOLTAGE_L1L2] = Float32Register(self, 30011 + offset, Scale.X1)
        self.registers[RegisterID.VOLTAGE_L2L3] = Float32Register(self, 30013 + offset, Scale.X1)
        self.registers[RegisterID.VOLTAGE_L3L1] = Float32Register(self, 30015 + offset, Scale.X1)
        self.registers[RegisterID.CURRENT_L1] = Float32Register(self, 30017 + offset, Scale.X1)
        self.registers[RegisterID.CURRENT_L2] = Float32Register(self, 30019 + offset, Scale.X1)
        self.registers[RegisterID.CURRENT_L3] = Float32Register(self, 30021 + offset, Scale.X1)
        self.registers[RegisterID.POWER_APPARENT_L1] = Float32Register(self, 30023 + offset, Scale.X1)
        self.registers[RegisterID.POWER_APPARENT_L2] = Float32Register(self, 30025 + offset, Scale.X1)
        self.registers[RegisterID.POWER_APPARENT_L3] = Float32Register(self, 30027 + offset, Scale.X1)
        self.registers[RegisterID.POWER_APPARENT_SUM] = Float32Register(self, 30055 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REAL_L1] = Float32Register(self, 30029 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REAL_L2] = Float32Register(self, 30031 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REAL_L3] = Float32Register(self, 30033 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REAL_SUM] = Float32Register(self, 30057 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REACTIVE_L1] = Float32Register(self, 30035 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REACTIVE_L2] = Float32Register(self, 30037 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REACTIVE_L3] = Float32Register(self, 30039 + offset, Scale.X1)
        self.registers[RegisterID.POWER_REACTIVE_SUM] = Float32Register(self, 30059 + offset, Scale.X1)
        self.registers[RegisterID.FREQUENCY] = Float32Register(self, 30047 + offset, Scale.X1)
`

The register values are updated using the following method 

`    def set_float32(self, address: int, value: float):
        """Assign the float value to the provided address

        Args:
            address (int): the address
            value (float): the value
        """
        builder = BinaryPayloadBuilder(byteorder=self.__byte_order)
        builder.add_32bit_float(value)
        payload = builder.to_registers()`

This is the related logging. I faced the same issue when using older version of the library. 

`2025-01-21 07:57:27,980.980 (transport.py:329) [DEBUG] recv: 0x0 0x12 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x35 0x0 0x24 old_data:  addr=None
2025-01-21 07:57:27,980.980 (base.py:91) [DEBUG] Processing: 0x0 0x12 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x35 0x0 0x24
2025-01-21 07:57:27,980.980 (decoders.py:113) [DEBUG] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersRequest(dev_id=0, transaction_id=0, address=30005, count=36, bits=[], registers=[], status=1) 
2025-01-21 07:57:27,980.980 (base.py:101) [DEBUG] Frame advanced, resetting header!!
2025-01-21 07:57:27,981.981 (context.py:120) [DEBUG] validate: fc-[3] address-30006: count-36
2025-01-21 07:57:27,981.981 (context.py:132) [DEBUG] getValues: fc-[3] address-30006: count-36
2025-01-21 07:57:27,982.982 (transport.py:386) [DEBUG] send: 0x0 0x12 0x0 0x0 0x0 0x4b 0x1 0x3 0x48 0x43 0x66 0x0 0x0 0x43 0x66 0x0 0x0 0x43 0x66 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3f 0xe2 0xfd 0xa8 0x40 0x62 0xfd 0xa8 0x40 0xaa 0x3e 0x3e 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3f 0x99 0xe4 0x61 0x40 0x19 0xe4 0x61 0x40 0x66 0xd6 0x91 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2025-01-21 07:57:28,004.4 (transport.py:329) [DEBUG] recv: 0x0 0x13 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x67 0x0 0x6 old_data:  addr=None
2025-01-21 07:57:28,004.4 (base.py:91) [DEBUG] Processing: 0x0 0x13 0x0 0x0 0x0 0x6 0x1 0x3 0x75 0x67 0x0 0x6
2025-01-21 07:57:28,005.5 (decoders.py:113) [DEBUG] decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersRequest(dev_id=0, transaction_id=0, address=30055, count=6, bits=[], registers=[], status=1) 
2025-01-21 07:57:28,005.5 (base.py:101) [DEBUG] Frame advanced, resetting header!!
2025-01-21 07:57:28,006.6 (context.py:120) [DEBUG] validate: fc-[3] address-30056: count-6
2025-01-21 07:57:28,006.6 (context.py:132) [DEBUG] getValues: fc-[3] address-30056: count-6
2025-01-21 07:57:28,006.6 (transport.py:386) [DEBUG] send: 0x0 0x13 0x0 0x0 0x0 0xf 0x1 0x3 0xc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0`

Thanks a lot

janiversen, 2025-01-21:
The pymodbus client uses the address you supply directly, there are no zero regulation.

However a lot of devices start with address 1 in the manual, but expect address 0 in the message.

janiversen, 2025-01-21:
Some of the data stores use an offset of 1, not the server. E.g. the simulator datastore do not add an offset.

but what is the problem (remembering a lot of devices do the same) ?

janiversen, 2025-01-23:
Your assumption is correct, but it seems wrong to change it in the client, it would be more logical to change your datastore setup.

closed 2025-01-23 by janiversen

filippobrogi, 2025-01-23:
Ok I know that but can you explain why I need to misconfigure the dataset to provide the right data to a client? For me it sounds like a bug when I request the content of 30006 and I get back the content of 30007, right?

janiversen, 2025-01-23:
No that is not a bug....that is a difference in how you DEFINE the data in the server and how you REQUEST the data from the client. As I have explained there are devices that do addressing as 1 -> 0, and others do 0 -> 0.

The old datastores follow the 1->0 addressing scheme, the new datastore will follow the 0->0 scheme. Please see #2535 

If you want the code to work differently, then feel free to submit a pull request, but please be aware that it would be nice that it does not break all current setups.