pymodbus-dev/pymodbus #2544: Incorrect Data Conversion in pymodbus 3.8.3

fineskorohodov, 2025-01-22:
Versions

python 3.12
OS: windows 11 x64
pymodbus: v3.8.3

Pymodbus speficic
Using the sync modbus server StartTcpServer

Issue

After upgrading pymodbus from version 3.7.4 to 3.8.3, incorrect behavior is observed when converting data from registers to FLOAT32. The previously used method BinaryPayloadDecoder.fromRegisters returned correct results, but the new method _client.convert_from_registers produces an incorrect result.

Code used in version 3.7.4:

`from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian

(response = ModbusTcpClient.read_holding_registers(address=_address, count=_count, slave=_slave_id, no_response_expected=False)  # return [36543, 31508]
)

registers = [36543, 31508]

decoder = BinaryPayloadDecoder.fromRegisters(
    registers,
    byteorder=Endian.LITTLE,
    wordorder=Endian.BIG
)
decoded_value = decoder.decode_32bit_float()
print(decoded_value)  # return -1.1100000143051147`

Code in version 3.8.3:
`
from pymodbus.client import ModbusTcpClient

(response = ModbusTcpClient.read_holding_registers(address=_address, count=_count, slave=_slave_id, no_response_expected=False)  # return [36543, 31508])

registers = [36543, 31508]

decoder = ModbusTcpClient.convert_from_registers(
    registers=registers,
    data_type=ModbusTcpClient.DATATYPE.FLOAT32,
    word_order="little"
)
print(decoder)  # Возвращает: 7.713551698059133e+35
`

Expected Behavior
The result should be -1.1100000143051147, as returned when using version 3.7.4 with the BinaryPayloadDecoder.fromRegisters method. (rounded -1.1)

Actual Behavior
The result from _client.convert_from_registers in version 3.8.3 is 7.713551698059133e+35, which is incorrect.

Suggestion
The issue seems related to changes in the interpretation of byte/word order in _client.convert_from_registers. It may require reviewing the implementation of byte/word order handling or updating the documentation to reflect the new behavior.

Contact
If additional information or examples are required, I am happy to provide them.

janiversen, 2025-01-23:
That might be a bug, I will take a look later.

janiversen, 2025-01-23:
Actually this is not a bug, registers are pr protocol definition big endian, which is why the convert functions do not have a byte_order parameter.

Word order is not part of the protocol definition, the "de facto" standard is to use big endian, however some devices do not, so the convert functions have a word_order parameter.

If you have a device that uses byte order == little endian, then the device does not follow the modbus standard and are thus not supported by pymodbus. That being said it is quite easy for you to do something like:
```
    register = int.from_bytes(register.to_bytes(2, "little"))
```

Closing as this is not a pymodbus bug.