pymodbus-dev/pymodbus #2555: Specify encoding when decoding registers to string

Breina, 2025-01-28:
When calling `convert_from_registers` with `data_type=DATATYPE.STRING`, the encoding `utf-8` is used.

Please parameterize this encoding so that it works with other text encoding formats (such as latin-1 in my case).

janiversen, 2025-01-30:
Pull requests are welcome, we use utf-8 because it is the most generic one, and please be aware that it is actually not really used, since the byte value is used directly.

janiversen, 2025-01-30:
Please be aware you are not forced to use the convert functions, you can easily do it in the app.

Btw saw the app is actually a custom component of HA, which means you are using a pretty old version of pymodbus (due to the component modbus) where among others the convert functions work quite differently.

Breina, 2025-01-30:
Thanks for the quick response!

I used `BinaryPayloadDecoder.decode_string` before and parsed that with python's `bytes.decode`. The warning message on `BinaryPayloadDecoder` advised me to use `convert_from_registers`, which did feel pretty clean.

My user has a starting byte of 0x80, which isn't ASCII nor UTF-8, so I need latin-1 extended ASCII specifically. I can create a PR for this.

>Btw saw the app is actually a custom component of HA, which means you are using a pretty old version of pymodbus

Not exactly sure how it works, but I can specify a required version of pymodbus and it will get that for me. I'm running on the latest version currently.

[image]

janiversen, 2025-01-30:
Interesting, if you can use multiple versions of the same lib. Did you actually check either pip what is installed ?

The python builtin decode function needs a character type like utf-8, if there is a "raw" or similar I quest that would be the best change.....we do our best to avoid extra parameters.

janiversen, 2025-01-30:
Just checked someone updated ha pymodbus version to 3.8.3, which is why you do not have problems.

janiversen, 2025-01-30:
A better solution would be to add "errors='ignore'" to the decode call, which should solve all your problems.

Breina, 2025-01-30:
I don't think this would solve my problem. I need the `0x80` byte to be decoded to `Ç`.

If you don't like the parameter, I can parse the bytes outside of your lib as you said, but I felt it would be cleaner to provide a way to configure it on your end.

janiversen, 2025-01-30:
A parameter with a correct default, would be accepted, please remember to add test cases (e.g. what happens if parameter="raw").

janiversen, 2025-02-09:
Solved.

closed 2025-02-09 by janiversen