pymodbus-dev/pymodbus #2556: Exceptions during decoding - struct.error: unpack requires a buffer of 4 bytes

Ambadrant, 2025-01-28:
# Original issue 

During the use of a serial server using the RTU framer I encountered the following exception:
```
Exception in callback SerialTransport.intern_read_ready()
  handle: <Handle SerialTransport.intern_read_ready()>
  Traceback (most recent call last):
    File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
      self._context.run(self._callback, *self._args)
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/transport/serialtransport.py", line 134, in intern_read_ready
      self.intern_protocol.data_received(data)  # type: ignore[attr-defined]
      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 304, in data_received
      self.datagram_received(data, None)
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
      cut = self.callback_data(self.recv_buffer, addr=addr)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 191, in callback_data
      used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/framer/base.py", line 76, in processIncomingFrame
      data_len, pdu = self._processIncomingFrame(data[used_len:])
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/framer/base.py", line 97, in _processIncomingFrame
      if (result := self.decoder.decode(frame_data)) is None:
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/pdu/decoders.py", line 112, in decode
      pdu.decode(frame[1:])
    File "/usr/local/share/virtualenvs/datf/lib/python3.12/site-packages/pymodbus/pdu/register_message.py", line 26, in decode
      self.address, self.count = struct.unpack(">HH", data)
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
  struct.error: unpack requires a buffer of 4 bytes
```

I only used the RTU framer, however I guess these kind of issues might also be present in the other framers.

# Reproducing the issue

I tried to reproduce this issue by running a simple program like this multiple times:

```python
from pymodbus.framer.rtu import FramerRTU
from pymodbus.pdu.decoders import DecodePDU
import random

framer = FramerRTU(DecodePDU(True))

for len in range(1, 20):
    for try_number in range(10000):
        data = random.randbytes(len)
        print(f"Trying data: {data}")
        framer.processIncomingFrame(data)
```

So far I found two payloads which cause similar exceptions:

```python
from pymodbus.framer.rtu import FramerRTU
from pymodbus.pdu.decoders import DecodePDU

data = b'\xd2\xd5\xd5\x02t%\x7fa\x87%\x03\xfad\xd5\xfe'
framer = FramerRTU(DecodePDU(True))
framer.processIncomingFrame(data)

# Traceback (most recent call last):
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/test.py", line 6, in <module>
#     framer.processIncomingFrame(data)
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/framer/base.py", line 76, in processIncomingFrame
#     data_len, pdu = self._processIncomingFrame(data[used_len:])
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/framer/base.py", line 97, in _processIncomingFrame
#     if (result := self.decoder.decode(frame_data)) is None:
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/pdu/decoders.py", line 112, in decode
#     pdu.decode(frame[1:])
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/pdu/bit_message.py", line 27, in decode
#     self.address, self.count = struct.unpack(">HH", data)
# struct.error: unpack requires a buffer of 4 bytes
```

```python
from pymodbus.framer.rtu import FramerRTU
from pymodbus.pdu.decoders import DecodePDU

data = b'\xd7C\xa55\xd7\xc1\x18\x8e{m\x9cG'
framer = FramerRTU(DecodePDU(True))
framer.processIncomingFrame(data)

# Traceback (most recent call last):
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/test.py", line 6, in <module>
#     framer.processIncomingFrame(data)
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/framer/base.py", line 76, in processIncomingFrame
#     data_len, pdu = self._processIncomingFrame(data[used_len:])
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/framer/base.py", line 97, in _processIncomingFrame
#     if (result := self.decoder.decode(frame_data)) is None:
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/pdu/decoders.py", line 112, in decode
#     pdu.decode(frame[1:])
#   File "/cygdrive/c/Users/treichet/home/w/repos/3rdparty/pymodbus/pymodbus/pdu/bit_message.py", line 27, in decode
#     self.address, self.count = struct.unpack(">HH", data)
# struct.error: unpack requires a buffer of 4 bytes
```

# Investigation 

As far as I understand `processIncomingFrame` does the following things:
1. calls FramerRTU.decode (this decodes the ADU and returns the raw PDU to step 2) (the actual root cause might be here)
2. calls DecodePDU.decode (this calls the correct decoder based on the PDU type)
    - here the exception occurs because the PDU decoder does not get enough data to decode

# Suggested Fixes

The easy way would be to change the ModbusPDU.decode() API to return a status for success/failure and handle the exceptions in the decode calls.
However, the real issue might be in FramerRTU.decode():

```
    class FramerRTU(FramerBase):
    
    ...

    def decode(self, data: bytes) -> tuple[int, int, int, bytes]:
        """Decode ADU."""
        data_len = len(data)
        for used_len in range(data_len):
            if data_len - used_len < self.MIN_SIZE:
                Log.debug("Short frame: {} wait for more data", data, ":hex")
                return 0, 0, 0, self.EMPTY
            dev_id = int(data[used_len])
            if not (pdu_class := self.decoder.lookupPduClass(data[used_len:])):
                continue
          
            # maybe this size calculation here is not correct, for all cases
            if not (size := pdu_class.calculateRtuFrameSize(data[used_len:])):
                size = data_len +1
            if data_len < used_len +size:
                Log.debug("Frame - not ready")
                return 0, dev_id, 0, self.EMPTY
            
            # this loop here, looks strange to me. Shouldn't we know the exact size here 
            # and check the CRC according to this known size ?
            for test_len in range(data_len, used_len + size - 1, -1):
                start_crc = test_len -2
                crc = data[start_crc : start_crc + 2]
                crc_val = (int(crc[0]) << 8) + int(crc[1])
                if not FramerRTU.check_CRC(data[used_len : start_crc], crc_val):
                    Log.debug("Frame check failed, possible garbage after frame, testing..")
                    continue
                return start_crc + 2, dev_id, 0, data[used_len + 1 : start_crc]
        return 0, 0, 0, self.EMPTY
```

# Testing

I would suggest to add some fuzzy tests for the decoders, to catch such issues.

-----
Sorry, that I can not give more detailed feedback or better suggestions for fixes right now. 
I neither know modbus, nor this code base well enough to make an actual PR.
I hope this issue is helpful. If you want me to do some more work in regards to this issue, let me know :)

janiversen, 2025-01-30:
Forgot to mention, the issue you describe is very common if you use the server on a multidrop line (multiple slaves), that is pr documentation not supported.

Ambadrant, 2025-01-30:
For me the issue is, that the server will just crash and as a user of the server I can't really do anything about it.
The decoder could just ignore such invalid frames, instead of throwing an exception. As far as I understand, it does so for other invalid frames, for example: frames with invalid length or checksum. It just doesn't seem to cover all edge cases.

janiversen, 2025-01-30:
Well problem is that the structure of the modbus protocol makes it impossible to catch all edge cases. The base problem is that modbus defines a delay of 3.5 char between messages....but after rs485 converter, drivers etc, this is. not reliable and so far we have found no other algorithm that actually worked.

An exception is thrown when something fundamental is wrong, which seem the right thing to do....and it is normally not a problem when the server is used for test purposes as defined.

janiversen, 2025-01-30:
But as usual if you want a change then fell free to submit a pull request. Since this is a very sensitive part of the library, please add tests to show the correct behavior.

Ambadrant, 2025-01-30:
Okay, I will try to look further into it. I need to read more about modbus first, maybe I just don't understand the protocol correctly.

closed 2025-01-30 by janiversen