pymodbus-dev/pymodbus #2559: [Bug] Off by one error datablock and slavecontext

nee2c, 2025-01-30:
Hello

When creating a data block and slave context there is a off by one error.

```
from pymodbus.datastore.store import ModbusSequentialDataBlock
from pymodbus.datastore.context import ModbusSlaveContext

_fx = {"co": 1, "di": 2, "hr": 3, "ir": 4}
mbd = ModbusSequentialDataBlock(0, [1, 2, 3, 4])
mbc = ModbusSlaveContext(di=mbd)

print(mbc.getValues(_fx["di"], 0, 2))
#prints "[2, 3]"
```

So basic I think the first value is going unused and is not addressable.

I think there are two ways of solving this

1. delete the lines such as [L131](https://github.com/pymodbus-dev/pymodbus/blob/4d56e0492c88ccd1bea98dd60b7520b8039b6a2a/pymodbus/datastore/context.py#L131) I think this change will break a fair amount of code

[image]

2. When data block is initialized or it gets the address of 0 bump it to 1

janiversen, 2025-01-31:
That is not a bug, that datastore is configured with address == 1 which corresponds to request address == 0

you might see it as strange, but it is not documented anywhere that the address used for configuration and request are the same and there are historic reasons for doing so.

Be aware the newer data stores start with 0.

janiversen, 2025-01-31:
Your example is very false DI is direct input, which is a coil, those are not addressed as registers, please see the modbus documentation.

janiversen, 2025-01-31:
If you do not want the offset use a new datastore like e.g the simulator.

closed 2025-01-31 by janiversen