pymodbus-dev/pymodbus #2562: Positional argument error in RemoteSlaveContext for modbus client read functions

hoesentoet, 2025-02-05:
When using the `RemoteSlaveContext` and trying to read any modbus register, a `TypeError` occures:
```
File ".venv\Lib\site-packages\pymodbus\datastore\remote.py", line 82, in <lambda>
    "h": lambda a, c: self._client.read_holding_registers(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 were given
```

pymodbus version: `3.8.3`

## Example to reproduce error
This is a minimal code example to reproduce this error:
```python
import asyncio
import logging

from pymodbus.client import ModbusSerialClient
from pymodbus.datastore import ModbusServerContext
from pymodbus.datastore.remote import RemoteSlaveContext
from pymodbus.server import StartAsyncTcpServer

logging.basicConfig()
_logger = logging.getLogger()
_logger.setLevel(logging.DEBUG)

async def run_forwarder():
    """Run forwarder setup."""
    rtu_port = "COM4"
    tcp_port = "5020"

    txt = f"### start forwarder, listen {tcp_port}, connect to {rtu_port}"
    _logger.info(txt)

    client = ModbusSerialClient(
        port=rtu_port,
        baudrate=38400,
        timeout=1,
        stopbits=2,
        bytesize=8,
        parity='N'
    )
    client.connect()
    assert client.connected

    store = RemoteSlaveContext(client)
    context = ModbusServerContext(slaves=store, single=True)

    await StartAsyncTcpServer(context=context, address=("localhost", tcp_port))
    # loop forever

if __name__ == "__main__":
    asyncio.run(run_forwarder())
```

## Possible fix
This error can be fixed by simply adding `count=` in the lambda function of `RemoteSlaveContext.__build_mapping()`:

(`pymodbus/datastore/remote.py`, `line 75`)
```python
self.__get_callbacks = {
    "d": lambda a, c: self._client.read_discrete_inputs(
        a, count=c, **params
    ),
    "c": lambda a, c: self._client.read_coils(
        a, count=c, **params
    ),
    "h": lambda a, c: self._client.read_holding_registers(
        a, count=c, **params
    ),
    "i": lambda a, c: self._client.read_input_registers(
        a, count=c, **params
    ),
}
```

janiversen, 2025-02-05:
Reported by another user and solved on dev.

janiversen, 2025-02-09:
Solved

closed 2025-02-09 by janiversen