pymodbus-dev/pymodbus #2565: Modbus broken since 3.6.9

jomach, 2025-02-08:
Hello Together, 

I think there is a big issue on pymodbus from version 3.7.0+. Following script works fine on 3.6.9: 

```
from pymodbus.client import ModbusTcpClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian
import logging

# Enable logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusTcpClient('192.168.1.200', port='4196')
client.connect()

# Read input register at address 0 (30001 corresponds to address 0)
result = client.read_input_registers(0, count=2)

if not result.isError():
    # Decode the register values to float32
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers)
    value = decoder.decode_32bit_float()
    value = round(value, 2)  # Apply precision
    print(f"Wallbox Phase 1 Line to Neutral Volts: {value} V")
else:
    print("Error reading input registers")

client.close()
```

The main problem is that Homeassistant uses your library and It's a huge problem... See https://github.com/home-assistant/core/issues/137765

can someone help ?

janiversen, 2025-02-08:
You do not write what the problem is, but I guess you did not update your script to the v3.7 standard.

Please be aware that v3.7.x is very old, current version is v3.8.3 and we cannot help you solve problems in old releases.

And just to be clear, I do not care what version homeassistant (but I happen to know it uses v3.8.3) uses, that is another project.

jomach, 2025-02-08:
Hello @janiversen, sorry by bad. It is on the other ticket from ha. 

```
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

here is the error

janiversen, 2025-02-08:
That because you device do not respond, as I wrote did you update your script to the v3.7 standard.

I do not follow HA, and when someone, without checking, cc me it's ignored.

janiversen, 2025-02-08:
Closing this for a couple of reasons:
- the example program will probably not work on any version beginning with v3.7.0, it needs to be updated
- the error reported (device not responding) is probably correct, since the example uses default parameters
- we do not look for bugs in old version so please update to v3.8.4 and see if the problem continues

closed 2025-02-08 by janiversen