pymodbus-dev/pymodbus #2575: convert_from_registers does not support INT8

PaulKGrimes, 2025-02-10:
I have a piece of equipment that reports certain information as a register containing two INT8s.  Using the BinaryPayloadDecoder I could do this conversion directly.  However, the new convert_from_registers and DATATYPE enum does not define INT8, and so I can no longer directly decode this information.

PaulKGrimes, 2025-02-10:
I can hack this using 
```            
temp = self._client.convert_from_registers(r, data_type=self._client.DATATYPE.INT16, word_order='little')
c = (temp >> 8) & 0xff
f = temp & 0xff
return c, f
```
But the loss of the previous functionality is annoying.

janiversen, 2025-02-10:
Just use .to_bytes() on the register, it is a python standard function.

It might be annoying to you, but then you should help with the maintenance and not just complain. We, the maintainers have more than enough to do, without features that are better done with standard functions.

janiversen, 2025-02-10:
Btw why do you use wordorder= when only reading a single register (word) that makes no sense.

closed 2025-02-10 by janiversen

PaulKGrimes, 2025-02-10:
> It might be annoying to you, but then you should help with the maintenance and not just complain. We, the maintainers have more than enough to do, without features that are better done with standard functions.

This response is incredibly dismissive and borderline abusive to your users. Do better, or resign as a maintainer.

MAKOMO, 2025-02-10:
> This response is incredibly dismissive and borderline abusive to your users.

I don't agree. Nobody is forced to use this lib. If one finds something that needs improvement one should actively contribute via PRs. Feature requests do count as rather minor contributions to a project.

I am quite happy that Jan is spending all this work as maintainer. Without him this project might just stall.

Anyhow, at any point you can fork your own and take over its maintenance.