pymodbus-dev/pymodbus #2577: trace_packet not called for incoming frames in synchronous client

wasilukm, 2025-02-15:
Considering the following sample code:
```
from pymodbus.client import ModbusTcpClient

def my_trace_packet(sending: bool, data: bytes) -> bytes:
    print(sending, data)
    return data

client = ModbusTcpClient('1.2.3.4', trace_packet=my_trace_packet)

client.read_holding_registers(0x100)
```
Observed:
my_trace_packet function is called for outcoming frames but is not called for received frames.

Expected:
my_trace_packet called both for outcoming and incoming frames

Probably line https://github.com/pymodbus-dev/pymodbus/blob/46aa4e82044eeb68ba44ca3bab070a14a020f007/pymodbus/transaction/transaction.py#L87 should be
> used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, databuffer))

Most likely execution of trace_pdu is also missing.

Found in pymodbus==3.8.6

janiversen, 2025-02-15:
pull requests are welcome.

closed 2025-02-16 by janiversen

wasilukm, 2025-02-16:
@janiversen thank you for the quick fix. Do you plan to release 3.8.7 or the fix will be in 3.9.0.

janiversen, 2025-02-16:
Right now I cannot say, it depends on if a major problem is found before 3.9.0