pymodbus-dev/pymodbus #2580: asyncio.exceptions.InvalidStateError: invalid state

jameshilliard, 2025-02-20:
Seeing this invalid state error [here](https://github.com/pymodbus-dev/pymodbus/blob/v3.8.3/pymodbus/transaction/transaction.py#L183), guessing we need to check if the future already has a result set?:
```python
Exception response 254 / 0
Exception in callback SerialTransport.intern_read_ready()
handle: <Handle SerialTransport.intern_read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/events.py", line 88, in _run
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/serialtransport.py", line 134, in intern_read_ready
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 304, in data_received
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
  File "/usr/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
asyncio.exceptions.InvalidStateError: invalid state
```

janiversen, 2025-02-20:
Please make a full debug log, it would be nice to understand how this happens.

It can happen if the client do not follow protocol and sends multiple requests without waiting for response.

Anyhow pull request are welcome.

jameshilliard, 2025-02-20:
Hmm, with debug logging enabled there's a lot of this spam in the logs coming from [here](https://github.com/pymodbus-dev/pymodbus/blob/v3.8.3/pymodbus/framer/rtu.py#L120) from the looks of it. It looks like some buffer maybe isn't getting cleared properly?

```python
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
ERROR:    Exception response 255 / 0
DEBUG:    Frame advanced, resetting header!!
ERROR:    Exception in callback SerialTransport.intern_read_ready()
handle: <Handle SerialTransport.intern_read_ready()>
Traceback (most recent call last):
  File "/usr/lib/python3.12/asyncio/events.py", line 88, in _run
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/serialtransport.py", line 134, in intern_read_ready
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 304, in data_received
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
  File "/usr/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
asyncio.exceptions.InvalidStateError: invalid state
DEBUG:    send: 0x1b 0x3 0x0 0x3a 0x0 0x30 0x67 0xe9
DEBUG:    recv: 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0x [... cut ...]
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
DEBUG:    Frame check failed, possible garbage after frame, testing..
```

jameshilliard, 2025-02-20:
Filtering out the frame check failed spam gives this:
```python
DEBUG:    recv: 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff old_data:  addr=None
DEBUG:    Processing: 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xfc 0xff 0xfe 0xff 0xfe 0xff 0xfe 0xff 0x [... cut ...]
  File "/usr/lib/python3.12/asyncio/events.py", line 88, in _run
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/serialtransport.py", line 134, in intern_read_ready
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 304, in data_received
  File "/usr/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
  File "/usr/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 183, in callback_data
asyncio.exceptions.InvalidStateError: invalid state
```

janiversen, 2025-02-21:
Your client is sending multiple requests without waiting for a response, which is not protocol conforming, unless the server is on a multipoint line, which is not supported

janiversen, 2025-02-21:
A thought are you trying to write registers leading to a packet bigger than 255 bytes.

jameshilliard, 2025-02-21:
> Your client is sending multiple requests without waiting for a response, which is not protocol conforming

Hmm, from my understanding there is no specific modbus slave reply timeout standard for modbus. These requests(if I'm tracing things right) are being wrapped in `asyncio.wait_for` with a 1 second timeout(so the request I think should be getting cancelled after 1 second without a valid reply) and have a surrounding lock to ensure multiple requests aren't in flight at the same time. I guess I should probably try and write a minimal reproducer to get a better idea of what specific conditions are triggering this since the debug logs are a bit too verbose to interpret easily.

> Of course it might also happen if your client sends random hex bytes, which seems to be the case here.

This looks to me like it's an issue with the response processing, not sure where you're seeing the client send random hex bytes.

This send in the logs looks normal to me at least:
```python
send: 0x1b 0x3 0x0 0x3a 0x0 0x30 0x67 0xe9
```

> A thought are you trying to write registers leading to a packet bigger than 255 bytes.

I think these are coming from `read_holding_registers` response processing or something. Maybe it's a slave sending weird responses or some other data corruption mucking with the response parser?

> Anyhow a simple solution would be to check the future and raise an exception, because clearly the client needs to be stopped.

Maybe what's going on here is that on a request cancellation the recv buffer needs to be purged(i.e. reset the client recv parser state)? I'm not sure why the recv buffer seems to be persisting across independent requests.

janiversen, 2025-02-21:
Ahh I got confused, I thought you talked about server not client.

pymodbus contains its own timeout mechanism, which are supposed to clear the buffer, however if the app cancels the request this do not happen. You should not need a wait in the app, but instead specify timeout.

jameshilliard, 2025-02-21:
> pymodbus contains its own timeout mechanism, which are supposed to clear the buffer, however if the app cancels the request this do not happen.

Yeah, that makes sense, where in the code is the buffer getting cleared exactly? I'm thinking it should be possible to catch the cancellation and clear the buffer when externally cancelling the request as well. I'll see if I can come up with a good solution for making pymodbus cancellation safe.

> You should not need a wait in the app, but instead specify timeout.

It's a bit tricky in that the global client timeout needs to be long enough for all requests to succeed, however we kind of have to override the timeout for some specific requests otherwise certain operations(such as operations probing a bunch of addresses for the presence of devices) will be far too slow.

jameshilliard, 2025-02-21:
> pymodbus contains its own timeout mechanism, which are supposed to clear the buffer,

Looking at the transaction code in more detail this doesn't appear to actually be how the current code behaves, the buffer only appears to get cleared on client close at the moment(I fully inhibit client close on request timeout which likely makes this issue more severe than would be typical), I think we need to do something like #2581 so that we clear the buffer on timeouts as well so that we have an empty recv buffer when re-sending the request.

PR #2582 (2025-02-23): Don't set_result on completed futures.
We need to guard self.response_future.set_result with a `done()` check to prevent `asyncio.exceptions.InvalidStateError: invalid state` errors.

Similar to #2581 I think we also need to flush the `recv_buffer` if the transaction is cancelled by the caller(say by the caller executing a transaction inside of another `asyncio.wait_for` that times out before the `asyncio.wait_for` inside of the `execute` call), in this case however we should re-raise the exception after the flush so that no retries are attempted.

Since `self.response_future` is cancelled in  [... cut]
closed 2025-02-23 by janiversen