pymodbus-dev/pymodbus #2585: Communication issue with Newer pymodbus version.

vivekgourav, 2025-02-24:
The communication seems to work fine in 3.7.4 version (Writing into coil /reading input registers) at 50hz(also tested at different speeds). However, there is an issue with version 3.8.0 and above, where the data isn't continuous (both reading /writing), after a couple of seconds it completely looses connection with the slave.
Throws the following error:
"No response received after 3 retries, continue with next request"  

Do you have any idea why this is happening ?

janiversen, 2025-02-24:
The typical error is that your device do not respond. Are you sure you address it correctly?

Without sample code and a debug log (as the issue template asks for), we cannot help you.

vivekgourav, 2025-02-25:
The device responds for few seconds and then the modbus fails. THis happens only on the newer versions and works fine with 3.7.4
Here is the error "
Slave 10: Coils set successfully with values [True, False, True, False, True, False, True, False].
Slave 10: Coils read successfully with values [True, False, True, False, True, False, True, False].
Exception in Slave 10: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
No response received after 3 retries, continue with next request
No response received after 3 retries, continue with next request
Exception in Slave 10: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request"

# Modbus communication settings
COM_PORT = '/tmp/ttyUR'
#COM_PORT = '/dev/ttyUSB0'
#BAUD_RATE = 460800
BAUD_RATE = 115200
#BAUD_RATE = 115200
TIMEOUT = 1# Seconds

 try:
            # Step 1: Generate dynamic coil values
            coil_values = [(self.total_packets[slave_id] + i) % 2 == 0 for i in range(coil_count)]

            # Write to coils
            write_result = self.client.write_coils(coil_address, coil_values, slave=slave_id)
            #time.sleep(0.001)
            if write_result.isError():
                print(f"Error writing coils to Slave {slave_id}")
                return

            print(f"Slave {slave_id}: Coils set successfully with values {coil_values}.")

janiversen, 2025-03-01:
???

janiversen, 2025-03-03:
Seems this is solved.

closed 2025-03-03 by janiversen