pymodbus-dev/pymodbus #2590: ModbusSequentialDataBlock off by one?? Swallowing element [0] ??

eabinga, 2025-02-28:
I am quite new to pyModbus, actually first day working with it.

I have the following code which doesn't return the result I am expecting. Perhaps I am expecting the wrong thing.

```
from pymodbus.datastore import ModbusSlaveContext, ModbusSequentialDataBlock

store = ModbusSlaveContext( 
    hr = ModbusSequentialDataBlock(0, list(range(1, 11)))
  )  # 10 Holding Registers initialized to 10

register_values = store.getValues(3, 0, 10)  # Function Code 3 = Holding Registers

print(register_values) # Should print [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]

print(f"Available Registers: {len(register_values)}")  # Should print 10
```
When I run this, I get this result

```
python main.py 
[2, 3, 4, 5, 6, 7, 8, 9, 10]
Available Registers: 9
```

And I actually expect this:

```
python main.py 
[1, 2, 3, 4, 5, 6, 7, 8, 9, 10]
Available Registers: 10
```

Is it just me?

eabinga, 2025-02-28:
I just saw someone with a similar issue that was not declared a bug: (#2559) and this was a reply.

I am just trying to understand what it means.

```
That is not a bug, that datastore is configured with address == 1 which corresponds to request address == 0

you might see it as strange, but it is not documented anywhere that the address used for configuration and request are the same and there are historic reasons for doing so.
```

Does this mean that register_value[1] is actually returned for request for 0  per spec, and that register_value[0] is simply dropped (as a python artifact) ?

janiversen, 2025-03-01:
There is a difference in how you define the registers as to how you address them, due to the modbus history.

If you define shared memory, the definition and addressing gets even further apart, because coils are adressed as bits  but defined as registers so a register contains 16 coils.

The simulator datastore is more complicated but works as you want.

janiversen, 2025-03-01:
Closing as this is not a bug, please also read the documentation.

closed 2025-03-01 by janiversen

eabinga, 2025-03-01:
# **Issue: Off-by-One Addressing Issue in `ModbusSlaveContext.getValues()`**

I have discovered an off-by-one issue in `ModbusSlaveContext.getValues()`, which causes the first entry in `ModbusSequentialDataBlock` to be skipped when using a starting address of `0x00`. This behavior results in an unexpected shift in register values and an inaccessible last register.

I am posting this as a follow-up for the benefit of others who might run into the same pitfall. Since this behavior is not clearly documented, it can cause confusion for new users, like me, initializing Modbus data stores.

---

## **Description**
I first encountered this issue while working with PyModbus for the first time. I initialized a `ModbusSequentialDataBlock` as follows:

```python
from pymodbus.datastore import ModbusSlaveContext, ModbusSequentialDataBlock

store = ModbusSlaveContext(
    hr=ModbusSequentialDataBlock(0x00, list(range(1, 11)))  # 10 Holding Registers
)

register_values = store.getValues(3, 0, 10)  # Function Code 3 = Holding Registers

print(register_values)
print(f"Available Registers: {len(register_values)}")
```

### **Expected output**
```
[1, 2, 3, 4, 5, 6, 7, 8, 9, 10]
Available Registers: 10
```

### **Actual output**
```
[2, 3, 4, 5, 6, 7, 8, 9, 10]
Available Registers: 9
```
The first value (`1`) is missing, and only 9 values are returned instead of 10.

---

## **Root Cause**
After investigating, I found that `ModbusSlaveContext.getValues()` modifies the requested address before retrieving values:

```python
def getValues(self, fc_as_hex, address, count=1):
    address += 1  # This shifts all accesses forward by 1
    return self.store[self.decode(fc_as_hex)].getValues(address, count)
```

Because of this `+1` shift:
- A request for **register 0** actually retrieves **register 1**.
- The last register in the range gets **cut off** and is never accessible.

---

## **Conflicting Examples in PyModbus**
There are multiple places in PyModbus where a sequential block is initialized with `0x00`, which does not work correctly due to this shift:

### **1. `setup_updating_server` Example**
```python
datablock = ModbusSequentialDataBlock(0x00, [17] * 100)
```
- This suggests that register `0x00` should be valid.
- However, due to `+1`, a request for `0x00` actually retrieves `0x01`, and the last register is lost.

### **2. `setup_server` (Async Server)**
```python
if args.store == "sequential":
    datablock = lambda: ModbusSequentialDataBlock(0x00, [17] * 100)
```
- Here, the register map starts at `0x00`, but the `+1` shift means that requests for `0x00` access `0x01`, leaving the last register (`0x63`) unreachable.
- The initialization should have used `0x01` instead.

---

## **Why This is a Problem**
I do not believe that it is the intent for some of the initialization data in a sequential block to be lost due to this shift. The fact that official examples use `0x00` as a starting address suggests that the expectation is for these values to be retrievable. However, due to the `+1` shift, part of the data is effectively dropped.

I also considered the workaround of manually **padding the list with an extra dummy value**, but this does not feel like a proper solution. The initialization data should be fully available without requiring artificial adjustments.

---

## **Proposed Fix**
### **1. Update PyModbus Documentation**
- Clearly state that `ModbusSequentialDataBlock` should be initialized with `1`, not `0`, due to the `+1` shift.
- Clarify that users initializing with `0x00` will lose their first register unless they manually pad the data.

### **2. Fix PyModbus Examples**
- Modify `setup_updating_server` and `setup_server` to use:
  ```python
  datablock = ModbusSequentialDataBlock(0x01, [17] * 100)  # Corrected
  ```
- This ensures that all values remain accessible.

### **3. (Optional) Remove the `+1` Shift in a Future Release**
- This would make addressing more intuitive.
- However, I understand that this would be a breaking change, so it would require careful consideration.

---

## **Final Thoughts**
This issue caused confusion while I was getting started with PyModbus, and I suspect other new users might run into the same problem. The fact that PyModbus' own examples use `0x00` as a starting address suggests that this is not just a misunderstanding on my part. 

I would appreciate any clarification on whether this behavior is truly intentional or if it should be addressed. At the very least, I believe the examples and documentation should be updated to reflect how the addressing actually works.

Thanks for your time!

janiversen, 2025-03-02:
Your proposal would break all existing uses, so it is really not interesting.

You forgot to mention that there are different data stores with different functionality and the simulator datastore works as you want it to work.

Also you need to consider how coils should work in your world, since they are addressed as bits not registers.

janiversen, 2025-03-02:
Your suggestion to fix the examples looks good, feel free to submit a pull request