pymodbus-dev/pymodbus #2592: `handle_local_echo` does not work with `ModbusSerialClient` client

brt-han-mai, 2025-03-03:
Environment:
pymodbus 3.8.6
pyserial 3.5

Test script:

```python
from pymodbus.client import ModbusSerialClient
from pymodbus.framer import FramerType
from time import sleep
import logging

FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

try:
    client = ModbusSerialClient(
        port= '/dev/ttyUSB1', # Change with your settings
        framer= FramerType.RTU,
        baudrate= 9600,
        bytesize= 8,
        parity= 'E',
        stopbits= 1,
        name= 'testmodbus',
        timeout= 1,
        handle_local_echo= True,
    )

    if client.connect():
        print(f">>> RESULT: {client.read_holding_registers(address= 0x0000, count= 1,  slave=126)}\n\n")
        sleep(1)
        print(f">>> RESULT: {client.read_holding_registers(address= 0x0002, count= 1,  slave=126)}\n\n")
        sleep(1)
        print(f">>> RESULT: {client.read_holding_registers(address= 0x0004, count= 1,  slave=126)}\n\n")
        sleep(1)
finally:
    print('Bye.')
```

Output:

```
$ python test.py
2025-03-03 16:34:00,650 MainThread      DEBUG    base           :91       Processing: 0x7e 0x3 0x0 0x0 0x0 0x1 0x8f 0xc5
2025-03-03 16:34:00,650 MainThread      DEBUG    decoders       :113      decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)
2025-03-03 16:34:00,650 MainThread      DEBUG    base           :101      Frame advanced, resetting header!!
>>> RESULT: ReadHoldingRegistersResponse(dev_id=126, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)

2025-03-03 16:34:01,651 MainThread      WARNING  serial         :272      Cleanup recv buffer before send: 0x7e 0x3 0x2 0x0 0x7e 0x2d 0xae
2025-03-03 16:34:01,674 MainThread      DEBUG    base           :91       Processing: 0x7e 0x3 0x0 0x2 0x0 0x1 0x2e 0x5
2025-03-03 16:34:01,675 MainThread      DEBUG    decoders       :113      decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)
2025-03-03 16:34:01,675 MainThread      DEBUG    base           :101      Frame advanced, resetting header!!
>>> RESULT: ReadHoldingRegistersResponse(dev_id=126, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)

2025-03-03 16:34:02,677 MainThread      WARNING  serial         :272      Cleanup recv buffer before send: 0x7e 0x3 0x2 0x0 0x3 0xed 0x8f
2025-03-03 16:34:02,695 MainThread      DEBUG    base           :91       Processing: 0x7e 0x3 0x0 0x4 0x0 0x1 0xce 0x4
2025-03-03 16:34:02,696 MainThread      DEBUG    decoders       :113      decoded PDU function_code(3 sub -1) -> ReadHoldingRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)
2025-03-03 16:34:02,697 MainThread      DEBUG    base           :101      Frame advanced, resetting header!!
>>> RESULT: ReadHoldingRegistersResponse(dev_id=126, transaction_id=0, address=0, count=0, bits=[], registers=[], status=1)

Bye.
```

As you can see the "Cleanup recv buffer before send" is actually the response data of the previous request. Local echo was recognized as a response.

janiversen, 2025-03-03:
Well I really cannot see that local echo was received as response, because I cannot see what is actually being send. Please add a debug log, as the issue template asks for (of course it might also be your log editing was far too restrictive).

Looking at the code, without a debug log is difficult, but I think you found a bug. The bug is only in the sync client, NOT in the async client.

closed 2025-03-03 by janiversen

brt-han-mai, 2025-03-04:
It seems that the github issue template does not work when I create the issue. I updated the script and the result. The issue does not happen with `AsyncModbusSerialClient`. I see you have done the fix. Thank you.

PR #2593 (2025-03-03): correct handle_local_echo for sync client.

fixes #2592