pymodbus-dev/pymodbus #2594: New Modbus client for DTU usage

hrbonz, 2025-03-04:
I have several customer cases now where a DTU (Data Transmission Unit) is creating a TCP connection to a server, that connection is then used to transmit modbus requests from the server to the DTU. The machine(s) connected over RS485 behind the DTU receive those requests and answer accordingly. This is a type of architecture that I haven't seen covered in python modus packages (umodbus / pymodbus). It allows to communicate with devices behind  a simple router, firewall for example.

The way I want to implement this is by simply subclassing `ModbusTCPClient` and provide it the socket directly rather than letting the client class create it. I can then use this class in a `handle()` method of a `socketserver.StreamRequestHandler`.

There are a few more features that I want to add to my implementation but I don't know if they would make sense for pymodbus. A DTU can send some sort of packet authentication in the form of a string at connection or leading any packet sent. The first case can be managed in the `setup()` method of the TCP server, the second case could probably be handled with a specific `trace_packet` function. A DTU can also send a heartbeat message, I'm not sure where to handle that yet as I'm not using that feature at the moment.

Would you be interested in a PR covering this use case?

janiversen, 2025-03-04:
I have no idea what umodbus is, but it is not true that pymodbus do not cover it, we even have an example showing a skeleton implementation working at datastore level.

However the current implementation works at request/response level and only for data requests. A proper implementation would have to be done at pdu/framer level. The pdu is transport independent (framer + comms) and are therefore the correct point to act as a switchboard.

We are always open to pull requests, but you need to consider the level you want to achieve, because a true "serial forwarder" (the name normally used) is surely a couple of pull requests.

It is in our plans to make a true serial forwarder but to be honest a very low priority so if you want to take on the challenge please.

janiversen, 2025-03-04:
And btw I think you are mixing the word "server" in modbus it is the client that request data from the server.

janiversen, 2025-03-04:
Reread your text....if I get it right, you want the modbus server (sitting behind a firewall and connecting to rs485) to connect to the client. This is not modbus standard but is clearly doable.

Now the client normally connect to the server, and instead you want it to listen.....that breaks all the retry/reconnect code, so it is a bit more complicated than just supplying the socket.

What I still do not understand is whether you want to make a new client or a new serial forwarder (what you call DTU) or both....

A more modern variant would be to use TLS for security and add a virtual host in the firewall.

hrbonz, 2025-03-04:
I think there's some misunderstanding on what I'm working on, I'm not surprised, it took me a minute to wrap my head around it the first time. I'll try to use TCP server/client and modbus client to make sure the explanation makes sense.
A DTU unit allows to connect a modbus network to a TCP/IP network through Wifi/RJ45/LoRa/mobile, in that sense, it acts as a sort of bridge. The DTU can usually work in different modes such as:
* TCP server: you can then connect to the DTU as you would to a modbus server and query data
* TCP client: the DTU will connect to a TCP server and forward any data you send on the line to the RS485 side. Some DTUs can do some clever Modbus RTU to TCP conversion both ways.
* UDP server & UDP client have similar use.
* some even have HTTP or MQTT capacities

Overall, the DTU makes your modbus network a bit of an IoT network.

In my case, I have some energy and water meters to collect data from. I can't use the TCP server mode because that would require exposing an unsecure port to internet on the meter side. For that reason, I use the DTU as a TCP client that connects to one of my servers running a generic TCP server. Using `socketserver.StreamRequestHandler`, this is trivial and will fork a process for each connection. When the DTU connects to the TCP server, I now have a socket available. I can use that socket to create a modbus client and send requests to the DTU that will be relayed to the RS485 side. The answers are similarly relayed. The forked socket calls a `handle()` method on a handler class you provide to the server. The handler is an infinite loop (think `serve_forever()`) that sends a modbus request, reads the response then sleeps for a defined amount of time.

At the end of the day, you can use a standard TCP/RTU PDU, you just don't initiate the TCP connection from your side.
It's exactly like the `ModbusTcpClient` with the only difference that the socket is already created by the TCP server. So instead of calling `ModbusTcpClient` with host and port, I am calling `ModbusSocketClient` (name TBD) with a [`socket.socket`](https://docs.python.org/3/library/socket.html#socket-objects) object which is what the `connect()` method on your TCP client does anyway. Everything else is untouched

See below for the architecture:

```
                                                   xxxx    xxxxxx                                                                    
                                             xxxxxx     xxxx     x                                                                   
                                            xx                    x                                                                  
                                           xx                     xx x xxx                                                           
                                           x                             xx                                                          
                                           x                              x                                                          
                                          x                               x                                                          
                                         xx                              xx                                                          
┌────────┐  RS485   ┌─────┐              x             Internet         xx             ┌────────┐                                    
│ modbus │          │ DTU │    TCP       x                                x     TCP    │ python │ socket handler                     
│ device ┼──────────┼     ├──────────────xxxx                             xx───────────┼  TCP   │ loop to manage modbus communication
└────────┘          └─────┘               xx                              xx           │ server │                                    
                                          x                               x            └────────┘                                    
                                          x                            x                                                             
                                           x            x           x                                                                
                                            xx        xxxxx      x                                                                   
                                               x xx x      xx                                                                        
```

[umodbus](https://github.com/AdvancedClimateSystems/uModbus) is another modbus python package that I've been using but I find it has a lot of annoying behaviors that make debugging really difficult.

I'll probably just push a PR with my code, in the end it's actually fairly simple for `ModbusSocketClient`. I still wonder if the code handling the TCP client connection should reside in your package or maybe just as an example.

hrbonz, 2025-03-04:
> What I still do not understand is whether you want to make a new client or a new serial forwarder (what you call DTU) or both....
The DTU is a piece of hardware so I'm limited in the ways it behaves on the TCP side.
See those common machines:
* https://www.cdebyte.com/Modem
* https://www.pusr.com/products/serial-to-ethernet-converters.html

hrbonz, 2025-03-04:
> Reread your text....if I get it right, you want the modbus server (sitting behind a firewall and connecting to rs485) to connect to the client. This is not modbus standard but is clearly doable.
Correct

> Now the client normally connect to the server, and instead you want it to listen.....that breaks all the retry/reconnect code, so it is a bit more complicated than just supplying the socket. And I am pretty sure you also need a new transaction manager.
I don't think it's super complicated, the retry/reconnect is simply out of scope for this use.
I'll look more at transaction manager code to understand what you're saying, I've only been looking at this project for a couple days.

janiversen, 2025-03-04:
Your DTU is exactly what in modbus terms is called a serial forwarder....a serial forwarder is basically a modbus server and multiple modbus clients connected together with the added logic that the server connect to the client.

Of course you can modify the client to accept a socket instead of a host/port that is simple....BUT if the socket is controlled outside of pymodbus you need to replace all the retry/reconnect code, and that happens outside the modbustcpclient class.

If you submit a PR please do not forget to add tests, that shows it works..and especially that reconnect/retry works.

janiversen, 2025-03-04:
Something is off in your drawing....the "python tcp server" is surely not a modbus server because a modbus server is a device. 

You miss the modbus client, which I assume sits after your "python tcp server" and in that case could just be a normal client.

janiversen, 2025-03-04:
I think it would be a lot easier for you to make a "dtu server". It should listen for dtu connects as well as modbus client connects using e.g the tcp/ip address as identifier to control which dtu talks to which client (or all dtu talk to one client). Once the dtu server have the 2 sides connected it simply passes the data.

janiversen, 2025-03-04:
Exposing the socket is not a feature we generally want, we have had the request many times and every time the complexity when problems occurs (like a bad response) raise sharply.

That said a special client with very restricted parameters is an option.....if it comes with a lot of testing. We are simply trying to secure it do come with added maintenance costs, which we have seen far too often in the past.

janiversen, 2025-03-04:
Closing as this is not an issue with pymodbus, but more a discussion about a new feature.

closed 2025-03-04 by janiversen

hrbonz, 2025-03-05:
@janiversen hopefully you see in the PR I'll push that there is a benefit here and actually very little impact on the overall code.