pymodbus-dev/pymodbus #2606: async serial crc check may hang

dcneeme, 2025-03-12:
ubuntu 20.4 on arm, python3.9, pymodbus 3.8.2

got stuck into CRC calculation (pymodbus/framer/rtu.py)? 100% CPU, everything frozen. happens daily under heavy load.
	
(venv_pymodbus3) root@K20kyte:~/pyapp# py-spy dump --pid 82492
Process 82492: python3 /root/pyapp/iomain3.py
Python v3.9.5 (/usr/bin/python3.9)

Thread 0x7F7D26E010 (active+gil): "MainThread"
    compute_CRC (pymodbus/framer/rtu.py:156)
    check_CRC (pymodbus/framer/rtu.py:142)
    decode (pymodbus/framer/rtu.py:122)
    _processIncomingFrame (pymodbus/framer/base.py:94)
    processIncomingFrame (pymodbus/framer/base.py:76)
    callback_data (pymodbus/transaction/transaction.py:187)
    datagram_received (pymodbus/transport/transport.py:338)
    data_received (pymodbus/transport/transport.py:304)
    intern_read_ready (pymodbus/transport/serialtransport.py:134)
    _run (asyncio/events.py:80)
    _run_once (asyncio/base_events.py:1890)
    run_forever (asyncio/base_events.py:596)
    run_until_complete (asyncio/base_events.py:629)
    <module> (iomain3.py:345)
Thread 0x7F6E05A1E0 (active): "asyncio_0"
    _worker (concurrent/futures/thread.py:75)
    run (threading.py:892)
    _bootstrap_inner (threading.py:954)
    _bootstrap (threading.py:912)

janiversen, 2025-03-12:
Please add a debug log as pr issue template, so we can see the frame causing the problem.

Without knowing the frame, we have no idea what the problem might be...there are no infinite loops in the crc calculation.

janiversen, 2025-03-12:
This is the loop:
```

for data_byte in data:
            idx = cls.crc16_table[(crc ^ int(data_byte)) & 0xFF]
            crc = ((crc >> 8) & 0xFF) ^ idx

```
it is hard to see how that can cause a system to freeze, unless there was some really big data was received.

janiversen, 2025-03-12:
It is more likely that the problem is not in the crc calculation but that pymodbus receives data continuously, a debug log will show thst clearly.

dcneeme, 2025-03-12:
i never send another request before a response or timeout is received. 
i'll try to catch the debug log, it may be a bit difficult.

janiversen, 2025-03-16:
seems to be solved.

closed 2025-03-16 by janiversen