pymodbus-dev/pymodbus #2607: No support for RTU over TCP?

silvioporto, 2025-03-14:
Did support for the FramerType.RTU_OVER_TCP get dropped? The documentation still says "support all standard frames: socket, rtu, rtu-over-tcp, tcp and ascii", but the only options in the FramerType class are "socket, rtu, ascii, tls".

janiversen, 2025-03-14:
Try to read the documentation, that helps.

Framertype.rtu with a tcp client, is magically rtuovertcp.

janiversen, 2025-03-14:
Closing as pymodbus support the rtu_over_tcp protocol (not as a separate framer though, which is not needed).

closed 2025-03-14 by janiversen

silvioporto, 2025-03-14:
I was referring to the "Common features" section in https://pymodbus.readthedocs.io/en/latest/ and also the "Client protocols/framer" table in https://pymodbus.readthedocs.io/en/latest/source/client.html. The way those are described made me think that there was a dedicate FramerType class for the RTU_OVER_TCP. I had tried to use a TCP client with the FramerType.RTU but didn't have any success. Thanks for the clarification though, the library is great, i'll keep trying.

janiversen, 2025-03-14:
Well you did not read the header it says "protocol" not framerType.

TID, PID is not part of the RTU protocol nor RTU_OVER_TCP (as this is identical to RTU, just another transport).

Seems you have a misunderstanding of the standard, TID/PID is part of the socket protocol.... so either you need TID/PID and use framertype.socket or not and use framertype.rtu.