pymodbus-dev/pymodbus #2608: serial_forwarder broken after updating to 3.8

rhpijnacker, 2025-03-14:
To connect my Home Assistant to my Heat Recovery unit in the attic, I've been running the serial forwarder for over a year now.
See code at https://github.com/rhpijnacker/homeassistant-addon-pymodbus-serial_to_tcp_forwarder.

Recently I have updated it to 3.8, and since then it is regularly failing:

```
Exception response 128 / 4
Datastore unable to fulfill request: Modbus Error: [Input/Output] ERROR: No response received of the last requests (default: retries+3), CLOSING CONNECTION.; Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/pymodbus/server/requesthandler.py", line 112, in handle_request
    response = await self.last_pdu.update_datastore(context)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/pdu/register_message.py", line 39, in update_datastore
    values = cast(list[int], await context.async_getValues(
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/datastore/context.py", line 43, in async_getValues
    return self.getValues(fc_as_hex, address, count)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/datastore/remote.py", line 47, in getValues
    self.result = func_fc(_address, _count)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/datastore/remote.py", line 82, in <lambda>
    "h": lambda a, c: self._client.read_holding_registers(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/pymodbus/transaction/transaction.py", line 118, in sync_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received of the last requests (default: retries+3), CLOSING CONNECTION.

```

Any idea what could be causing this?
Is this still expected to be working?

janiversen, 2025-03-14:
It tells you in the last line "no response received".

You do not write from which version you upgrade, but did you remember to correct your code. All 3.x versions have smaller API changes (3.x.y do not).

rhpijnacker, 2025-03-14:
Sorry, I updated from 3.6.9.
I did have to make some small [code changes](https://github.com/rhpijnacker/homeassistant-addon-pymodbus-serial_to_tcp_forwarder/commit/83d0e1d6dda06699f81a5b4777f619da00a5afca) to make things work with 3.8.

Please let me know if more is needed.

janiversen, 2025-03-14:
I am not going to review your code sorry, that is purely your problem.

If there is a bug in pymodbus, please open an issue with debug log.

rhpijnacker, 2025-03-14:
Fair enough.

Since I'm not exactly sure how all of this works, and I basically just copied the serial_forwarder example, I will just revert to 3.6.9.

closed 2025-03-14 by rhpijnacker