pymodbus-dev/pymodbus #2609: Passing in a `device_id` to `read_holding_registers` causes error `got an unexpected keyword argument 'device_id'`

8none1, 2025-03-15:
I'm using version 3.8.6 from pip and talking to an EW11 modbus TCP adapter.

According to the [docs ](https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.read_input_registers) I can pass in a `device_id`.

However, when I do this I get the error:

```text
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'device_id'
```

When ommiting `device_id` altogether things work as expected.

## Example code

```python
from pymodbus.client import ModbusTcpClient
c = ModbusTcpClient(host="hostname")
c.connect()
r = c.read_holding_registers(47, count=7, device_id=1)
```
### Error

```text
Traceback (most recent call last):
  File "<python-input-4>", line 1, in <module>
    r = c.read_holding_registers(47, count=7, device_id=1)
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'device_id'
```

janiversen, 2025-03-15:
Looks a lot like the pymodbus version is NOT v3.8.6

look at the code:
```

def read_holding_registers(self,
                               address: int,
                               *,
                               count: int = 1,
                               device_id: int = 1,
                               no_response_expected: bool = False) -> T:
        """Read holding registers (code 0x03).

        :param address: Start address to read from
        :param count: (optional) Number of registers to read
        :param device_id: (optional) Modbus device ID
        :param no_response_expected: (optional) The client will not expect a response to the request
        :raises ModbusException:

        This function is used to read the contents of a contiguous block
        of holding registers in a remote device. The Request specifies the
        starting register address and the number of registers.

        Registers are addressed starting at zero.
        Therefore devices that specify 1-16 are addressed as 0-15.
        """
       
```

janiversen, 2025-03-15:
Are you sure you do not another version installed e.g. globally that that precedence.

Did you verify with pip that v3.8.6 is actually installed?

8none1, 2025-03-15:
I thought the same thing, but everything I've tried says it's `3.8.6`:

```text
$ pip3 show pymodbus
Name: pymodbus
Version: 3.8.6
Summary: A fully featured modbus protocol stack in python
Home-page: https://github.com/pymodbus-dev/pymodbus/
Author: Galen Collins, Jan Iversen
Author-email: 
License: BSD-3-Clause
Location: /home/will/source/growatt_modbus/.venv/lib/python3.13/site-packages
Requires: 
Required-by: 
```

```text
$ python3
Python 3.13.2 (main, Feb  5 2025, 01:23:35) [GCC 14.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymodbus
>>> print(pymodbus.__version__)
3.8.6
>>> 
```

🤷

8none1, 2025-03-15:
... and yet, when I go in to `site_packages` and look in `mixin.py`:

```python
    def read_holding_registers(self,
                               address: int,
                               *,
                               count: int = 1,
                               slave: int = 1,
                               no_response_expected: bool = False) -> T:
        """Read holding registers (code 0x03).

        :param address: Start address to read from
        :param count: (optional) Number of registers to read
        :param slave: (optional) Modbus slave ID
        :param no_response_expected: (optional) The client will not expect a response to the request
        :raises ModbusException:

        This function is used to read the contents of a contiguous block
        of holding registers in a remote device. The Request specifies the
        starting register address and the number of registers.

        Registers are addressed starting at zero.
        Therefore devices that specify 1-16 are addressed as 0-15.
        """
```

I'm clearly not running the actual `3.8.6`.

So I dunno.  Something funky happened.  I'll delete the venv and start again.  If I can recreate this issue I will re-open the issue but otherwise this is something at my end.  Sorry for the noise.

closed 2025-03-15 by 8none1