pymodbus-dev/pymodbus #2613: Read too many holding registers does not result in correct Exception Code

MuellerDaniel, 2025-03-17:
A ReadHoldingRegistersRequest of more than the allowed 125 registers does not result in an Exception Response 0x03 (Illegal value).
The `encode()` method of ReadHoldingRegistersRequest calls the `ModbusPDU.validateCount()`, which raises a ValueError. 

For WriteMultipleRegistersRequest, the count validation is performed in the `update_datastore()` method and the correct exception code is returned.

Is this behaviour intended?

janiversen, 2025-03-17:
You mix things, pymodbus detects that the message is too big, and do not execute it. The response you refer to comes from the device.

janiversen, 2025-03-17:
If you feel the writemultioleregisters should have a limit check, then feel free to submit a PR.

closed 2025-03-17 by janiversen