pymodbus-dev/pymodbus #2614: The `execute()` method's return type hint doesn't include `ExceptionResponse`

david-drinn, 2025-03-19:
This is important for auto-completion and self-discovery of available attributes. It also results in false errors/warnings when referencing `ExceptionResponse` common attributes, such as `exception_code`.

For example, the last line will result in a warning of "Unresolved attribute reference 'exception_code' for class 'ModbusPDU'" in PyCharm IDE:

```
import pymodbus
import pymodbus.pdu.file_message as pdu_file_msg

def test_illegal_function(client: pymodbus.client.ModbusSerialClient):
    mb_exc = client.read_fifo_queue(slave=pydm.Pdm.DEV_ID)

    assert type(mb_exc) == pymodbus.pdu.ExceptionResponse
    assert mb_exc.function_code == pdu_file_msg.ReadFifoQueueRequest.function_code + 0x80
    assert mb_exc.exception_code == pymodbus.ExceptionResponse.ILLEGAL_FUNCTION
```

But the code works, since the expected response is an `ExceptionResponse` in this case.

janiversen, 2025-03-19:
Feel free to make a pull request.
, but it is the wrong place to change signature, that change will not help with your example, since it does not change the signature of your call.

janiversen, 2025-03-19:
But apart from that the signature of the call is correct, because ExceptionReponse inherits from ModBusPDU:+

see code:
`
class ExceptionResponse(ModbusPDU): 
`

at least in v3.8.6 which is the latest version.

PR #2615 (2025-03-19): Expose exception_code to API.

fixes #2614
closed 2025-03-19 by janiversen