pymodbus-dev/pymodbus #2620: How to implement a custom server that can be busy?

shyd, 2025-03-31:
Hi,
in my project, I have custom contexts that communicate with external hardware. Therefore I derive from `ModbusSparseDataBlock` and overwrite `validate`, `setValues` and `getValues` with my business logic.

For example I have a coil that takes several seconds to process data on the external hardware. A response look like this even the device is busy.

```
WriteSingleCoilResponse(dev_id=1, transaction_id=1, address=6, count=0, bits=[True], registers=[], status=1)
```

Now my question is how to use busy flag that the client gets the according response that the device is currently busy? If I use a lock the response would timeout because the lock cannot be acquired in time.

Thank you very much for this project!

janiversen, 2025-04-01:
Modbus do not have a busy flag, please explain which message/flag you are referring to ?

janiversen, 2025-04-01:
Yes, but that is received as an exception response, not a normal response. BUT the pymodbus server do not use this exception response.

If you receive that from the device in your datastore-client, you should pass it on...which is not very easy. It is something that is not foreseen in the current server implementation.

If you just need to wait the only option is a wait_for with timeout.

janiversen, 2025-04-01:
The server context or datastores are designed to respond immediately, delays are not permitted. Also datastores are not designed to return anything but a value (or False in the validate function), so your point of attack seems wrong.

janiversen, 2025-04-01:
Did you check the ExceptionCode.

closed 2025-04-01 by janiversen