pymodbus-dev/pymodbus #2621: Default ModbusSlaveContext will share the same dataBlocks for every instance

maximeLeurent, 2025-04-01:
Hello,

For my project, I used several Modbus Server in one unique process, each server in their own thread.

I realise that that each device was searching the same memory, as two devices writing at the same register location different values would rewrite their values.

I found the cause of this sharing memory in the definition of the __init__ function of the class ModbusSlaveContext

```
class ModbusSlaveContext(ModbusBaseSlaveContext):
    def __init__(self, *_args,
                    di=ModbusSequentialDataBlock.create(),
                    co=ModbusSequentialDataBlock.create(),
                    ir=ModbusSequentialDataBlock.create(),
                    hr=ModbusSequentialDataBlock.create(),
                    zero_mode=False):
        """Initialize the datastores."""
        self.store = {}
        self.store["d"] = di
        self.store["c"] = co
        self.store["i"] = ir
        self.store["h"] = hr
        self.zero_mode = zero_mode
```
I wasn't calling ModbusSlaveContext with any di, co, ir, or hr variable
therefore all self.store["d"]  was using the same instance of ModbusSequentialDataBlock

it is the same problem of default argument of this piece of code:
```
def f(l =[]):
    l.append(3)
    print(l)

f()
>>>[3]
f()
>>>[3, 3]
```

I propose you the following code

```
class ModbusSlaveContext(ModbusBaseSlaveContext):
    """Create a modbus data model with data stored in a block.

    :param di: discrete inputs initializer ModbusDataBlock
    :param co: coils initializer ModbusDataBlock
    :param hr: holding register initializer ModbusDataBlock
    :param ir: input registers initializer ModbusDataBlock
    :param zero_mode: Not add one to address

        When True, a request for address zero to n will map to
        datastore address zero to n.

        When False, a request for address zero to n will map to
        datastore address one to n+1, based on section 4.4 of
        specification.

        Default is False.

    """

    def __init__(self, *_args,
                    di: ModbusSequentialDataBlock=None,
                    co: ModbusSequentialDataBlock=None,
                    ir: ModbusSequentialDataBlock=None,
                    hr: ModbusSequentialDataBlock=None,
                    zero_mode=False):
        """Initialize the datastores."""
        self.store = {}
        self.store["d"] = di or ModbusSequentialDataBlock.create()
        self.store["c"] = co or ModbusSequentialDataBlock.create()
        self.store["i"] = ir or ModbusSequentialDataBlock.create()
        self.store["h"] = hr or ModbusSequentialDataBlock.create()
        self.zero_mode = zero_mode
```

janiversen, 2025-04-01:
Pull requests are welcome.

closed 2025-04-01 by janiversen

janiversen, 2025-04-01:
Closing as this is not a pymodbus problem.

maximeLeurent, 2025-04-01:
I am quite on rush with work today but I will provide you an piece of code to show the problem

janiversen, 2025-04-01:
Well they are 2 different python objects, so in that respect there are not the same, but they are of same type and with same content so python "is" operator declares they are the same, but look in with a debugger and you will see they have different addresses.

You are not testing pymodbus but python and its capability to make objects.

You missed to notice something far more important, namely the scope of self.values.

janiversen, 2025-04-01:
You do not need to insist, but you are referencing an article written in 2018, python have evolved a bit since then, however as I wrote earlier pull requests are welcome, without that this is just talk.

Be aware though that the datastores are being replaced by a class that allows you to model your device, this is targeted for v4.0.0.

And if you do not want to make a pull request, then add the parameters.