pymodbus-dev/pymodbus #2623: convert_from_registers wrong transformation

milannvidia, 2025-04-03:
print(client.convert_from_registers(registers=[0,8194], data_type=client.DATATYPE.BITS, word_order="big"))
>>> [False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, True, False, False, False, True, False, False, False, False, False, False] 

whilst it should give 
>>>[False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, False, True, False, False, False, False, False, False, False, False, False, False, False, True, False]

Is this a bug?

janiversen, 2025-04-03:
Sounds like a bug, but why do you want to convert bits ??? when you use e.g. read_coils it is already unpacked.

Pull requests are welcome.

A single register show the same problem (from dev):
```
>>> print(client.convert_from_registers(registers=[8194], data_type=client.DATATYPE.BITS))
[False, False, False, False, False, True, False, False, False, True, False, False, False, False, False, False]
```

janiversen, 2025-04-03:
Bug confirmed:

```
>>> from pymodbus.utilities import pack_bitstring, unpack_bitstring
>>> 
>>> a = int.to_bytes(1409, 2, "big")
>>> print(a)
b'\x05\x81'
>>> b = unpack_bitstring(a)
>>> print(b)
[True, False, True, False, False, False, False, False, True, False, False, False, False, False, False, True]
>>> c = [True] + [False]*6 + [True, True, False, True] + [False]*5
>>> d = pack_bitstring(c)
>>> print(d)
b'\x81\x05'
>>> 
```

This is really a very very old bug (since v1.something). Good catch.

janiversen, 2025-04-07:
#2627 solves the problem. Thanks for catching it, it was really a deeply buried bug and most likely affect a number of apps using pymodbus.

Extensive testing have been added to ensure it does not happen again.