pymodbus-dev/pymodbus #2630: read_coils() report wrong bit order in 3.9.0

chkr-private, 2025-04-08:
My modbus device is reporting the following coils starting at address 0x100: true, false, true, false, true, true, ... .

pymodbus 3.8.6 returned the bits in the correct order:

```
>>> client.read_coils(0x100, count=16).bits
[True, False, True, False, True, True, True, True, True, True, True, True, True, True, True, True]
```

pymodbus 3.9.0 returns the values byte-wise swapped:
```
>>> client.read_coils(0x100, count=16).bits
[True, True, True, True, True, True, True, True, True, False, True, False, True, True, True, True]
```
It seems, that pymodbus 3.9.0 misinterprets the on-wire protocol.

According to the Modbus Application Protocol 1.1b document, section 6.1, the data bytes should be interpreted one by one starting with the first byte and the bits should be interpreted starting at the LSB for each byte.

In my case, the device returns 2 data bytes: 0xf5, 0xff and so the bits should be interpreted as follows:
- starting with the first byte (0xf5), interpreting the bits starting at LSB: true, false, true, false, true, true, true, true
- then the 2nd byte (0xff), starting at LSB: true, true, ...

The modbus dissector in wireshark confirms it:

[image]

It looks like that commit d47dc65a7995a19e9d004318ff7b050d8f82f2a5 has broken the bit order in 3.9.0.

janiversen, 2025-04-09:
hmm, could be a bug, I will take a look later

janiversen, 2025-04-09:
bug confirmed, this will be released as v3.9.1

closed 2025-04-09 by janiversen